jr8ppg/zLog #43: FT-991のUSBサウンド対応

jr8ppg, 2020-11-06:
from zlog-reiwa
FT-991のサウンド対応

→WAVファイルは出力先の選択ができた(11/4)
→MP3はどうする？？

7M4KSC, 2020-11-07:
MP3は、Win32 APIが無いので、外部.DLL(アプリのDLL)をロードするしか再生方法が無いみたいです。
[https://qiita.com/tera1707/items/90453fa9351209e0373c](https://qiita.com/tera1707/items/90453fa9351209e0373c)

jr8ppg, 2020-11-11:
WAV,MP3の対応含め目処が立った

jr8ppg, 2020-11-13:
完了

jj1cvh, 2020-11-13:
Escキーを押すとapplication error画面がポップアップする問題が発生しています。

(動作環境) Nov 13 2020 v2510 d97ee57, Win10 64bitマシン

(再現手順)
1. zlog.iniを削除。
2. zlog.exeを実行し起動。
3. SSB/FM/AMモードの状態でEscを押すと、application error画面がポップアップする。
4. ボイスメモリ#1を一度再生すれば、その後はEscを押しても大丈夫。

(application error画面の内容)
32bitビルド:
> There was an error during the execution of this program.
The application might become unstable and even useless.
It's recommended that you save your work and close this application.
>
> アドレス FF3EE892 でアドレス FF3EE892 に対する読み取り違反がおきました。.

64bitビルド:
> There was an error during the execution of this program.
The application might become unstable and even useless.
It's recommended that you save your work and close this application.
>
> nil を TTimer に代入することはできません.

zlog.exeと同じフォルダにzlog.exe-exception-hogehoge.txtが出力されています。
もし再現しない場合はファイルをお送りしますので、お知らせください。

jr8ppg, 2020-11-13:
現象確認しましたので修正しました。

jj1cvh, 2020-11-13:
Nov 14 2020 v2510 8555154 の32bitビルドと64bitビルドで不具合が解消していることを確認しました。
ご対応ありがとうございました!

closed 2020-11-13 by jj1cvh