jr8ppg/zLog #58: マルチOPde

jr8ppg, 2020-11-06:


jr8ppg, 2020-11-21:
完成！

jj1cvh, 2021-02-14:
従来のzlog.opは無視されて、zlog_oplist.iniに新たにオペリストを保存する必要がある、という理解でよろしいでしょうか？
マルチオペ局にとってはzlog.opのフォーマット変更はインパクトがあるため、wavファイルを切り替える必要が無い場合は既存のzlog.opファイルをそのまま使えると嬉しいのですが。

また、ACAGで起動して試したのですが、
zlog.iniの[Profiles]のデフォルト値
 Power1.9MHz=H
 ...
 Power10GHz=H
が優先されてしまい、Opごとに設定した
 Power=HHHHHHHMMLPPP
が反映されませんでした。Opごとのパワーの書き方はどうすればよろしいでしょうか？

jr8ppg, 2021-02-14:
了解しました。

1点目
zlog_oplist.iniが無い or 0件の場合は、zlog.opファイルも参照するようにします。
（ロードのみで次からはzlog_oplist.iniに保存される）
年齢は対応無しで良いですかね。（いずれにしても毎年修正が必要です）

2点目
これはバグですね。
OPを切り替えてPowerが変わるのは確認したのですが、何を見てOKとしたかは不明です。（お恥ずかしい）

jj1cvh, 2021-02-14:
ご確認ありがとうございます。

1点目と2点目、了解です。zlog.opに
JJ1CVH H
と記録されていた場合、[Profiles]の設定に関わらず切替可能な全バンドでpwr欄がHになる感じですかね。

年齢の入力欄は特に必要としていません。私自身がALL ASIAに参加する場合、「平均年齢」を事前に決めて、全オペが同じ送信ナンバーを送っています。

jr8ppg, 2021-02-14:
> zlog.opに
> JJ1CVH H
> と記録されていた場合、[Profiles]の設定に関わらず切替可能な全バンドでpwr欄がHになる感じですかね。
そうですね。
バンド数分のPowerコードが無い場合は、無い部分について１文字目が使用されます。
と書きながら確認するとそうなっていないので修正しました。

jj1cvh, 2021-02-20:
Version 2.6 Beta3で、全市全郡・Multi-Opを選択して検証しました。3点ご報告します。

1. zlog.opに
JJ1CVH M
と記載されていた場合、1.9Mには反映されますが、他のバンドはPreferencesタブの設定値(zlog.iniの[Profiles])のHになってしまう点が思った動作と異なります。

[Categories]タブでJJ1CVHをダブルクリックしてOKを押せば、
M------------
に自動的に変換されて、他のバンドもMが選択できるようになります。

2. zlog.opに
JJ1CVH/1            M
JJ1CVH/2            M
JJ1CVH/3            M
JJ1CVH/4            M
JJ1CVH/5            P
と記載されていて、JJ1CVH/3を選択した場合、1.9はM、3.5-5600はHになりますが、10GだけはなぜかPになってしまいます。

3. zlog.opの最終行に1行空行を入れると、
zlog起動->コンテスト選択画面で[OK]-->zloファイルを指定して[OK]
の後に、
モジュール 'zlog.exe' のアドレス 00758AE7 でアドレス 00000000 に対する読み取り違反がおきました。.
のエラー画面がポップアップします。[OK]を押せば引き続き使用できます。

jr8ppg, 2021-02-20:
現象確認しました。
１、３はその通りですね。
２は再現に至っていませんが１と同様原因かと思います。

closed 2021-03-02 by jr8ppg