jr8ppg/zLog #65: サイドトーン機能の復活

jr8ppg, 2020-11-06:
サイドトーン機能は復活できないか？

Media Foundationを検討してみた
https://sourceforge.net/projects/mfpack/files/MFPack/

jj1cvh, 2020-11-06:
On 2020/08/26 0:22, JJ1CVH wrote:
マルチオペの6D参加時かつ2.4G&UpかつF2キーイングが必要な場合限定ですが、Win XPマシンで利用し続けております。

2.4G&Upは局数は少ないので、古いXPマシンを割り当てて、リグのマイク端子に接続する自作インタフェースを介して、F2キーイング時のCWサイドトーン機能をボイスCQと同様に(稀にですが)利用しております。

jr8ppg, 2020-11-09:
win64のreleaseモードで原因不明のアプリケーションエラーとなるので、WAVEデータ格納用バッファを３倍に増やす事でとりあえずは回避。出力先デバイスの選択はこれから。

jr8ppg, 2020-11-13:
完了

closed 2020-11-13 by jr8ppg

jj1cvh, 2020-11-13:
サイドトーン機能の復活、ありがとうございます!
Nov 13 2020 v2510 d97ee57, Win10 64bitマシンで動作確認し、PC Phone出力からCWメッセージが聞こえることを確認しました。

4点ほどフィードバックします。

1. USBIF4CWに接続したパドルの操作時にサイドトーンがPCスピーカから聞こえません。2.2eでは聞こえていたような気がします。

2. PAUSE CWボタンとCW Keyboardボタンの間にあったSide toneボタンも復活していただけると嬉しいです(優先度低で結構です)。

3. ビルドは成功するのですが、32bit、64bitともにビルド時の警告が10個出ています。このままでOKでしょうか？
32bitビルド時の警告:
> [dcc32 警告] ToneGen.pas(734): W1024 符号付型と符合無し型の演算による、オペランドの拡張
[dcc32 警告] ToneGen.pas(761): W1024 符号付型と符合無し型の演算による、オペランドの拡張
[dcc32 警告] ToneGen.pas(771): W1024 符号付型と符合無し型の演算による、オペランドの拡張
[dcc32 警告] ToneGen.pas(781): W1024 符号付型と符合無し型の演算による、オペランドの拡張
[dcc32 警告] ToneGen.pas(800): W1024 符号付型と符合無し型の演算による、オペランドの拡張
[dcc32 警告] ToneGen.pas(848): W1024 符号付型と符合無し型の演算による、オペランドの拡張
[dcc32 警告] ToneGen.pas(858): W1024 符号付型と符合無し型の演算による、オペランドの拡張
[dcc32 警告] ToneGen.pas(869): W1024 符号付型と符合無し型の演算による、オペランドの拡張
[dcc32 警告] ToneGen.pas(871): W1024 符号付型と符合無し型の演算による、オペランドの拡張
[dcc32 警告] ToneGen.pas(985): W1024 符号付型と符合無し型の演算による、オペランドの拡張

4. もし、Side tone機能未対応のOS(XPとか)があれば、沢山PCを使うマルチオペ局での2.2eから令和EditionへのPC毎の移行判断材料となりますので、制限事項としてどこかに記載していただけると助かります。

jr8ppg, 2020-11-13:
取り急ぎ
1.2.は後で確認します。
3.は問題ありません。
4.は特に制限はありませんが、MP3についてはXPで確認した方が良いですね。

jj1cvh, 2021-03-28:
> USBIF4CWに接続したパドルの操作時にサイドトーンがPCスピーカから聞こえません。2.2eでは聞こえていたような気がします。

Mar 27 2021 0bb9a5c をビルドし、Win10 64bitマシンで サイドトーンを確認してみましたが、やはりUSBIF4CWに接続したパドルの操作時にサイドトーンがPCスピーカから聞こえませんでした。

jr8ppg, 2021-03-28:
パドルのサイドトーンですね。
旧zLogはエレキー動作をzLogでやっていたので、サイドトーンを鳴らすことが一応できていましたが、
今はパドル操作時のエレキー動作をUSBIF4CWの機能でやっているので無理ですね。
（zLog側ではパドルのON/OFFはとれるけど、USBIF4CWが行うキーイング信号は取れないため）