jr8ppg/zLog #66: QSO Rateグラフの過去ログとの重ね書き表示機能

jr8ppg, 2020-11-06:
V2.6候補
QSO Rateグラフの過去ログとの重ね書き表示機能が欲しい

具体的なデザインとかは未検討

jr8ppg, 2021-05-02:
◎棒グラフをバンド毎の積み上げにしてほしい(バンド毎は色別)
〇JARL4大の開始終了時刻が固いものは、21時スタートで左端から表示開始にし
てほしい。
〇過去のデータをターゲットファイルとして、毎時棒グラフを2本たてて比較し
ながらすすめたい。

JO3JYE, 2021-06-07:
rate画面を見るの楽しみが増えますね。
ロカコンでマルチバンド（28-1200MHzの5バンド)参加したログを読み込ませてみたところ。
・マウスのホイールで、グラフが上下に動きます。
・ＱＳＯ数の棒グラフが上にはみ出してしまうケースあります。目盛の自動調整の問題だと思います。
この２点が気になりました。

jr8ppg, 2021-06-07:
おっと気が付きましたか。

> ・ＱＳＯ数の棒グラフが上にはみ出してしまうケースあります。目盛の自動調整の問題だと思います。

こちらはコンテスト中に気になったので直しました。最新コミット見てみて下さい。https://github.com/jr8ppg/zLog/commit/5d0a26f43a4d060e22dc56be4cd6cbf4f6730202

ホイールはやってみます。

JO3JYE, 2021-06-07:
単に、ちょいと古いコードを使ってしまったようです。
棒グラフのはみ出しは、最新のコードでビルドしたらokでした。

ホイールは、グラフが上下に動きます。

JO3JYE, 2021-06-09:
ホイールでの移動、しなくなりました。ありがとうございました。

jr8ppg, 2021-06-11:
積み重ねはいったん終了。
過去ログの重ね書きは別Issueにします。

closed 2021-06-11 by jr8ppg