jr8ppg/zLog #72: AMDインターフェースの検証

jr8ppg, 2020-11-06:
AMDインターフェースの件の検証
http://jn2amd.html.xdomain.jp/usbrigcwsound40.htm

双方共にKEY:DTR, PTT:RTSなのにキーイングできないとは？

誰か持っていませんか？

7M4KSC, 2020-11-07:
[PCWI01](https://www.jh4vaj.com/pcwi01_01)のDTR->Keyは動作しているので、AMDインターフェース ver.4でも、動作するハズですが、どうも頒布されていない？（「配布」のページに、ver.4の頒布情報が掲載されていません）
もうちょっと、回路を見てみます。

jr8ppg, 2020-11-07:
調査ありがとうございます。RI, DCD, DSRは使ってないですね。
理屈上は動く物と考えています。
もし関係があるとすればDSRかと思いますが、データを送らないので関係なさそうです。
前にAMDさんに聞いてもらったことがあるのですが、動作テストした状況とかが不明でした。
現物で追試するしか無いと思っています。

7M4KSC, 2020-11-07:
AMD-USB-RIG-CW-SOUND Ver4.0の頒布に関して、AMD OMにメールを出しました。

7M4KSC, 2020-12-11:
ＡＭＤ－ＲＳ２３２Ｃ－ＳＯＵＮＤ　Ｖｅｒ１．０では、zLogで(RISC CPU USBでは無く)DTRでキーイングされているので、どうも最近のAMD OMのインターフェースでは、本当にキーイング出来ないみたいです。ちょっと本気で、入手してみますかね。

jr8ppg, 2020-12-11:
？？
なんで駄目なんですかね。不思議です。

7M4KSC, 2020-12-23:
ちょっと遅くなりましたが、ＡＭＤ－ＵＳＢ－ＲＩＧ－ＣＷ－ＳＯＵＮＤ　Ｖｅｒ３．５の完成品
を注文しました。（各IFの違いがよくわかって無くて、発注遅延～）

7M4KSC, 2020-12-28:
上記、AMD-USB-RIG-CW-SOUND Ver3.5で、CWのキーイングが出来る事を2.6 beta1で確認しました。

7M4KSC, 2020-12-28:
AMD OMに動作した事を報告済みなので、closeでも良いかと思います。

jr8ppg, 2020-12-28:
了解しました。ありがとうございます。

closed 2020-12-28 by jr8ppg