jr8ppg/zLog #74: WinKeyerの対応

jr8ppg, 2020-11-06:
WinKeyerの対応
https://hamcrafters2.com/WKUSBX.html

ニーズはあるか？

JO3JYE, 2020-11-07:
追加を希望します。
手元に実機もあります。ただ、所有しているものは古いバージョンのファームかもしれません。
zlogでは使えないので埃をかぶってしまいました。hi

jr8ppg, 2020-11-07:
WinKeyer3 IC (V3.1)にUpdateが必要かもしれませんね。

JO3JYE, 2020-11-07:
埃をかぶっている状態なので、まずは、押し入れからひっぱり出して、動作確認から始めます。
ちゃんと動かすまでに、時間かかるかもしれません。

7M4KSC, 2020-11-07:
マニュアルのリンクは、WinKeyerUSBの調査 #56に有り。

jr8ppg, 2020-11-07:
マニュアル確認しました。うーん、どうしたら良いですかね～
https://hamcrafters2.com/files/WKUSB_Manual_v1.3.pdf

7M4KSC, 2020-11-07:
サポートは、かなりシンドいと思います。ファンクションキーF1のCQ TEST $M TESTのASCII文字列をそのまま、WinKeyにxon/xoff制御付きで送る感じです。USBIF4CWと同じクラス扱いにして、仮想関数を全てオーバーライドする感じ？（すみませんソース見ないで言ってます）

JO3JYE, 2020-11-08:
手元にある実機、やはりファームは古かったです。K1ELのサイトから最新版を取得しアップデートしました。
マニュアル見ていますが、これはなかなかハードルが高いですね。

jr8ppg, 2020-11-08:
夕べマニュアルを見てわかったこと
・USBはHIDではなくてCOMポートとして認識される
・普通に送信したい文字列を送れば良い
・１６０バイトバッファがあるのでバッファ制御不要（USBIF4CWは８バイトなので何かしないといけない）
・スピード変更や中断のコマンドはどれ？　マニュアル最後の方にexampleがあった
・zLogの制御コマンドをWinKeyer用に置換する処理

意外と簡単かもしれないです。
WinKeyer用のzLogSendStr()を作って分岐すれば良いかな。
中身はZComKeying.SendString('CQ TEST DE JR8...')で良さそう。

jj1cvh, 2020-11-09:
ご検討いただきありがうございます。私自身はWinkeyerを使う機会はゼロですが、先進的なマルチオペ局はきっとN1MM+に移行していて、zLogがWINKEYに対応すればzLogに戻ってくるマルチオペ局もいると想像してWinkeyer対応をリクエストさせていただいていました。

Winkeyer IC データシートのリンクです。
Winkeyer1 https://www.k1elsystems.com/files/Winkey10.pdf
Winkeyer2 https://www.k1elsystems.com/files/WK2_Datasheet_v23.pdf
WinKeyer3 https://www.k1elsystems.com/files/WK3_Datasheet_v1.3.pdf

Arduinoを使ったキーヤはK3NG Winkey Emulationを使ってCTESTWINやN1MM+で動作していると思いますが、Winkey 1 modeがデフォルトなので、Winkeyer1(1200baud)のコマンドに対応するのが無難な選択かなと思います。
https://github.com/k3ng/k3ng_cw_keyer/wiki/370-Feature:-Winkey

Winkeyer1データシートによると、スピード変更は以下ですね。5-99のWPMを16進に変換すればいいみたいです。
```
Set WPM Speed <02><nn>  nn is in the range of 5-99 WPM
                        Example: <02><12> set 18WPM
Set a new Morse operating speed, this command takes effect as soon as WinKey receives it. If speed is set 
to zero then WinKey will take it’s speed setting directly from the speed pot., this is the reset default.
```

JO3JYE, 2020-11-09:
手元のwinkey相手に試しました。
お察しの通り、pcからは単なるCOMとして見えています。シリアルはFTDIのチップです。
起動するとスタンドアローン状態、起動用の電文をPCから投げてオンラインにしないとpcからのコントロールできません。
ただ、jj1cvh局ご指摘の通り、起動に必要な電文を投げる作法があるみたいで、巷にあるバイナリが通るターミナルを使って手動でコマンドを投げるという手だと私のスキルではオンライン状態が作れませんでした。

オンライン状態が作れれば、あとは既存のオブジェクトの流用・手直しで実装そうな気がします。(気がするだけですのであしからず)

jr8ppg, 2020-12-13:
１台借りてきました。と言うか押し貸し・・・

jr8ppg, 2020-12-20:
試作レベルだけどできたところ
・F1キーでCQ出す
・TABキーで応答、しかし送出電文が変
・下キーで確定
・SPEED DIALも動く

次のTODO
・SPEED DIALに応じたUI更新
・特殊コマンドのWinKeyer対応
・ショートカットキー（CTRL+Y/U）対応
・OptionsウインドウへのUI追加

全体的にCW送出APIの仕様を整理した方が良い。
困ったな・・・

jr8ppg, 2020-12-20:
winkeyerブランチです
https://github.com/jr8ppg/zLog/tree/winkeyer

jr8ppg, 2020-12-22:
今日できたところ
・SPEED DIALに応じたUI更新
・ショートカットキー（CTRL+Y/U）対応

次のTODO
・特殊コマンドのWinKeyer対応
・OptionsウインドウへのUI追加

あわせてリファクタリングして整理することにする

jr8ppg, 2020-12-29:
今日できたところ
・OptionsウインドウへのUI追加

次のTODO
・特殊コマンドのWinKeyer対応
・リピート処理

jr8ppg, 2020-12-30:
今日できたところ
・特殊コマンドのWinKeyer対応，2番メッセージ対応

次のTODO
・リピート処理

jr8ppg, 2020-12-31:
今日できたところ
・リピート処理

次のTODO
・もう少し整理したい
・もう少しいじってみる
・JYEさんいじってみて

JO3JYE, 2020-12-31:
なんだか呼ばれたみたいです(^^;;;。

先ほど（12/31のPM９時頃）GETした最新のコードでビルドしたところ、Winkeyでのキーイングができませんでした。
つないだパドルからの送信はできます。winkeyのトーンの雰囲気からは、初期化がおかしいと直感？
でも、VRを回すと、速度が変わっていくところはzlogで追従できている。（ということは初期化はokなの？）

今朝(12/31のAM7時過ぎ)にGETしたコードでビルドした場合。
・普通にQSOする分には正常（TABでナンバーを送って、↓でQSO確定、F3メッセージ送信)
・ナンバーフィールドにフォーカスがあるときは、TABキーを押しても相手コールサインのみ送信し、NRが送れない。
　(コールサインフィールドの場合は正常です)
・DUPE時の動作、コールサインの後のメッセージ送信がない。(AlowToDupeがONでもOFFも変わらずだったかな）
・リピート利かない（でも、それはそうですよね。対応前のコードだから）
・速度が勝手に変わることがある。どのタイミング？というのはまだ見極めきれず。

こちらの実機では、PCからのコントロールの有無で内蔵モニター音の音程が違います。PPGさんところの個体はいかがでしょうか？

jr8ppg, 2020-12-31:
こんばんは。年末の忙しいところありがとうございます。
はい、呼んでますよ（笑

最新ビルドではサイドトーンの有無のコマンド送るのを追加しているので、
Use Side ToneをONにすると音がでます。OFFにすると何も起こってないように見えます。
違いと言えば朝のバージョンとの違いはそれ位です。
LEDでも付いているとありがたいのですが。
最終的には通常使うときサイドトーン要らないですよね。

あと、Switch TAB/; with CW bankをONにすると動きがおかしくなると思います。
どう考えてもバグと思うのですが、もう少し検証が必要です。ここはOFFで使って下さい。

> ・ナンバーフィールドにフォーカスがあるときは、TABキーを押しても相手コールサインのみ送信し、NRが送れない。
> 　(コールサインフィールドの場合は正常です)
> 
このパターンは気がつきませんでした。やってみます。

> ・DUPE時の動作、コールサインの後のメッセージ送信がない。(AlowToDupeがONでもOFFも変わらずだったかな）

これは大丈夫だったような気がするけど、、、
デバッガ上で実行すると、TABキー押して相手コールが送出された後の処理が行われないケースはありました。原因不明。
zlog.exe単独で実行すると問題ないようです。

> こちらの実機では、PCからのコントロールの有無で内蔵モニター音の音程が違います。PPGさんところの個体はいかがでしょうか？

おっと、パドルはまだ繋いでいません。
正月休みの間でやってみようと思います。

JO3JYE, 2020-12-31:
最新ビルドの件。
winkeyのサイドトーンで確認していたので、OFF設定だったら判りませんね。
ということで、動いているけど、動いていないと早合点している可能性、かなり高いです。
FT-817でもつないでモニターしみてます。失礼しました。

DUPEの件、もうちょっと試してみます。
こちらでのビルドはwin32、デバッグではなくリリースにしています。

jr8ppg, 2020-12-31:
Aboutウインドウにリビジョン表示を追加しました。
手元のWinKeyer USBは A0 でした。
どう言うバージョンなのかな。ハムフェア2009のシールが貼っているのでその頃のバージョンとは思います。

jr8ppg, 2020-12-31:
今見るとリビジョン 17 でした、、、

JO3JYE, 2020-12-31:
こちらのは　C５　です。

jr8ppg, 2020-12-31:
そちらは最新ですよね。まずはそこからか、、、

jr8ppg, 2020-12-31:
リビジョン0x17(=23 で2.3)なのでWinKeyer IC 2.3と言う事がわかりました。
これってWinKeyer IC 3.0にできるのですかね。（普通無理ですね）

JYEさんのC5ってなんだろう。
CなのでSTATUSっぽいな、、、

うーん、どうするかな。

JO3JYE, 2020-12-31:
もう一回見てみたら　CF　になっています。
純正の　Winkyer3　Tools　Utility　で見ると　F/W　ver　31.02　です。
K1ELのサイトに最新のファームがあるはずです。

jr8ppg, 2020-12-31:
Wk3Toolで見ると、WK3 onlyと出ました。やはり手元のはWK2で古いですね。
https://hamcrafters2.com/WK3tools.html

新しいの買えや～って言ってみます。

jr8ppg, 2021-01-01:
WinKeyerにパドルを繋いでみました。
トーン音はPCキーイングと同じ音でした。

JO3JYE, 2021-01-03:
久しぶりの常置場所。
昨日版のcodeでビルドして、TS-590SGとつないでみました。
例の動かないかも？問題は、単なるモニターOFFになっていることに気が付いていないだけでした。失礼しました（^^;;;

DUPE時の動作。AlooDupeOnの有無で異なる感じです。
ONのとき、コールサインのみの送出で止まります。
OFFの時は、Ｆ４の中身を返していると思われる動作です。

TABキーの動作。フォーカスがあるフィールドで違う件。昨日のcodeでビルドしたところ再現は確認できずです。

　WinkeyからのBEEPオンモニターですが。今のところ,pcからの時は消音、パドル操作時はbeep音のモニターありになっています。モニター音設定のコマンドが違うのでしょうか？

明日。時間があれば。HGコンテストで実戦テストを少しやってみます。

jr8ppg, 2021-01-03:
こんばんは。
確認しました。

> DUPE時の動作。AlooDupeOnの有無で異なる感じです。
> ONのとき、コールサインのみの送出で止まります。
> OFFの時は、Ｆ４の中身を返していると思われる動作です。

こちらは私の方で勘違いをしていまして別のパターンを確認していました。
ALLOW DUPE ONで、DUPEの時の処理を追加しました。

> TABキーの動作。フォーカスがあるフィールドで違う件。昨日のcodeでビルドしたところ再現は確認できずです。

こちらは修正済みなのですが、元の動作の意図が不明なのでとりあえず下記の様にしました。
問題は無いと思いますが、、、

従来
・CALLSIGNにフォーカスがある時：DUPEチェックをして2番か4番を送出
・CALLSIGN以外にフォーカスがある時：2番を送出　←何か意味があるのか無いのか？

修正後
・CALLSIGNにフォーカスがある時：DUPEチェックをして2番か4番を送出
・CALLSIGN以外にフォーカスがある時：DUPEチェックをして2番か4番を送出

> 　WinkeyからのBEEPオンモニターですが。今のところ,pcからの時は消音、パドル操作時はbeep音のモニターありになっています。モニター音設定のコマンドが違うのでしょうか？

コマンドを再度確認すると、"パドルのみ"か"両方"かしか設定できません。（Page7 Sidetone Control参照）
あとできそうなのはSidetone Volumeをいじる位ですが、1～4の設定なのでOFFにはならないかもしれません。

Page9 Set PinConfig でもできそうな気がしますが、ICのCFGになるのでいじりたくないですね。
やるならWk3Toolsの方が無難。

参照マニュアル
https://hamcrafters2.com/files/WK3_Datasheet_v1.3.pdf

jr8ppg, 2021-01-03:
ツールバーのサイドトーンON/OFFボタンも対応しました。

jr8ppg, 2021-01-03:
そうだ、PTT未確認なんですが、動いていますかね、、、

JO3JYE, 2021-01-03:
すみません、PTTは未テストです。
運用がセミあるいはフルブレークイン派なので、テストするには接続ケーブルの工作が必要です。今も自宅じゃないのでパーツのストックが少ないから材料あるかな？

AlowToDupeの件、対応ありがとうございます。

さて、新しいコードでビルドしなおします。

JO3JYE, 2021-01-04:
対応作業ありがとうございます。
ちょうど、常置場所(JCC2713)にいたので兵庫コンテストで実戦投入してみました。(7MHzCWで2時間ひたすらＲＵＮのみ。約160QSOS）
1/4の9時のコンテスト開始前にcodeを取得してビルド、PTTも部材が足りたので接続ケーブルを作成。

○動作確認できました。
・PTT機能
・DUPE時の動作もok。

○要修正と感じたこと。
・TABキーにてコールサイン→ＮＲ送信している途中で、ESCキーを押しても完全にキャンセルできない
　(コールサイン送出中のESCキーだと、コールサインが止まって、NRが送出される。もう一回ESCでNR送信もキャンセルできる）
・コールサイン送出中にコールサインを修正しても修正が反映しない。
　例、JR8PPNと入力して最後までコールサインを送出する前に、JR8PPGと直してもJR8PPNと送信する。

○？なところ。再確認あるいは、ほかの人にもテストをお願いしたいところ。
・パドル入力がワンテンポ遅れる。取りこぼす（これはパドルの接触に問題がある可能性も否定しません)
　あるいはマシンパワーの問題？？Core2Duo(E8400)/MEM4GB/Win7pro(32bit)。CドライブはＳＳＤという環境。
　裏で、WindowsUPDATEが動いてしまったからかもしれません。
・なぜか、「？」が送信できないことがあった。
・勝手に速度が25wpm→22wpmとなったことが数回あった。(手動で25へ戻す)
・cw送信中にフォーカスを動かし、QSO編集画面を開くと一旦停止することがある

　気になる点をメモしながらRUNしていました。フル参加できればよかったですが、午後イチで大阪の自宅へ戻ってきました。

jr8ppg, 2021-01-04:
細かいところまでありがとうございます。

> ・TABキーにてコールサイン→ＮＲ送信している途中で、ESCキーを押しても完全にキャンセルできない
> 　(コールサイン送出中のESCキーだと、コールサインが止まって、NRが送出される。もう一回ESCでNR送信もキャンセルできる）
> 

こちら了解しました。修正します。

> ・コールサイン送出中にコールサインを修正しても修正が反映しない。
> 　例、JR8PPNと入力して最後までコールサインを送出する前に、JR8PPGと直してもJR8PPNと送信する。
> 

こちら困りましたね。
WinKeyerに全部送ってしまってるので、そうなっちゃうよな、、、です。
Page14のPointer Commandに callsign correctionって書いてあるのでこのコマンド使ってなんかやれって事みたいです。
ちょっと考えます。

> ・パドル入力がワンテンポ遅れる。取りこぼす（これはパドルの接触に問題がある可能性も否定しません)
> 　あるいはマシンパワーの問題？？Core2Duo(E8400)/MEM4GB/Win7pro(32bit)。CドライブはＳＳＤという環境。
> 　裏で、WindowsUPDATEが動いてしまったからかもしれません。
> 

パドル処理はzLog関係なくてWinKeyer内での処理になります。
改めてやってみた所、手元のWK2でも同じですね。
20WPM位でも立ち上がり一瞬遅れるのでとても使いにくいです。
zLogを起動しなくても同様でした。何かパラメータ設定で調整できるのかもしれません。
詳しい方いませんかね。

> ・なぜか、「？」が送信できないことがあった。

こちらは様子見かな。
関連して、zLogでは"."を"?"の代用にできるのですが、その対応が漏れています。

> ・勝手に速度が25wpm→22wpmとなったことが数回あった。(手動で25へ戻す)
> 

WinKeyerのダイヤルを回さなくても変わったと言う事ですかね。
ダイヤルのvolumeが安定しないのかな。
他にzLogにはFixedSpeedと言う機能があって、"!"か"-"で決められたSpeedにダイレクトに変更できます。
初期値は20wpm。この操作はどうですか？

> ・cw送信中にフォーカスを動かし、QSO編集画面を開くと一旦停止することがある

CQ中にQSO編集画面を出してみましたがそんな感じはしませんでした。
様子見でお願いします。

jr8ppg, 2021-01-04:
マニュアルのPage14

> A detailed description of the pointer command is detailed in a separate application note.

にある、a separate application note.ってどこにあるのだろう、、、

jr8ppg, 2021-01-04:
パドルの立ち上がりが悪い件判明。
before delay timeが効いていますので、0にすると良くなります。

JO3JYE, 2021-01-05:
いろいろと対応ありがとうございます。

○TABキー操作をESCキー　一回でキャンセルする動作
動作を確認しました。
ただ、以下の事象が出ます。ちょっと意地悪ですけど。

１．tab-キーでコールサインの送出が完了し、ナンバーを送っているときにESCキーでキャンセル
２．フォーカス位置は不問で、もう一度TABキーを押す。
３．コールサインの送出は行うけど、ナンバーが送出されない。
４．もう一回TABキーを押す、ここは正常動作。

この現象は続かず、一回だけ発生すると思われます。

○勝手に速度が変わる件、

optionを開いて、okをクリックしたら、速度はwkのVRの位置へ吸い込まれる。(当然そうなりますよね)
当日は、QSOしながらoption画面も開くということをしていたので、okをクリックしたことでｗｋのvrの位置へ吸い込まれたのではないかと思います。逆に、VRが早い位置でzlog側で速度を落とした場合は、option開いてokすると、速度が上がります。
wk本体のvrでも速度が変わり、zlog上でも速度が変わる、両立が難しい。なんだか悩ましい。

○コールサイン送出中のコールサイン修正に追従する件
comポートやusbif4cwではokですので、なんとかなると非常にうれしい限りです。

○パドル立ち上がり悪い件

before delay time、どこにありますか？ドキュメント内で探しているが見つからず。。。
私のwk3だと単体では違和感なく、zlogからコントロールするとワンテンポ遅れるかんじ。

wkのドキュメントやwk_tool、wk_demo。ファームウェアバージョンとソフトのバージョン、何か相関あるかも？
最新ファームにしてから、wk_demoが使えない状態になっています。

もうちょっとこちらでも悩んでみます。
場合によっては押し貸しの第二弾でしょうか？(笑)

jr8ppg, 2021-01-06:
おはようございます。
取り急ぎ２点ほど。

> ○勝手に速度が変わる件、

お気づきの通りでWinKeyerのVRを正とするしかないですね。

> ○パドル立ち上がり悪い件

zLogのOptionsウインドウ、Hardwareタブで、下の方にPTT Delay time beforeとafterがあります。
たぶんbeforeが初期値で10か20になっています。

まだ押し貸しは不要な段階です。Hi

JO3JYE, 2021-01-06:
beforeはZLOGの設定でしたか・・失礼しました。
こちらの設定は25になっていましたので、０としたところズレは解消しました。
この設定、EnablePTTControlのチェックに連動して設定なのですが、チェックが外れても設定値が反映しているように思いますが、いかがでしょう？

押し貸しはまだ不要ということで了解です。ｈｉ

jr8ppg, 2021-01-06:
確かにPTT Delayの設定間違っています。OFFの時はbefore,after共に0にするようにしました。

JO3JYE, 2021-01-06:
PTT Dealyの修正ありがとうございました。
動作確認もできました。

jr8ppg, 2021-01-06:
早速ありがとうございます。残るはコールサインの訂正ですね。
winkeyerのwebサイト見て回ってもドキュメント見当たらないのです。
困りましたな。

jr8ppg, 2021-01-09:
backspaceコマンドをやっと見つけた。これを実装すればコールサインの訂正はＯＫかな。
Page9
● Backspace <08> Backup the input buffer pointer by one character.
This command is only meaningful if there is something in the serial input buffer, otherwise it is ignored.

JO3JYE, 2021-01-09:
backspace、おーーーそういう手がありますね。あっ、リファレンスに書いてあるわ(^^;;;。

K1ELのサポートページの投稿を検索したら、pointerコマンドのドキュメントはどこにある？とか、使い方は？といった質問が数件見られます。なんだけど、全部スルーされて回答がありません。いったい、どうなっているんでしょうね？

jr8ppg, 2021-01-09:
いやー、仕方ないからK1ELにメール書こうと思ってマニュアル見直していたら発見しました。

jr8ppg, 2021-01-09:
TABキー押下後、Number欄にフォーカスが移動するまでの間、Backspaceキーが使用できます。

jr8ppg, 2021-01-09:
PTT制御はこれでいいのか不明。

jr8ppg, 2021-01-11:
今更ながら見てきました。確かに全部スルーされてますね。書き込んだら個別にメール来るのかな。
https://groups.io/g/k1elsystems/search?q=pointer+command

N1MM用？のundocumentedなパラメータがあるみたいです。
https://groups.io/g/k1elsystems/message/4707?p=,,,20,0,0,0::Created,,pointer+command,20,2,0,20342215

JO3JYE, 2021-01-11:
対応作業お疲れ様です。
Backspaceキーでの修正、確認できました。
Backspace以外のキーで修正した場合も含めていただけると助かります。今だと、JR8PPDと入力した後、カーソルキーとdelキーで、JR8PPGと修正した場合は、JR8PPDGと符号が出ます。この点は、backspaceでＯＫだということがないとコーディングしにくい部分だと思いますが。よろしくお願いします。
Backspaceは連打しても追い付いているので、大丈夫かなと思います。

　Pointerコマンドの件はどうでしょう？？mailしてもスルーだったら、データシートには書いているけど、pointerコマンドは実装されていない可能性もありませんか？
時々、pointerコマンドの質問があるけど、ブラックホールに吸い込まれるかのようにみんなスルーされていますね。hi

JO3JYE, 2021-01-11:
N1MM用の非公式コマンドの件。
独自に、通信をキャプチャーしたらドキュメントにないコマンドが流れている。。という突っ込みもスルーされていますね。
なにを見つけたのか？すごく気になります。hi

jr8ppg, 2021-01-11:
Backspaceコマンドを見つけたのでまだメールはしていないです。
末尾の修正に限るならdelキーも対応できますが、そう言うことでは無いですよね。
サフィックスの真ん中を修正位は想定できますね。

pointerコマンドの説明は動作モードが３つあるしか書いていなくて、書き込み位置をどこにするとかいった肝心な部分がundocumentedなパラメーターなんだと思っています。
メールしてみますか。

jr8ppg, 2021-01-11:
メール出してみました。moderate制なので時間掛かりそうです。

JO3JYE, 2021-01-11:
カーソルキー＋del　が想定です。
例えば、JO3JYE　を　JO3AYE　と変えるのなら　Backspace　3回　のあと　AYE　を送り直す、という手順はいかがでしょう？修正された文字から後ろを全部切り捨てて、送り直すという発想です。
これなら、backspaceだけで実装できそう。ただ、力技で、美しくない、めんどうですが(^^;;

jr8ppg, 2021-01-11:
その手法も考えたのですが、Backspaceコマンドを送った後、再度コールサインを送るタイミングがわからないのです。
あと、左右のカーソルキー操作を左左右とかやられると、もう何がなんだかわかりませんね。
---
JR8OPG   画面上：JR8OPG 、WK：JR8OPG
←キー3回　画面上：JR8PPG 、WK：JR8
DELキー   画面上：JR8PG 、WK：JR8
訂正  Pを入力 画面上：JR8PPG、WK：JR8P
残りをいつ送る？

JO3JYE, 2021-01-11:
wkに送った文字列と修正後の文字列を比較して、文字が入れ替わった最初の場所から後ろを差し替える、というロジックで考えてみましたが。。
どの操作をもって修正終わりとするか？という問題がご指摘の通り残ります。

一文字ずつ送るのもの違いますし。
やはりpointerコマンドしかないのでしょうか。

jr8ppg, 2021-01-11:
> どの操作をもって修正終わりとするか？という問題がご指摘の通り残ります。

そうなんです。この問題が残るのです。

> 一文字ずつ送るのもの違いますし。

それ良いかもしれません。コールサインは１文字ずつ送る。
WKからはキーイング完了のお知らせが来るので、そのタイミングで次の文字を送る。
これならBackspaceコマンドもいりませんし、好きに修正できます。
ただ、やってみないとわかりませんが、若干間が空いて不自然になる可能性があります。（1200bpsなんで）
やる価値ありそうなのでやってみます。

jr8ppg, 2021-01-11:
一文字ずつの送信にしてみました。
目論見通りの動きになりましたが、やはり若干間が空いて丁寧に１文字ずつ打っている様な感じです。
https://github.com/jr8ppg/zLog/tree/winkeyer2

WK3は1200bpsから変更できるので19200bps位にすれば自然な感じで打てそうです。

jr8ppg, 2021-01-11:
winkeyer2ブランチです

jr8ppg, 2021-01-11:
WK3でも9600bpsにしか変更できない模様。（データシートのpage18）

[20 more replies]

closed 2021-02-19 by jr8ppg