jr8ppg/zLog #76: アルインコの無線機のCAT調査

7M4KSC, 2020-11-06:
アルインコの無線機のCAT調査を行う。

7M4KSC, 2020-11-06:
・DX-SR9 J/Mのクローン端子で、PC制御が出来そうであるが、ドキュメント内にコマンド表等は存在せず。
・PC制御ソフトは、[存在する](https://www.nick-bailey.co.uk/driver8/index.html)。
・上記PC制御ソフトは、DX-SR8もサポートするので、DX-SR8もPC制御出来そうである。
・DX-SR8のコマンドも、現状では判明せず。（ドキュメントが存在しない。どうやってPC制御ソフトを作った？）

jj1cvh, 2020-11-06:
どうやって解析したかはさておき、DX-SR8のコマンド表。
https://forums.qrz.com/index.php?threads/alinco-dx-sr8t-ans-omni-rigv2-or-v1.695478/
https://yo5ptd.wordpress.com/2017/02/12/alinco-dx-sr8/

hamlibではYO5PTDのコマンド表を参照して2020年に対応したようです。
https://www.w8wjb.com/wp/2020/06/14/beta-support-for-alinco-dx-sr8t-in-hamlib/

fldigiのリグ定義ファイルは2015年ですね。
https://sourceforge.net/projects/fldigi/files/xmls/alinco/

closed 2020-11-07 by 7M4KSC