jr8ppg/zLog #78: ユーザー定義コンテストでの局種係数の取り扱い

jr8ppg, 2020-11-08:
ＦＤ以外で局種係数を使っているコンテストがあり、E-LOGに出力されない。（想定外の使い方）

修正案
・ユーザー定義コンテストで局種係数を明示的に入力した場合は
   <FDCOEFF>タグを出力するようにする。(R1.0,R2.0ともに)
・FDコンテストモードで起動時に入力した局種係数(1 or 2)を、
　過去の"保存"したデータにかかわらずこの画面に正しく反映さ
　せる。もしくは、"保存"データとしない。

などなど

jr8ppg, 2020-12-08:
#81 の対応じゃだめかな

jj1cvh, 2020-12-10:
E-Log (JARL 1.0)画面とE-Log (JARL 2.0)画面で局種係数を入力すると、emファイルの<TOTALSCORE>に反映されることを確認しました。

ただ、ユーザ定義コンテスト選択時に、E-Log (JARL 1.0)の場合は<FDCOEFF>がemファイルに出力される一方、E-Log (JARL 2.0)の場合は出力されませんでした。
E-Log (JARL 2.0)の場合も、入力した局種係数を<FDCOEFF>タグで出力するようにしていただけませんでしょうか？

それからField Day選択時に、コンテスト選択画面のScore Coeff.に入力した係数が、E-Log (JARL 1.0)画面とE-Log (JARL 2.0)画面の局種係数欄に反映されるわけではないのですね。
E-Log (JARL 1.0)画面は常に2、E-Log (JARL 2.0)画面は常に1になりました。
コンテスト選択画面のScore Coeff.はどのように機能しているのでしょうか？

jr8ppg, 2020-12-10:
やっぱり入り口側も直さないとだめですね。
出口側の対応は #81 を基本とすればいいかな～と考えています。

jr8ppg, 2020-12-11:
これでOKかな

jj1cvh, 2020-12-12:
コンパイル時に以下の警告が出ますが、OKでしょうか？
- [dcc32 警告] UELogJarl2.pas(306): W1036 変数 'nFdCoeff' が初期化されていない可能性があります

以下2点の改善をお願いします。
- どのコンテストを選んでも、E-Log (JARL 2.0)で出力したemファイルで\<TOTALSCORE>0\</TOTALSCORE>となる。
- E-Log (JARL 1.0)の局種係数欄に小数(1.5など)を入力できない
(京都コンテストの例) ニューカマーマルチプライヤー: x2.5, x1.5, x1.2, x1.0

以下は正常動作を確認しました。ありがとうございます。
- ユーザ定義コンテスト選択時に、E-Log (JARL 1.0)もE-Log (JARL 2.0)も局種係数欄に入力すると、\<FDCOEFF>がemファイルに出力される
- Field Day選択時に、コンテスト選択画面のScore Coeff.に入力した係数が、E-Log (JARL 1.0)画面とE-Log (JARL 2.0)画面の局種係数欄に反映される

jr8ppg, 2020-12-12:
係数1.2等のコンテストあるのですね。
規約を見たところ端数処理は切り上げとのことなので、とりあえず切り上げにします。

jj1cvh, 2020-12-12:
総得点の端数処理ですが、たまたま見つけたのは京都コンテストでしたが、他のコンテストは調べられていないので、どうあるべきかは微妙です。
JA1ZLOのリアルタイムコンテスト第1回規約案は、多分小数のままですね。

総得点は小数のままにしておき、必要に応じてemファイルをテキストエディタでユーザが手修正する方が良いような気もしますが、いかがでしょう。

ちなみにZPRINTWも、総得点は小数のままでした。
`<SCORE BAND=50MHz>1,1,1</SCORE>`
`<SCORE BAND=TOTAL>1,1,1</SCORE>`
`<FDCOEFF>1.5</FDCOEFF>`
`<TOTALSCORE>1.5</TOTALSCORE>`

jr8ppg, 2020-12-12:
では端数処理無しで。
取り急ぎ直しました。続きは来週。

jj1cvh, 2020-12-12:
ご対応ありがとうございます。以下2点の正常動作を確認しました。
- E-Log (JARL 2.0)で出力したemファイルの\<TOTALSCORE>で総得点が正しく記録される
- E-Log (JARL 1.0)とE-Log (JARL 2.0)の局種係数欄に小数(1.5など)を入力し、正しくemファイルが出力される

zlog.iniの[SummaryInfo]のFDCoefficientは、今回の改修により使われなくなったと理解して良いでしょうか？
であれば、E-Log (JARL 1.0)とE-Log (JARL 2.0)画面の[保存]を押した時、FDCoefficientを記録する必要はないかもしれませんね。

#78 に関して、他に気付いた不具合はありません。

jr8ppg, 2020-12-13:
そうですね。zlog.iniの記録は削除しました。

jj1cvh, 2020-12-13:
2d2deda にてzlog.iniのFDCoefficientの記録が削除されていることを確認しました。
また、書き忘れましたが、その前の 4bfcd01 で「[dcc32 警告] ... 'nFdCoeff' が初期化されていない...」が発生しなくなったことも確認しました。

jr8ppg, 2020-12-30:
これもOKかな

closed 2020-12-30 by jr8ppg