jr8ppg/zLog #83: Allow to log dupesがオフの場合の強制確定時の点数

jr8ppg, 2020-11-12:
Allow to log dupesがオフの場合、DUPE局をSHIFT+ENTERで強制確定すると、DUPEなのに点数が付いてしまう不具合
zloファイル再ロードまたはQSO編集で正常化する

Allow to log dupesがオンの場合のDUPW局は０点で記録され正常

jr8ppg, 2020-11-13:
TQSO.Reserve2の$FFをクリアしなければ良い話なのだが、どう言う影響があるか、、、

jj1cvh, 2020-11-14:
#83 は、dupeかつSHIFT+Enter時にallowdupeと同様の処理をすれば良いはずと思いまして、修正案を作ってみました。#80 は考慮していません。
```
--- zlog/main.pas	2020-11-13 23:46:17.509367800 +0900
+++ zlog-after/main.pas	2020-11-14 09:23:56.000000000 +0900
@@ -5125,7 +5125,7 @@
    EditedSinceTABPressed := tabstate_normal;
 
    _dupe := Log.IsDupe(CurrentQSO);
-   if (_dupe = 0) or (CurrentQSO.Reserve2 = $FF) then // $FF when forcing to log
+   if (_dupe = 0) then
    begin
       if (MyContest.MultiForm.ValidMulti(CurrentQSO) = False) and (CurrentQSO.Reserve2 <> $FF) then begin
          WriteStatusLine('Invalid Number', False);
@@ -5138,12 +5138,13 @@
          CallsignEdit.SetFocus;
          exit;
       end;
-      if CurrentQSO.Reserve2 = $FF then begin
+
+   med:
+      if CurrentQSO.Reserve2 = $FF then begin // $FF when forcing to log
          CurrentQSO.Reserve2 := $00; { set it back }
          CurrentQSO.Memo := '* ' + CurrentQSO.Memo;
       end;
 
-   med:
       MyContest.SetNrSent(CurrentQSO);
 
       repeat
@@ -5284,7 +5285,7 @@
       end;
    end
    else begin
-      if dmZLogGlobal.Settings._allowdupe = True then begin
+      if (dmZLogGlobal.Settings._allowdupe = True) or (CurrentQSO.Reserve2 = $FF) then begin // $FF when forcing to log
          CurrentQSO.Dupe := True;
          CurrentQSO.Points := 0;
          CurrentQSO.NewMulti1 := False;
```

jr8ppg, 2020-11-14:
検討ありがとうございます。
概ね良いと思うのですが、下記の行があるのでforce to logの時も画面にDUPEと出てしまうのです。
ここをどうするか考え中です。
CurrentQSO.Dupe := True;

jr8ppg, 2020-11-14:
再ロード時にTLog.SetDupeFlags;内で再度DUPEチェックが行われて強制入力した物が生き返ってしまいますね。

jr8ppg, 2020-11-15:
修正案作成
https://github.com/jr8ppg/zLog/commit/6822c52e923fc75305124c7d811bf4e6d2e8a7e6

jj1cvh, 2020-11-16:
「Allow to log dupesがオフの場合、DUPE局をSHIFT+ENTERで強制確定すると、DUPEなのに点数が付いてしまう不具合」について、shift_enter_plan 6822c52 で改善していることを確認しました。ありがとうございます。

#83とは直接関係しませんが、その他のDUPE QSOの挙動を調べてみたところ、軽微な不具合が2点ありました。
・PhoneのDupe時にTABを押してもStatusLineに何も表示されない
・Allow to log dupes有効かつPhoneのDupe時にTABを押すと、F2ではなくF4の音声ファイルが再生されてしまう
PhoneでF2やF4を再生する機能を使っている人がいるかどうかは不明で、TABを押す人もレアだとは思うのですが、本来期待される動作とは異なるので一応ご報告します。

動作条件: Allow to log dupes無効または有効(どちらも), SSBモード, F4メッセージ登録「無」
CALL欄にDUPE局を入力してTABキーを押した時:
 (2.2e) StatusLineに過去QSOを赤字で表示する
 (6822c52) StatusLineに何も表示しない

動作条件: Allow to log dupes無効または有効(どちらも), SSBモード, F4メッセージ登録「有」
CALL欄にDUPE局を入力してTABキーを押した時:
 (2.2e) F4を再生し、StatusLineに過去QSOを赤字で表示する
 (6822c52) F4を再生するが、StatusLineに何も表示しない ※Allow to log dupes有効時はF2を再生すべき

jr8ppg, 2020-11-16:
確認ありがとうございます。
↓については過去に修正しているのでおかしいなと思ったらSSBですか。Voiceは最近復活したので、、、CWは７月頃に修正しています。了解しました。
> 動作条件: Allow to log dupes無効または有効(どちらも), SSBモード, F4メッセージ登録「有」
> CALL欄にDUPE局を入力してTABキーを押した時:
> (2.2e) F4を再生し、StatusLineに過去QSOを赤字で表示する
> (6822c52) F4を再生するが、StatusLineに何も表示しない ※Allow to log dupes有効時はF2を再生すべき

jr8ppg, 2020-11-16:
修正できました。（v2510へは未マージ）

Fixed: Allow to log dupes有効かつPhoneのDUPE時にTABを押すと、F2ではなくF4の音声ファイルが再生されてしまう件の修正
https://github.com/jr8ppg/zLog/commit/90874c52009e832b2edb03ee0c47ae7396d024fa

Fixed: PhoneのDupe時にTABを押してもStatusLineに何も表示されない不具合の修正
https://github.com/jr8ppg/zLog/commit/187bc82261001d2fa652f827a4fa43f5dae10391

jj1cvh, 2020-11-16:
shift_enter_plan 187bc82 でPhoneの軽微な不具合2点が改善されていることを確認しました。
ご対応ありがとうございました!

closed 2020-11-16 by jr8ppg