jr8ppg/zLog #85: OmniRigを選択するとTrapする

7M4KSC, 2020-11-13:
OmniRigを選択するとTrapしますが、詳しい手順は後ほど。

jr8ppg, 2020-11-15:
残念ながらOmniRigは一度も動かしたことありません。

7M4KSC, 2020-11-15:
了解です、JR1LQKさんのmasterにもOmniRigの残骸があるのですが、旧zlog ver.2.2eには選択肢が無いので、旧zLogのその後の開発途中で、作成中止なった様ですね。時期からして、OmniRig V1なのは判明しました。

7M4KSC, 2020-11-15:
http://www.dxatlas.com/OmniRig/ のOmniRig Clientを立ち上げると、IC-7000の周波数/modeも制御出来るので、
一度八重洲のリグでやってみます。（時間かかります、Hi）
OmniRigが動作すると、アルインコ、Elecraft、Ten-Tec辺りが救えます。

jj1cvh, 2020-12-15:
HDSDRとCW SkimmerでOmniRigをちょっと使ったことがある程度の経験値しかありませんが、zLogがOmniRig対応になると、zLog以外のOmniRig対応ソフトとzLogがVSPEを使わずにリグコンを共用できるはず、という点で本件(#85)には興味を持っています。

JR1LQK/zLogが公開されたときにOmniRigが含まれているのを見た時の私の印象は、zLogのリグコンはOmniRigのソースコードを活用してたのね、ということでした。
ソースコードをちゃんと追いかけた訳ではありませんが、残骸というよりむしろzLogのリグコンの本体という気がします。

ちなみにOmniRigのソースコードはDelphiでgithubに公開されてるようです。
https://github.com/VE3NEA/OmniRig

Delphi5のサンプルプログラムも配布されています。
http://www.dxatlas.com/OmniRig/

7M4KSC, 2020-12-16:
なんと、VSPE使わずに、共用できるかもしれないのですね。興味沸いてきました。
残骸というのはちょっと言い過ぎました、icomでもモードは取得出来ていたので、zLog本体の方を見てみます。
色々情報ありがとうございます。

jr8ppg, 2020-12-16:
OmniRig_TLB.pasというファイルがあるのですが、これを最新のOmniRigのTypeLibから再度作成して入れ替えれば動くかな～と思います。そこまでは調べたのです。

jr8ppg, 2020-12-31:
最新のOmniRigをインストールしてOmniRig_TLB.pasを再作成した物の、色々違いすぎて中断。参ったなー。

jr8ppg, 2020-12-31:
インポートの仕方が駄目だった。やり直してOmniRig_TLB.pas差し替え成功。

7M4KSC, 2020-12-31:
AMD-USB-RIG-CW-SOUND Ver3.5とTS-850Sを繋ぎ、zLog 2.6 beta1 releaseでTS-850を正常の制御出来る事を確認した後に、OmniRig経由にしてみましたが、ICOMと同じく周波数制御が出来ずに、Trapしますね。新しい、OmniRig_TLB.pasの方でやってみますが、これは、OmniRig V2の物なのでしょうか？

jr8ppg, 2021-01-01:
えーと、OmniRig V1.19と出ました。
これで良いですよね、、、

jr8ppg, 2021-01-01:
あーわかりました。
分岐したV2があるのですね。
http://www.hb9ryz.ch/multirig/index.html

jr8ppg, 2021-01-01:
手直しをして例外が飛ばないようにしました。
OmniRIg V1.19 + FT-2000でとりあえず動きました。

7M4KSC, 2021-01-01:
ビルド環境を再構築し、winkeyerブランチをビルドしたところ、OmniRig V1で、TS-850が動作しました！
周波数を取得出来るのも確認しました。修正ありがとうございました。いったんcloseします。

7M4KSC, 2021-01-01:
OmniRigで動作し、TRAPしなくなったので、クローズとします。
確認機種、FT-2000, TS-850

closed 2021-01-01 by 7M4KSC