jr8ppg/zLog #89: JARL国内コンテストに1.9MHz追加

jj1cvh, 2020-12-04:
- ALL JA、Field Day、全市全郡について、
-- zLog Menu画面のBandで1.9MHzを選択可能にする
-- メイン画面のBand欄で1.9MHzを選択可能にする

- (参考情報)
-- https://www.hamlife.jp/2020/12/04/jarl-kokunai-contest-2021-160mb/
-- https://www.jarl.org/Japanese/1_Tanoshimo/1-1_Contest/Contest.htm

jr8ppg, 2020-12-05:
↓の様な所を全て削除でいいのかな
https://github.com/jr8ppg/zLog/blob/master/zlog/main.pas#L7057

あとはコンテストを選ぶ画面
https://github.com/jr8ppg/zLog/blob/master/zlog/UMenu.pas#L296

jj1cvh, 2020-12-05:
自信ないのですが、多分そういうことだと思います。
zLog Menu画面のBand選択に関して、1つバグを見つけました。

- 前回終了時にAll JAを選択していた場合、次回起動時のzLog Menu画面で全Bandが選択可能な状態になってしまう。

[image]

jr8ppg, 2020-12-05:
思ったより修正量あった
完了

jj1cvh, 2020-12-06:
95cb5bd で、以下3点の正常動作を確認しました。
- zLog Menu画面のBandで1.9MHzを選択可能にする
- メイン画面のBand欄で1.9MHzを選択可能にする
- 前回終了時にAll JAを選択していた場合、次回起動時のzLog Menu画面で全Bandが選択可能な状態になってしまう。

別件ですが、関連する問題点を2つ見つけました。

- ALL JAを選択して起動した後、zLogメイン画面でShift+Bを押すと1.9M-10G&Upの全バンドをトグルしてしまう(Band欄クリック時も全バンドが選択可能)。
- (ひとまずALL JA、6m&Down、Field Day、全市全郡で確認) zLog Menu画面(コンテスト選択画面)でシングルバンドを選択して起動すると、zLogメイン画面のバンド欄が常に最初のバンド(ALL JA、Field Day、全市全郡は1.9MHz、6m&Downは50MHz)が選択された状態で起動してしまう。Shift-Bを押すと正しいBandが選択される。

2点目については、zLog Menu画面で[OK]を押し.zloファイルを開くダイヤログボックスが表示された時点では、メイン画面のBand欄に正しいBandが表示されています。しかし、新規zloファイルを指定して[OK]を押すと、メイン画面のBand欄が最初のバンド(1.9M or 50M)に変化してしまいます。

jr8ppg, 2020-12-06:
確認ありがとうございます。

> ALL JAを選択して起動した後、zLogメイン画面でShift+Bを押すと1.9M-10G&Upの全バンドをトグルしてしまう(Band欄クリック時も全バンドが選択可能)。
たぶんこれは前からです。
OptionsのActive Band（QRV可能バンド）で絞れるから良いかな～と放置してあった物です。
正しくは、各コンテストの使用周波数とActive Band両方がONの場合にトグルするになるかと思います。
#91 としました。

> (ひとまずALL JA、6m&Down、Field Day、全市全郡で確認) zLog Menu画面(コンテスト選択画面)でシングルバンドを選択して起動すると、zLogメイン画面のバンド欄が常に最初のバンド(ALL JA、Field Day、全市全郡は1.9MHz、6m&Downは50MHz)が選択された状態で起動してしまう。Shift-Bを押すと正しいBandが選択される。
おそらくこれも前からですね。
これは直した方が良さそう、、、
#92 としました。

2点ともリグコントロールとの兼ね合いが出てきますが、ややこしくなるのでリグ制御はしません。
シングルバンドでもリグ側を別のバンドにするとzLogもバンドが変わってしまいます。

jr8ppg, 2020-12-09:
完了

commit 263c0ecd (2020-09-01): Fixed: #089 バンド変更と#090 モード変更の初期値が逆になっていたのを修正
closed 2020-12-09 by jr8ppg