jr8ppg/zLog #103: CW bank B のメッセージ　＃１の件

JO3JYE, 2021-01-13:
CQモードでもSPモードでも、CW　BANK　Ｂの#1が、F1キーで送信できなくなっています。
bank　Bを選んでいても、F1を押すとBANK　Aの#1となります。
マウスでボタンをクリックした場合はBANK　Ｂの#1が出ます。
本件は、Magical　Callingを実装してからだと思います。

RUNしているにもかかわらず、BANK　Bをセレクトしている使い方を見たことがあります。（見てからから、なるほど！と思い、自分自身そうしていることもあります)
runしているとCQは単発よりループで出すから、CTRL+ZかSHIFT+Z。コールされたら押すのはTAB、最後は↓キーですし。
BANK　Aを表に出す必要がないよね…ということでした。

ただ、CQモードとSPモードの切り替えトリガにF1キーも含まれているので、条件整理も必要かもしれません。

jr8ppg, 2021-01-13:
今開発中の物からF1のデフォルト割り当てが#116のバンクAのCQ送信に変わっています。ご指摘の通りMagical Callingで変更した物です（CQモードの切り替えトリガのため）
これを#010のバンクAのメモリ内容の送信01にすると従来通りの動きになります。
CTRL+ZでCQかけるなら、それで問題ないと思います。

JO3JYE, 2021-01-13:
説明ありがとうございました。
設定変更でokとなりました。
脱線しますが、クイックリファレンスが長らく更新されていないと思います。
デフォルトから変わらない部分とkey-editで変えた部分を足し合わせて、自分用のクイックリファレンスを作る機能というのは他の方からの需要あるでしょうか？個人的にはほしいのですが。

jr8ppg, 2021-01-14:
クイックリファレンスとは ZLOGHELP.TXT のことですかね？

JO3JYE, 2021-01-14:
はい、その通りです。よろしくお願いします。

jr8ppg, 2021-01-14:
zlog_keyedit.exeから出力すると良いですかね。とりあえず話題が違うのでこのIssueはクローズします。

closed 2021-01-14 by jr8ppg