jr8ppg/zLog #107: Dell Inspiron 5502（ノートPC）でのアプリケーションエラー

jkitagawa, 2021-01-22:
以下のノートPCで令和版を起動するとアプリケーションエラーが出るようになりました．
Dell Inspiron 5502（Win10 Home，Intel Core i7-1165G7）

最近，Dellからのデバイスドライバ更新などがあったので，それが影響しているかもしれません．
以前は出ていなかったように思います（普段使っていないPCなので不確かです）．

・オリジナルの2.2c，2.2eではエラーなしでした
・令和版はバージョンに関わらず，エラーが出ます

zlog.exe-exceptionを添付します．2.6 beta2をソースからビルドしたもので，
32bit版，64bit版共にエラーが出ました．
[32bit_zlog.exe-exception.txt](https://github.com/jr8ppg/zLog/files/5854716/32bit_zlog.exe-exception.txt)
[64bit_zlog.exe-exception.txt](https://github.com/jr8ppg/zLog/files/5854717/64bit_zlog.exe-exception.txt)

jr8ppg, 2021-01-23:
こんばんは。
ちょっと調べたところ、謎のUSB HIDデバイスがあるようです。
困りましたね、、、　どうしたらいいんだろう。

参考
https://issuetracker.delphi-jedi.org/view.php?id=6459

jr8ppg, 2021-01-23:
指紋センサーかな

jkitagawa, 2021-01-23:
謎のUSB HIDですか．確かに同じJEDI Issue Tracker

https://issuetracker.delphi-jedi.org/view.php?id=5894

にあったHID Device Testerを試したところ，Device cannot be identifiedと同じ
エラーになるので，HIDの問題のようですね．調べてみます．ちなみに，指紋
センサーは搭載していないモデルです．ありがとうございました．

jr8ppg, 2021-01-24:
指紋センサー無し了解です。
切り分けですが
・USBIF4CWを繋がなくても起きるかどうか
・外付けキーボード・マウス等その他のUSB HIDデバイスの有無は関係するか
・私の方でビルドしたzlog.exeでも発生するかどうか
と言ったところでしょうか。
うちもDELLなのでいつか起きるのかもしれないですね。
最近やたらとUPDATEが来ているので、、、

jkitagawa, 2021-01-24:
・USBIF4CWを繋がなくても起きるかどうか
-> 繋がなくても起きます．

・外付けキーボード・マウス等その他のUSB HIDデバイスの有無は関係するか
->外付けキーボードやマウスは付けていません．他に外部に付けたことがあるのはUSBメモリとUSBIF4CW位です．

・私の方でビルドしたzlog.exeでも発生するかどうか
発生します．

jr8ppg, 2021-01-24:
了解です。
と言う事は本体側の何かと言うことになりますね。
うーむ。

jkitagawa, 2021-01-24:
はい，そのようです．また何か変化がありましたら，こちらでお知らせします．

実はもう1つ，完全に別件でAMDのCPU搭載のPCで仮想COMからのCWキーイングがうまくいかない問題があります．こちらはMLのほうに書きますね．

js2ggd, 2021-01-27:
DELL のノートPC を使用しています。OS は Windows10 64bit版です。オリジナル版(v2.2) の起動ができず、令和版だと起動できるという問題があり、要因を探ったところ、Bluetooth を OFF にすると起動できることを見つけました。令和版はもちろん、オリジナル版でもBTデバイスは何も触っていないはずですが、デバイスの存在が問題を引き起こす可能性がある、ということで一応お知らせしておきます。(2.6 Beta2 はまだ試していません。)

jkitagawa, 2021-01-27:
ありがとうございます．当方とは逆のパターンもあるのですね．お陰様でヒントになり，こちらの原因も分かりました．

*HID PCI Minidriver for ISS*

でした．ジャイロスコープ，加速度センサ，電子コンパスのドライバのようです．このPCは2in1ではなく，画面の
自動回転もないですし，そんなセンサは搭載していないはずなのですが．とりあえず，このドライバを無効にすると
エラーが出なくなりました．ずっと無効にし続けるのもすっきりしないので，今後もっと良い対策があるか考えます．

武藤さんが指紋センサーかもと言った時に，もう少しHIDのドライバについて，詳しく調べれば良かったです．

jr8ppg, 2021-01-27:
なるほど。
実体デバイスが無いから Device cannot be identified と言うことですね。無いのだから無効でいいのではないかと思います。
アプリケーションエラーになるのはどうかと思いますが、JEDIで直してくれないかな。

jkitagawa, 2021-01-28:
そういうことですか! そう考えればDevice cannot be identifiedで正しいことを言っていますね．
確かに実体が接続（内蔵）されていないわけですから無効でいいですね．
それにしても，これでアプリケーションエラーになるとはPascalに似てか，かなり厳格なのですね．

その後，無効で様子見していますが，特に問題なさそうなので，これでいきます．ありがとうございました．

closed 2021-01-29 by jkitagawa