jr8ppg/zLog #110: AMD CPU・チップセット搭載PCでの仮想COM経由CWキーイング不具合

jr8ppg, 2021-01-31:
問題
AMDのCPU（チップセット）を搭載したPCの一部で，仮想COM経由のCWキーイングを
した場合に，符号が正常に出ないことがあります．令和版もオリジナル版2.2cも同じ症状です．
ファンクションキーや令和版のメッセージパッドだと正常な場合もあり，キーボード
キーイングでの不具合が顕著です．

不具合の出るPC
1.  デスクトップPC 自作機，CPU; AMD Ryzen 5 3400G，チップセット; B450，Win10 Pro

2. ノートPC HP ENVY x360 13-ar0115AU，CPU; AMD Ryzen 7 3700U，Win10 Home

試したIF
1. 八重洲 SCU-17（USBブリッジはSilicon Labs）
FSK出力（DTR）からCWを出し，PTT出力（RTS）も使用

2. USB<->232C変換（BSURC06，変換チップはFTDI）+自作IF
D-subの4番（DTR）でCW，7番（RTS)でPTT

RIGは接続ケーブルの都合上，FTDX5000 MPで試しましたが，RIGは
関係ないように思っています．SCU-17も他社RIGでも使えると思います．

症状
・例えば，`Q'を送信しようとしても`K'になったり，そもそも符号が
  ほとんど送信されずに切れたり，一瞬送信状態になるだけだったりで，
  キーボードキーイングではまともに符号が送出されません．

・CTESTWINやDXLogではAMD CPUと仮想COM（上記IF）において，
   キーボードキーイング含め全て正常です

・仮想COM（上記IF）ではなく，USBIF4CW（Ver.2.2）を使うと
   zLogとAMD CPUでも正常です

・Intel CPU・チップセット搭載のPCだと，zLogと仮想COM（上記IF）
  でも正常です

jr8ppg, 2021-01-31:
同様症状の方。
https://twitter.com/JJ1DPG/status/1323990669640101889

この事例は他のPCでは問題無いとのことで、レッツノートR7とSerial-USB I/Fの相性かな？程度で追及終わりとなった。

jr8ppg, 2021-01-31:
USBセレクティブサスペンド機能によるサスペンドからの復帰時に頭切れする可能性を確認したが関係無かった。

jr8ppg, 2021-01-31:
7M4KSCさんの追試で問題の再現性は確認できた。

マザーボード：Gigabyte AB350M-D3H (一番古いPCI付の安マザー)
CPU：Ryzen5 2400G
RAM：DDR4 8GB
IF：JN2AMD AMD-USB-RIG-CW-SOUND Ver.3.5
zLog：2.6 beta2

jr8ppg, 2021-01-31:
7M4KSCさんからの情報

> 続報、
> なんとですが、JH4VAJ OMのPCWI01のWCH社のUSBシリアルCH340Gだと、
> Ryzen 2400G AMD CPUでも、DTRでもキーイング出来て、CW Keyboardも動作します。
> JN2AMD OMのFTDIのUSBシリアルの時だけ、変な動作になる。
> 他のソフトは動作するので、回避方法あるのかな・・・

jr8ppg, 2021-01-31:
CPDrv.pasの検証
特段不審な点は見当たらないが、下記の4点を試す。
1. CP2105のドキュメントで、SetCommState()後は60milisec待てとの記述　→　改善しない
2. FHandleを取得直後に、有無を言わせず、PurgeComm()する　→　改善しない
3. CreateFile()時にFILE_FLAG_OVERLAPPEDを追加　→　改善しない
4. SetCommTimeOuts()しない　→　改善しない

うーむ

jr8ppg, 2021-01-31:
リグコントロールは問題ない模様、、、DTR/RTSのON/OFFのみ障害？

jr8ppg, 2021-01-31:
Core2 DuoのPCにBuffaloのUSB-Serialアダプタ(FTDI)で試すと現象再現。

jr8ppg, 2021-01-31:
原因について検討しました。一応合理的に説明はできていると思います。

現象の発生プロセス
1.zLogは一定間隔でタイマーをかけてCWのキーイングをします。
2.タイマー処理で送信OKフラグがOFFならタイマー処理を終了します。
3.タイマー処理で送信する文字が無ければDTRをOFFにします。
4.この時、送信OKフラグをOFFにしないので、次のタイマー処理時にまたDTRをOFFにします。
5.何か送信操作が行われるまで、これ（1～4）を繰り返します。
6.そうすると仮想COMポート側には多数のDTR OFF命令が届きます。これらは未処理で溜まっていきます。
　正しくは次々DTR OFF処理は行われるのだけど、溜まっていく方が多いようです。
7.次にキーイングを開始してDTRがONになります。
8.しかし、6で溜まっているDTR OFFの命令の影響で、瞬時にDTRがOFFになります。（並列処理？）
9.結果として符号にならなくなります。

機種やUSB-Serial Converterの種類で現象が出ないのは、、、
仮想COMポートドライバの実装で、無駄なDTR ON/OFF命令は無視され、溜まらない様になっているのではないかと思います。
あとは十分に処理が早いとか。

旧zLogで問題なかったのは、当時はUSBではなく本物のCOMポートばかりだったので、
DTR OFFの命令が溜まる様なことは無いためと考えられます。

どう直すのか？
上記の3.の処理の後で、送信OKフラグをOFFにする様にします。
これで無駄にDTR OFF命令が出なくなります。

jkitagawa, 2021-01-31:
原因の検討，すばらしいと思います．キーボードキーイングで確かに同じキーを押し続けていると正常に出ることがありましたので，その時はたまっていたDTR OFFがなくなって正常にキーイングできていたのでしょうね．

jr8ppg, 2021-01-31:
そうですね。それで合っていると思います。
結果としてはAMDだからとか、そういうのは関係ありませんでした。
クローズします。

closed 2021-01-31 by jr8ppg