jr8ppg/zLog #113: CFG で定義するDXコンテストの問題

js2ggd, 2021-02-14:
1. "cty" のファイルを指定する引数の問題

令和edition では cty は「廃止」と扱われている (ファイル名固定で cty.dat を読み込む)が、CFGで定義するコンテストでは引き続き cty を指定しなければ読み込まれないため、「廃止」ではなく、「引数無しのキーワードに変更」と記載すべきではないでしょうか。

2. "cty" と 他のマルチ定義ファイル(*.DAT) を併用する場合の multi info window の問題

cty キーワードと dat キーワードを併用してDX entity と local マルチを定義すると、multi info window においてまず上方に local マルチが表示され、"Unknown Country" の表示の下にctyを参照した entity のリストが表示されます。このとき、以下の**3つの問題**があります。

a)  entity のマルチにチェックがある行(いずれかのバンドで交信済み)が赤く表示されるが、DATを併用した場合、**赤い表示が下に一行ずれてしまう。** (DATを併用しなければ正しく表示される。)
b) entity の表示にはentity名、CQ ZONE, 大陸名 のみが表示されるが、**ITU ZONE** をコンテストナンバーとしているコンテストにとっては不便である。(ctyコマンドで任意のファイルを読込・表示出来なくなったことによる弊害。対策として、cty.dat に記述されている両ZONEをともに表示することでとりあえずの問題は対処できる。)
c) entity と、送信コンテストナンバーの**両方にマルチの意味がある場合に、現状では対応できない**。(例: EUDXコンテスト。)

c については令和edition 固有の問題ではないため、別途の対応になるかと思いますが気づいた点をまとめて記載しました。

jr8ppg, 2021-02-15:
１．について
改めて確認したところ、ご指摘の通りなので下記のページの記述を修正しました。
https://github.com/jr8ppg/zLog/wiki/CFG%E3%83%95%E3%82%A1%E3%82%A4%E3%83%AB%E4%BB%95%E6%A7%98#%E3%83%87%E3%83%BC%E3%82%BF%E3%83%95%E3%82%A1%E3%82%A4%E3%83%AB%E6%8C%87%E5%AE%9A

２－ａ、２－ｂについて
これは修正したいですね。

２－ｃ
これは現状仕方ないかな。

jr8ppg, 2021-02-15:
２－ａ、２－ｂを再現できるcfgとdatをもらえませんか？

jr8ppg, 2021-02-15:
２－ａは現象確認
２－ｂはどうしますかね？

現状こうなっているのを

> JA     Japan          25 AS 

こんな感じで？

> JA     Japan          25 AS 45

js2ggd, 2021-02-15:
JA Japan 25 45 AS  とどちらかですが、、、CQ/ITUが同時に使われることはまずないということを意識すると、ASを挟むPPGさんの案の方が区切りにもなっていいですかね。

jr8ppg, 2021-03-13:
ようやく修正完了！

jr8ppg, 2021-03-14:
２－ａの方は、DATファイルの最後の行がCTY.DATの１行目で上書きされていたため１行ずれていました。

js2ggd, 2021-03-16:
ご対応ありがとうございました。一旦 Close にさせて頂きます。

closed 2021-03-16 by js2ggd