jr8ppg/zLog #125: OptionsウィンドウとCW/RTTYタブの開き方

jj8dan, 2021-03-10:
Windows-OptionsウィンドウとCW/RTTYタブのお話です。
Windows的にはOptionsは開いたらフォーカスがOptionsに行き、メインウィンドウは操作できません（Beepがなる）。これはこれで正しい。ただ困ったことにWindows-Options-CW/RTTYタブが存在し、これはオペレート中に見ながら操作したい。
22E版のころはOptionsを開きっぱなしで操作できたので開きっぱなしで（CW/RTTYタブを見ながら）操作してました（Optionsが開きっぱなしなのは行儀悪いですね）

さて、CW/RTTYタブは、どうあるべきか。思いつくのは3案で3案目がスマートと考えています。

1. オペレート中には操作できない、いったんOptionsに入る必要がある（現行のまま、使いにくい）
2. Optionsを開きっぱなしにできるようにしてしまう（Windowsの礼儀作法からは外れる）
3. いっそCW/RTTYタブを独立したウィンドウにしてしまう（大改造の予感）

メインウィンドウの緑丸にマウスオーバーすればメッセージは見れますがCW/RTTYタブに比べるとわかりにくい。
（Cf. Voiceタブも同じ問題でしょう。）

jr8ppg, 2021-03-11:
こんばんは。
この手の話はちょくちょく出てくるのですが、Optionsを開いておきたいのは結局の所、何ができれば良いのでしょうか。
CW/RTTYの送信電文を頻繁に変えたいのでしょうか？　それとも設定内容を確認したい（見たい）のでしょうか？
それによりできる対応が見えてくるかと思います。

jj8dan, 2021-03-11:
こんばんは。どちらかといえばFAQと理解しました。

何ができれば良いかといえば、Options-CW/RTTY-Messagesの中身が「みたい」、です。

電文編集は最初に調整すれば良いでしょう。頻繁に変える必要はありません（※１）
焦った時に例えばNR?を打ちたいのだがファンクションキーどれだっけどれだっけ、になります。
対応するファンクションキーがどれであることが知りたい(ちなみにDefaultだとF5)。

他のSpeed類はメインウィンドウで変わるし、その他も最初に設定すればいいから
別に開いておく必要はないですね。
Shift+F でA/Bが切り替わることは、マルチOPだと始める前に教育しないといけないですね。

（※１）マルチOPだと変わる必要があるやもしれない。運用中にそれぞれ自分の電文に
切り替えたい(≠書き換えたい)需要はあるかどうか。zlog_oplist.iniが活躍しますかね(Issues #58)
　(蛇足：Abbreviation（019)も趣味が出ていてOAN派、TAN派、019派(短縮ナシ)と別れるが、普通のマルチOPでそこまで区別する必要はないでしょう。それこそOptionsを開いてやればよろしい)

jj1cvh, 2021-03-26:
機能要望兼ディスカッションです。
特にCWで、Functionキーのメッセージ内容を見たいというニーズは結構ありそうに思いますので、簡略表示機能があるとベターだと思います。

私のクラブ局運用経験の範囲では、メッセージ内容をOpごとに変更することは無くてクラブ局内で統一したメッセージを割り当てていますが、クラブ局によって流儀が違うので、下記のようなtxtファイルを用意しておき、ディスプレイ下部に3行で表示する工夫をしています。
※zLog win 2.2eの例ですので、F9とF10は未設定です。

(クラブ局A)
```
   F1    F2    F3     F4    || F5    F6    F7    F8    || F9    F10   F11   F12|
A: CQ    $C+NR TU+$M  $C+NR || NR?   $M  R_5NN$X 5NN$X ||             ?     TU |SHIFT+F: 送信メモリバンクA/B切り替え
B:       $M    TU           || NR?   $M  R_5NN$X 5NN$X ||             ?     TU |SHIFT+F1～F8:裏バンクのメモリ内容の送信
```
(クラブ局B)
```
   F1    F2    F3     F4    || F5    F6    F7    F8    || F9    F10   F11   F12
表 CQ    $C+NR TU+$M  QSOB4 || NR?   ?     5NN$X TU    ||             $C    $M
裏                          ||       AGN   $X    QSL   ||             $C    $M $M
```
裏バンク(メモリバンクB)も設定してあるのですが使うオペはごく少数で、メモリバンクAに設定したCQとS&Pのメッセージだけで交信しているオペが多いです。

DOS版のZLOGCG.EXEは、Functionキーのメッセージ略語が画面下部に表示されていたので、大変分かりやすかったです。
https://www.jarl.com/nttrd/denden-test/picture/zLogdos1.png

N1MM+もボタン表示で[F1 Cq][F3 Tu]などが表示されるので分かりやすいですね。

CTESTWINは、ツールバーのCWボタンの文字列(4文字程度)をカスタマイズする機能や、Functionキーの割当内容を別Windowで表示する機能[1]があります。

[1] http://e.gmobb.jp/ctestwin/CTWHELP/Allfunc.html
　４９．Functionキーの割当内容の表示

jr8ppg, 2021-03-27:
うーん、話はわかるのですが、具体的にどんな物にすればいいかで悩み中です。
とりあえず何か試作してみますか。

jr8ppg, 2021-03-28:
functionpanelブランチで何か作ってみます。

jr8ppg, 2021-03-31:
とりあえずベース部分はできた。zlog_keyedit.exeでパネル表示内容を設定できるようにする予定。

ex)
zlog_key.ini
[text]
116=CQ開始

jr8ppg, 2021-05-02:
@jj8dan @jj1cvh 
まずはこんな形でどうですかね。
・Windows－Function Key Panelメニューで表示
・Function Key PanelはF1～F12までに割り当てられたCW Message又はzlog_keyeditで設定したパネル表示名を表示します
これじゃよくわかりませんね。
資料作ります。

jr8ppg, 2021-05-02:
説明資料です。
https://github.com/jr8ppg/zLog/blob/images/%E3%83%95%E3%82%A1%E3%83%B3%E3%82%AF%E3%82%B7%E3%83%A7%E3%83%B3%E3%82%AD%E3%83%BC%E3%83%91%E3%83%8D%E3%83%AB.pdf

jr8ppg, 2021-05-02:
後は、現在バンクとSHIFTキー押下状態で表示と動作が切り替わればＯＫかな

jj8dan, 2021-05-04:
PDFはいけんしました。素晴らしい。

見ていると次のような考え方も出てきました：（ただし当初の目的から外れる）「CWのメッセージではなくその要約を書きたい人が出て来るのでは（他のfunctionキー設定と同等）」。やりすぎかな、たんにCWはCWメッセージ表示に特化した方がいいな。

zLog_keyeditで設定した機能とパネル表示するのは良いです。パネル表示名のDefaultと標準に戻すボタン
必要かどうか。

jj1cvh, 2021-05-05:
ご対応ありがとうございます。説明資料の動作で宜しいかと思います。

jr8ppg, 2021-05-08:
一旦完成！

closed 2021-06-08 by jr8ppg