jr8ppg/zLog #126: 再起動時に入力欄のBandが変わってしまう

jj8dan, 2021-03-12:
zLogを終了、再起動した際に最後にQSOしたBandが入力枠のBand欄に残っていてほしいところですが
(あるいは最後に切り替えたBand)、

実際はOptions-Preferences-Active bands and powersチェックボックス(*1)のチェックされた
最も低いバンドになってしまう。
例えば(*1)のすべてにチェックを入れた場合、再起動時は入力枠のBand欄は最後の交信にかかわらず
1.9 (MHz)となる。(*1)の1.9 MHzを除くすべてにチェックを入れた場合、3.5 (MHz)となる。
(*1)の5600 MHzのみにチェックを入れた場合、Band欄は5600 (MHz)となる。
モードは維持される(SSB, CW, ～)

他の条件
全市全郡のMultiOPで確認した。Operatorと(*1)のpowersは同じにした(WARC以外)(HHHHHHHMMLPPP)
リグコントロールなし、zServerなし

2.2eでは発生しない。2.6 β2, 2.6β4にて発生を確認した。

jr8ppg, 2021-03-14:
現在は、終了時のバンド・モードを記憶していません。
各コンテストが必要としているバンドと、自分がQRVできるバンド（Active Band）の両方がONのバンドのうち、一番低いバンドが起動時に選択されるバンドです。旧zLog（2.2）ではそのあたりがいい加減で、起動時のバンドがActive Band外になったりしていましたので直した物です。

旧zLog（2.2）でも終了時のバンド・モードは記憶していないと思いますが、、、

リグコントロール（自動でリグのバンドに合わせる）を使用されている方が多いためか、終了時のバンド・モードを記憶していて欲しいと言う話はまだありません。

jj8dan, 2021-03-15:
こんばんは。今までも意図した動作であったこと、わかりました。
Changedとのことで了解です。ありがとうございます。

closed 2021-03-31 by jr8ppg