jr8ppg/zLog #131: 「memo」欄に「(0.0)」が毎入力ごとに入力される

jj8dan, 2021-04-05:
1交信の入力完了時にmemo欄に(0.0)が入力されるようになってしまった。

継続はしていますが再現していなません。次のような操作を行いました。
「File-New Contest-」で開いたzLog Menuで[Cancel]キーを押して終了させて(勝手に終了する)
次回起動すると、1交信ごとです。

ContestはDX Peditionと全市全郡を選んでみました。
zlog.iniを作り直せば元に戻るようです（自信なし）。

jj8dan, 2021-04-05:
具体的にはこんな感じです
[image]

jr8ppg, 2021-04-06:
こんばんは。レポートありがとうございます。

確認したところ下記の条件で再現しました。
１．リグコントロールを行う設定になっている
２．通信条件が違うなどの理由でリグから情報が取れていない（周波数表示は00000.000KHz）
３．Options-Rig controlタブ-Record rig frequency in memoがON

zlog.iniを削除すると上記１と３がクリアされますので起こらなくなります。

対処
通信条件などを見直してリグから周波数を取得できる様にして下さい（上記２）

です。

jr8ppg, 2021-04-06:
そうそう、(0.0)はリグの周波数です。正しく取れていれば、(7123.4)と言った感じで表示されます。

jj8dan, 2021-04-06:
こんばんは。
その通りです。1.と2.  リグコントロールON のままリグの電源がオフになっていた。 3. ☑ Record rig frequency in memo。
言い訳をするとzlog_v26_beta4_release_note.pdfの31ページ「#100 CABRILLOファイルにACTUAL FREQUENCYを出力する」を読んで（紙ログで周波数細かく書いとく癖があるので）これ便利だなで終わってしまいmemo欄に影響するまで読み下せませんでした。早とちりで申し訳ないです。FAQになりそうな気がします。

closed 2021-04-06 by jj8dan