jr8ppg/zLog #138: WinKeyerで無変換キーでのPTT制御ができない

jr8ppg, 2021-04-17:
無変換キーでPTTが動作しない、通常のキーイング時はPTT制御が行われる。

jr8ppg, 2021-04-17:
マニュアルには
https://www.k1elsystems.com/files/WK3_Datasheet_v1.3.pdf
PTT On/Off    <18><nn> nn = 01 PTT on, n = 00 PTT off
The command is operational only when sidetone and PTT are disabled (See PINCFG command)
と書いてある。

jr8ppg, 2021-04-17:
ALL JA対策用
https://github.com/jr8ppg/zLog/releases/tag/ZLOG2607

JO3JYE, 2021-04-18:
対策お疲れ様です。
PINCFGのところ。　
10101100($AC)　ではパドル入力が止まってしまいます。
10101110($AE)　だとパドル入力が再開。
BIT1のサイドトーンenableを追加していますが、何ででしょう？？
パドル入力だけサイドトーンが止められないと何か関係あるかも？
BIT0のPTTenableは触っていないので、本来の修正目的であるPTT動作には支障がないはずですが。

jr8ppg, 2021-04-18:
sidetone and PTT are disabled　なのでそうしたんですよね。
さっき試してみたら、パドルは動作しました。
ただ、パドル割り込み後、PTTがOFFになりませんでした。これはまずいかな。

JO3JYE, 2021-04-18:
やっぱりダメでしたか・・困りました。

JO3JYE, 2021-04-18:
PHONEとCWを切り替えるたびに、PINCFGをセットし直すしかないのでしょうか。

jr8ppg, 2021-04-18:
ところで、JYEさんのところではパドル動きますか？？
PHとCWでの違いは無いように思いますが、、、

パドル入力のお知らせを受信して、PTTをOFFにすればいいのですが、そんなロジック無いですね。

JO3JYE, 2021-04-18:
なんだか、こちらのチェックが怪しく何か勘違いしているかも?なので、公開されているPTT対策版でもう一度最初からやり直してみました。

・WK3だとパドルは受け付ける
・サイドトーンは　パネルのボタンを押しても出ない
・pttは動作する。ただし、パドル入力時は動作しない。
・ptt　ONの状態でF1等のメッセージを送り終えると、PTTはOFFになる。

このような感じです。

JO3JYE, 2021-04-18:
最後のところ
　無変換でptt　onにしてからf1メッセージ再生　したケースです。

jr8ppg, 2021-04-18:
同じですね。
> pttは動作する。ただし、パドル入力時は動作しない。
これが解せないところです。
FT-2000なんですが、[BKIN]をONにすると、パドルでもPTT動作します。OFFにすると動きません。
CQ等は[BKIN]のON/OFFにかかわらず、PTT動作します。不思議。

jr8ppg, 2021-04-18:
パドル割り込み対応入れました。

jr8ppg, 2021-04-18:
branchにすれば良かった。

jr8ppg, 2021-04-18:
結局の所、無変換キー押下時（ON）は、SETPINCFGコマンドでPTT Disableにして、OFF時はPTT Enableとすることで対応。
あとはWinKeyerにお任せ。
アプリから制御することをあまり考えていないみたい。

jr8ppg, 2021-04-18:
これで決着
https://github.com/jr8ppg/zLog/releases/tag/ZLOG2609

jr8ppg, 2021-06-01:
一旦、済

closed 2021-06-01 by jr8ppg