jr8ppg/zLog #163: WinKeyerでTABキー押下後、ナンバー送る際にPTTが効かない

jr8ppg, 2021-05-26:
WinKeyerでTABキー押下後、コールサイン送出時はPTTが動作するが、ナンバー送る際にPTTが効かない。

jr8ppg, 2021-05-26:
コールサイン送出時はWinKeyerControlPTT()を呼んでいる。UzLogKeyer.pasの2569行目。https://github.com/jr8ppg/zLog/blob/master/zlog/UzLogKeyer.pas#L2579

それに対して、ナンバー送出時は#18#01のみ。2612行目。
https://github.com/jr8ppg/zLog/blob/master/zlog/UzLogKeyer.pas#L2612

なので、PTT ON=#09#AE#18#01、PTT OFF=#18#00#09#AD にすると良いのかな。
https://github.com/jr8ppg/zLog/blob/master/zlog/UzLogKeyer.pas#L2616

jr8ppg, 2021-05-26:
違うな
コールサイン送出後、[2877行目](https://github.com/jr8ppg/zLog/blob/master/zlog/UzLogKeyer.pas#L2877)でナンバー送信し、[2881行目](https://github.com/jr8ppg/zLog/blob/master/zlog/UzLogKeyer.pas#L2881)でPTTをOFFにする。

と言う事は、[Main.pasの5892行目](https://github.com/jr8ppg/zLog/blob/master/zlog/main.pas#L5892)では、
PTT制御しないように、dmZLogKeyer.WinkeyerSendStr(S, False); で呼んでやれば良いということになりますかね。

週末に動かしてみます。

jr8ppg, 2021-05-26:
ESCを押すと、
WinKeyerClear();が呼ばれて、中でWinKeyerSetPTTMode(False);が呼ばれます。こいつがくせ者かもしれません。

JO3JYE, 2021-05-26:
対応ありがとうございます。

dmZLogKeyer.WinkeyerSendStr(S, False); 　なんですね。
実は、思い当たる節があり試したけどダメだったというのは。この部分を　False　ではなく　True　としてました。
事象１の対策はここしかないと思います。

＃読み解きがもう一息足りませんでした(^^;;。

ESCキーやstopによるキャンセルを考えると、WinKeyerClear()　の　WinKeyerSetPTTMode(False);　っているような・・
あと、この手続き内で、PTTのOFFが先か？キャンセル電文が先か？　気になり入れ替えましたが、結果は変化なし(T_T)
sleepでwaitを突っ込んでみたのは、この手続き内です。200ｍSだと一瞬、できた！と思いましたが速度を落としたらダメ。
ということで、見当違いなのかなぁとみています。あとPTTのレスポンスが遅くなる点でもよろしくないかと。

PTTが崩れる件、キーイング速度を遅くしたときの方が顕著かもしれません。
なかなか奥が深いです。

JO3JYE, 2021-05-26:
キャンセル電文の後　sleep(200)　として　ptt処理の電文を遅れて飛ばしてみた、という作業でした。
これは違うかなぁー、ということで。

jr8ppg, 2021-05-26:
了解しました。いろいろお試しありがとうございます。
これは無線機に繋がないと動作がわからないので、週末にでもゆっくりやってみます。
PTTなので電波が出てしまうんですよね・・・うーん。

JO3JYE, 2021-05-26:
無線機につないでしまうと、PTTのテストをすれば電波が出てしまいますね。
うちは、アンテナもリグも普段は押入れなので、先日、テスト治具を作りました。
単に電池駆動でledとブザーを鳴らすだけですが。(keyはブザーとLED、pttはLEDのみ）
CWやPTT関係はこの治具を作ってからは、試しやすくなりました。

JO3JYE, 2021-05-27:
＞PTT ON=#$09#$AE#$18#$01、PTT OFF=#$18#$00#$09#$AD にすると良いのかな。
この件、試しました。PTT　ONは大丈夫ですが、PTT　OFFが利かなくなりました。
(電文を最後まで再生しても、途中停止でも同じ)

JO3JYE, 2021-05-29:
なんと。。Option→Hardwareのタブにある、CW　PTT　control　のEnablePTTcontrol　のチェックを外すと、すべての動作がPTT　onで正しく動きました。(今までの話題に出た修正しては？という箇所は何もしていません。修正不要かと思います）
事象の１，２，３。今の段階では、いずれも発生しなくなりました。
逆に、PTTを使わないという選択ができない・・まぁ、つながなければ良いのですが。

EnablePTTcontrolのフラグの動きを追いかける必要があるように思います。

でも、普通は、PTTを使うとめいじするのだからPTTコントロールは絶対ONにしますよねー。

JO3JYE, 2021-05-29:
修正です。
optionでEnablePTTcontrolのチェックを抜くと、PTTはONになるけど電文終わりにOFFならない。
TABキーの動作、ナンバーまでちゃんとPTT　onでした。
書き換えた箇所を順番にたどり直します。
どうやら、修正を繰り返したコードで確認したので、私も見失っていると思います。う~ん、

JO3JYE, 2021-05-29:
optionでEnablePTTcontrolのチェックを抜くと、PTTのON/OFFも正しく動く場合と、上記のように電文終わりでOFFにならないケース。
両方あるみたい、発生条件がまだわからない。
コードは、触ったけど、戻し切っていました。(結局zipから展開し直して確認が一番早くて…)

dmZLogKeyer.WinkeyerSendStrでのPTTコントロールは効いていないように思います。
引数でPTTあり・なしを指定しても別のところで支配されている感じです。(TABにかぎらず、F1キーなどでも)

25府在住ですので、StayHome。一人呑みしながらcodeを見ていました。

jr8ppg, 2021-05-29:
こんにちは。
確かにEnable PTT ControlがOFFの方がうまく動いているように見えますね。
これは無変換キーでPTT ON/OFFの件で色々やったときにWinKeyerが勝手にPTT制御やっていると言う結論だったかと思います。#138

Enable PTT ControlがONの時は、無変換キー押下時と同じ事をやってやればいいのかな。
ちょっと考えます。

jr8ppg, 2021-05-29:
無変換キーと同じようにPTT制御を変えてみました。よさげになりましたがどうですかね。
https://github.com/jr8ppg/zLog/tree/wkpttfix 

コール送出後に一瞬PTTがOFFになるのが気になります。なんでかな。

JO3JYE, 2021-05-29:
対応ありがとうございます。
TABキー時のナンバーでもPTTがonになるようになりました。ただ、ナンバー送信終わりでPTTがOFFにならないことがあります。この状態の時はF1キーなどでの送出でも電文終わりでPTT　OFFになりません。
違いはなんだろ？動くときは動くのだけど…何かのきっかけでダメになります。
一度ダメになると、zlogの再起動だけでは不十分でwinkeykerの再起動も必要です。

相変わらず、Enable PTT ControlがOFFのほうが、まともにPTTが作動しているように見えます。
なんでだろう？？

jr8ppg, 2021-05-29:
> TABキー時のナンバーでもPTTがonになるようになりました。ただ、ナンバー送信終わりでPTTがOFFにならないことがあります。この状態の時はF1キーなどでの送出でも電文終わりでPTT　OFFになりません。

そういえば、この現象は出ていませんね。PTT After Delayが0だったりしませんかね。

jr8ppg, 2021-05-29:
PTT After Delayは関係ありませんね。

jr8ppg, 2021-05-29:
うーん、CW Keyboardで1文字送信するとフォーカスがメインウインドウに行ってしまうことに気がついてしまいました、、、なんでかな

jr8ppg, 2021-05-29:
> うーん、CW Keyboardで1文字送信するとフォーカスがメインウインドウに行ってしまうことに気がついてしまいました、、、なんでかな

原因判明。Speed PODの位置でWPMがドリフト？している感じ。25になって、すぐに26になる。
ちょっと回すと25や26で安定するので現象は起こらない。

JO3JYE, 2021-05-29:
確かに、speed podの位置によって、wpmのスライドバーがちらつく現象はあります。
ちょっとどちらかへ回すと落ち着きます。

JO3JYE, 2021-05-30:
いろいろとありがとうございます。
なんだか悩ましいです。
PTTがオフにならない現象はTABキーを押した後にでています。一度この状態に出くわすと、それ以降の送信でPTTがOFFになりません。

#138の通り、winkeyerが勝手にPTTコントロールするのなら、zlogからのコントロールをやめてはどうでしょうか？
試しに、uzlogkeyer.pas内の
TdmZLogKeyer.WinKeyerSendCallsign(S: string);　にて　ControlPTT(True);　をコメントアウトして制御しないようにしてみました。（2579行目）
この状態だと、こちらの環境ではPTTにまつわる不具合が解消しています。

jr8ppg, 2021-05-30:
それもそうですね。
そうすると、https://github.com/jr8ppg/zLog/blob/wkpttfix/zlog/UzLogKeyer.pas#L652 を手直しして、PTT Controlするしないにかかわらず、PTT Delayの設定をするようにした方がいいですね。
あとは、無変換キーがちゃんと動けばＯＫかな。

JO3JYE, 2021-05-31:
検討ありがとうございます。
こちらでは無変換キーのPTT　ON/OFFもできました。

PTTcontrolの有無を問わず、PTTdelayができればOKかと思います。
https://github.com/jr8ppg/zLog/blob/wkpttfix/zlog/UzLogKeyer.pas#L652　の部分の書き換え。
変化を見るだけなので、あんちょこに655行目は、652行目のコピペとして、IFを骨抜きにして試しました。
BEFORE側は、okかと思います。after側が、PTTcontrolのチェックで動作が違うかも？チェック有の時にafterの数値が反映していないように思われます。(パドル入力だけ数値が反映している感じ)
もうちょっと試してみます。

あと、option設定、PTTcontrolのチェックでbeforeとafterの設定入力の許可・禁止をしているので、プロパティの変更条件を見直さないと・・winkeyerのチェックも設定可否の条件に加えないといけませんね。

なんとか、ゴールが見えてきたかも！

jr8ppg, 2021-05-31:
色々確認ありがとうございます。ちょっと体調崩して寝込んでいます、、、

PTT Controlコントロール有りの場合は自分でAfter Delay Timeを取らないとだめみたいですね。

> あと、option設定、PTTcontrolのチェックでbeforeとafterの設定入力の許可・禁止をしているので、プロパティの変更条件を見直さないと・・winkeyerのチェックも設定可否の条件に加えないといけませんね。

確かに、、、

あと、この手の↓コードは全部要らないですね。電文中の"(" や ")"もいらんのかも。
```
if FPTTEnabled = True then begin
      ControlPTT(True);    又はWinKeyerControlPTT();
end;
```

JO3JYE, 2021-05-31:
体調不良はまずいですね。どうぞお大事に！

確かに、電文中の”（”とか　”）”は不要だと思います。デバッガーで見ていた時も、これって要るの？疑問でした。
uzlogkeyer.pas内のwinkeyer　PTT制御に関する部分で、使わないところはコメントアウトなどの対応しておいた方が良いと思います。

jr8ppg, 2021-06-01:
これでどうでしょうかね。

JO3JYE, 2021-06-01:
対応ありがとうございます。
こちらではPTTの動きは良くなりました。無変換キーもokです。
option設定のPTTcontrolのチェックの有無を問わずafterも反応しています。
TABキーを押してコールサインを送信しているときにstop(esc)をかけても、パドルをさわっても止まるまでの反応がかなり鈍いです。割り込みの関係だと思います。手直しができれば良いのですが。

Uzlogkeyer.pas内のTdmZLogKeyer.WinKeyerClear()　の最後に　PTT　OFFの操作が残っています。
今のところ、実害なさそうですが、どうしましょうか。(コメントアウトしてみたが、動作が変わった感はありませんでした)

これで、optionのプロパティ関係の残るは手直しでしょうか？

issueのテーマから外れますが、wkのサイドトーンのON/OFFに少し難があります。
サイドトーン無しでzlogを起動し、サイドトーンをONにしても、tabキー以外の操作だとサイドトーンが出ません。
無変換キーを押したら出るようになります。(ほかにも出るようになるきっかけの操作があるかも)
pincfgを送るタイミングなんでしょうか？一度、サイドトーンが出るようになると、後は問題なく動きます。
最初のところだけみたいです。
うちだけでしょうか？現象出ていませんか？

jr8ppg, 2021-06-01:
おはようございます。

> option設定のPTTcontrolのチェックの有無を問わずafterも反応しています。
> TABキーを押してコールサインを送信しているときにstop(esc)をかけても、パドルをさわっても止まるまでの反応がかなり鈍いです。割り込みの関係だと思います。手直しができれば良いのですが。
> 

コールサイン送出中は試してなかったかも。見てみます。

> Uzlogkeyer.pas内のTdmZLogKeyer.WinKeyerClear()　の最後に　PTT　OFFの操作が残っています。
> 今のところ、実害なさそうですが、どうしましょうか。(コメントアウトしてみたが、動作が変わった感はありませんでした)
> 

これは大丈夫です。WinKeyerSetPTTMode(False);は、PTT制御をWinKeyerやってね～なのでOK。

> これで、optionのプロパティ関係の残るは手直しでしょうか？

PTT Control ONにしないとDelayが入力できないので、PTT Controlしたい場合はONにしてもらって
Delay Timeを入力してもらう方が自然な感じがするのでこのままでいいかなと思います。

> issueのテーマから外れますが、wkのサイドトーンのON/OFFに少し難があります。
> サイドトーン無しでzlogを起動し、サイドトーンをONにしても、tabキー以外の操作だとサイドトーンが出ません。
> 無変換キーを押したら出るようになります。(ほかにも出るようになるきっかけの操作があるかも)
> pincfgを送るタイミングなんでしょうか？一度、サイドトーンが出るようになると、後は問題なく動きます。
> 最初のところだけみたいです。
> うちだけでしょうか？現象出ていませんか？
> 

これはやってみますね。

JO3JYE, 2021-06-02:
おはようございます。了解です。

送出中のキャンセル、見ていただけると助かります。操作的には十分にあり得るでしょうし。
PTTのメニューはそうですね。使わない時はチェックしませんし、このままのが判りやすいか…

サイドトーンの問題、前は無かったように記憶しています。最初のころはwk単体で見ることが多かったので、最初に出ない状態だと絶対に気が付いているはず。。無変換キーのPTT操作を追加した後のどこかからのような気がします。
まずは再現するかどうかですね。

jr8ppg, 2021-06-02:
Abort処理見直ししました。これで結構良くなったと思います。
ついでにサイドトーン問題も直ったような気がします。
（https://github.com/jr8ppg/zLog/blob/wkpttfix/zlog/UzLogCW.pas#L172 を追加したため）

JO3JYE, 2021-06-02:
おはようございます。対応ありがとうございます。
tabキーでのキャンセル動作の問題は解消しました。サイドトーンも、ほぼ直ったような気がします。出ないこともあったのですが、条件がまだ見つからず。

shift+zやcrtl+zでCQのループに入った時に、ESCを2度押ししたら、shift+zやctrl+zが反応しなくなります、F1などを押して違うメッセージを出すと復帰します。
多分、FWKabrot　のフラグを使うところが増えているので、ESCキーによるTrueが悪さをしているのかな？と思っています。
TdmZLogKeyer.SendStrLoop　のwk使用時に　winkeyerClear()を追加したらokのような気がします。(軽く試したところ、直った気がする)
他にも、TABキーも利かなくなるケースに遭遇しています。発生条件はまだつかみきれずです。
ここまで見たところで、現地待ち合わせの仕事があるので、今朝は時間切れです。

jr8ppg, 2021-06-03:
了解です。今日はほかに漏れがないか見てみます。

jr8ppg, 2021-06-03:
CQ LOOP開始時はOK
TABキーが効かなくなるのも再現したのでこれでOK（と思う）

JO3JYE, 2021-06-03:
おはようございます。対応ありがとうございました。
CQ　LOOPとTABキーの問題、発生しなくなりました。ともに、昨日のコードでビルドしてESCキーを連打したら再現できました。
サイドトーンの問題も、セットで直ったと思います。
これで、すっきりしました(^^)

jr8ppg, 2021-06-03:
了解です。一旦はこれでOKとしましょう。masterにマージします。
TABキーが利かなくなるのはWinKeyer関係ないですね。前からだったのかなと思います。

closed 2021-06-03 by jr8ppg