jr8ppg/zLog #171: 1200Mにバンド変更した際、リグのバンドが変わらない

jr8ppg, 2021-06-11:
1200Mにバンド変更した際、zLog→リグ方向の制御ができていない。
リグ側で1200MにQSYした場合は、zLog側は追従している。

IC-9700、TS-2000で現象報告あり。

JO3JYE, 2021-06-12:
以下、MLへ投稿されたJJ1CVHさんの解析結果からの予測です。

uRigControl.pasの冒頭の、BaseMHz　の配列の　1200MHzのところが　1200000000,　となっています。（96行目）
上記のところを　1295000000,　等にすればバンド外にならないような気がします。

TRig.SetBand(Q: TQSO)　のところを見ると、バンドごとに記憶している周波数がない場合はBaseMHzの配列から周波数を決めていますね。

こちらに1200MHzが使える実機があればテストしてからの投稿になったのですが・・試しようがないので、取り急ぎ。

jj1cvh, 2021-06-12:
ありがとうございます。
URigControl.pas line 96 を
     1295000000,
に修正してビルドした結果、zLogからIC-910のバンドを1200MHz帯に変更できるようになることを確認しました。

ついでのご相談ですが、1.9MHz帯のBaseMHzは1900000となっていますが、1800000の方が良いように思います。

また、1.9MHz帯や3.5MHz帯から周波数記録が無い他バンドに変更する際に、思わぬ周波数になってしまうことが気になってました。
100kHz台以下を保持する以下のコードが原因だと思いますが、現周波数とBaseMHzの差分に対してシフトするなど、もう1歩賢くなりませんでしょうか？
line 2314
      ff := (_currentfreq[_currentvfo] + _freqoffset) mod 1000000;
line 2317
      f := BaseMHz[Q.Band] + ff;

jr8ppg, 2021-06-12:
そう言うことですか。周波数記録が無い場合のバンド変更での話でしたか。
少しマシにしたいですね。

jj1cvh, 2021-06-12:
1200MHz帯は6m&Downで使う方が多いと思いますので、是非ご検討をお願いします。

BaseMHzの基本は、各バンドの下限周波数だと思います。

1200MHzに限っては、現状の1200000000をどう変更するか悩ましいです。
バンド下限の1260000000、CW帯下限の1294000000、呼び出し周波数の1295000000あたりが候補かと思います。
BaseMHzが下限でない場合を許容すると、
BaseMHz[Q.Band] + ff;
の計算で遷移する先の周波数が、バンド下限より低い周波数になるため、何らかの処理が必要そうですね。

JO3JYE, 2021-06-12:
早速確認していただきまして、ありがとうございました。

100kHz台以下の件。
line2314で　ff　を決めた後の処理で500kHzより大きい場合は、強制的に　０　にするから、バンドエッジに行ってしまいます。かと言って、０以外にしても、SSBかCWかで都合の良い着地点は変わるでしょうし。

今の周波数はどこか?ということを忘れて、周波数の記録がないバンドはモードごとにデフォルトの周波数へ落ちるという考え方はいかがですか?
BaseMHzの配列を増やして、modeごとにデフォルト値を決めてしまう、という案です。
50MHzであったようなコンテストバンドの変更にも対応しやすいかと・・

jr8ppg, 2021-06-12:
今回バンドプランと称してモード毎に上限値下限値をもっていますので、活用しますかね。

jr8ppg, 2021-06-12:
上限値と下限値の中間にするとか。

JO3JYE, 2021-06-12:
バンドプランを利用する案で良いかと思います。

JO3JYE, 2021-06-12:
バンド内で、ランダムなところに着地させる案はいかがでしょうか？

jj1cvh, 2021-06-12:
バンドプランのモード毎の下限周波数に1票

JO3JYE, 2021-06-12:
やはり、一般的な心理だと、バンドプランのモードごとの下限でしょうか。
自分自身、初めてQSYしたバンドでS&Pするならどこから？と聞かれたら「下限から」って答えますし。
実装をシンプルにする点でも、やはり下限でしょうか。
遊び心的にはランダムなんて言いましたが、実装を考えるとちょっと・・・ビミョーな案かもしれません。

jj1cvh, 2021-06-12:
> 500kHzより大きい場合は、強制的に　０

なるほど、そうなってますね。

> また、1.9MHz帯や3.5MHz帯から周波数記録が無い他バンドに変更する際に、思わぬ周波数になってしまうことが気になってました。

と書きましたが、多分逆で、例えば50.300MHzの状態から1.9MHzにzLogでバンドチェンジすると2.2MHzになってしまう、といった動きだった気がします。
周波数記録が無い場合のバンドチェンジのデフォルト周波数ですが、程よいと思われる周波数はモードとバンドで結構まちまちなので、下限周波数に決め打ちするのがシンプルかなと思いました。

(50M SSB) 50350-51000、(50M FM) 51000-52000 は真ん中だとちょっと高すぎるかなと思いますが、モード毎のバンドプランの真ん中にするのも悪くないと思います。

jr8ppg, 2021-06-13:
とりあえず下記の動きで作ってみました。全バンドCWのバンドプランは入っていると言う前提です。
・現在のモードの下限周波数が０以外ならそれをセット
・現在のモードの下限周波数が０の場合はCWの下限周波数をセット
・それも０なら周波数をセットしない

jj1cvh, 2021-06-13:
c36a6fb を32bitでビルドし、zLogからIC-910のバンドを1200MHz帯に変更できることを確認しました。
周波数記録が無いバンドに変更する際の周波数も、モード毎に気持ち良く変更されるようになりました。

ご対応ありがとうございました。

jj1cvh, 2021-06-13:
1.9MHz帯LSBでzLogを起動し、SHIFT+Bで順にたどると、14MHz-50MHz帯もモードがLSBのままでした。
旧版zLog 2.2eは、動きが変なところありつつもUSBに切り替わっていたようです。
多分、下記のコードで実現していたのではないかと思います。
https://github.com/jr8ppg/zLog/blob/78befad3afde161c577d71c99034108078ef29a2/zlog/URigControl.pas#L1722

JO3JYE, 2021-06-14:
予測を試してみました。(テストはPL2303によるUSBシリアル変換＋FT-817を使用)
おっしゃる通り、ssbモードのときだけ、setmodeを呼び出すようにしたら、バンドによるLSB/USBの切り替えが追従するようになりました。

jj1cvh, 2021-06-14:
プログラムの改良、ありがとうございます。

残念ながら、d3614f8 を32bitでビルドしてみましたが、IC-756ProII＋FT232RLでは相変わらず14MHz-50MHz帯もモードがLSBのままでした。

zLog令和のmain.pasに既に似たようなコードがあるのですが、(どういうわけか分かりませんが)それが機能していないことが問題なような気がしますので、d3614f8 は取り消した方が良いかもしれません。
https://github.com/jr8ppg/zLog/blob/3d7ca0c30295ef4c894072aca19acf8fc6993a14/zlog/main.pas#L1876

COMポートの通信データを見る限り、SSBの状態で、SHIFT+BやzLogバンド欄でバンド変更しても、CI-Vコマンド06(モード設定)が発行されている気配がありません。

一方、Delphi上でデバッグで実行した時、
もともとあったmain.pas L1876
         MainForm.RigControl.Rig.SetMode(CurrentQSO);
や、d3614f8 で追加された URigControl.pas L2310
    Self.SetMode(Q);
にブレークポイントを設定すると、バンド変更時にちゃんと引っかかるので、これらの行の実行はされているようです。

jr8ppg, 2021-06-14:
こんばんは。
JYEさんが直したところは元々その処理があったけど、どうしてかコメントアウトされていたところです。
（https://github.com/jr8ppg/zLog/commit/7a574d915db56cfd5e43ae37b75239ace55cdfa3 で消してしまいましたが）
私の方でも調べてみます。

jj1cvh, 2021-06-14:
4c0257c でコメントアウトされたようです。
d3614f8 の処理に加えて、ICOM機特有の追加処理が必要なのかもしれませんね。調査の程、宜しくお願いします。

JO3JYE, 2021-06-14:
ご指摘の件、もともと処理があるのにうまくいかない。何かありそうです。
main.pas L1876　と今回の追加、完全にダブっています。
FT-817でテストした時は、追加した部分の有無で挙動が違っていたのですが・・・
思ったより、根が深そうです。
よろしくお願いします。

jj1cvh, 2021-06-14:
main.pas L1869の TContest.ChangeBand() はSHIFT+Bの処理で、SetBandのあと、SSB判定してSetModeしてます。
2.7.0.2までのFT-817は、SHIFT+BでバンドチェンジするとLSB/USBが切り替わるが、Band欄でバンドチェンジするとLSB/USBが切り替わらない、ということはありませんでしょうか？

もしそうなら、URigControl.pasの d3614f8 の処理を追加した場合、SHIFT+BではSSB判定後のSetModeを2回実行していると想像します。
d3614f8 は活かして、main.pas L1869 のSSB判定を削除すべきかもしれません。

ICOM機は、SHIFT+BでもBand欄でもLSB/USBが切り替わりませんが、URigControl.pas L1869の TICOM.SetMode() を修正する必要がありそうに見えます。
どう修正すれば良いのかは分かりませんので、、いったんお任せします。

jr8ppg, 2021-06-14:
えーとですね。手元のYAESUとKENWOODは https://github.com/jr8ppg/zLog/commit/d3614f8ba7bdd356269e410e0f2772244c399414 の修正が無くても、SHIFT+Bで問題なくLSB,LSB,USB,USBと遷移します。

と言う事はICOM特有の現象と言う事になります。FPollingCount と言う変数がミソで、zLogのトランシーブモードがONの時は一度SetMode()が呼ばれるとFPollingCountが0になり二度と増えることがないので、それ以降はSetMode()が呼ばれても何もしません。
トランシーブモードがOFFの時は動く様な気がします。

jr8ppg, 2021-06-14:
はい、うちでは https://github.com/jr8ppg/zLog/commit/d3614f8ba7bdd356269e410e0f2772244c399414 の修正が無くても動きます。
で、ICOM機ですが無いのですよ。Ｈｉ

jr8ppg, 2021-06-14:
３行コメントにして動かしたら、それっぽい感じでした。２回呼ばれますが。
無線機繋ぐと動きが変わる可能性はありますが。
[image]

jj1cvh, 2021-06-14:
すみません、訂正です。
ポーリング動作であればICOM機でもLSB/USBが正しく切り替わりました。
また、3行コメントアウトすれば、トランシーブモード動作でもLSB/USBが正しく切り替わりました。

モード欄のマウスクリックでLSB/USBが正しく切り替わるYAESU機の機種は何でしょうか？
TFT817.SetMode()と見比べてみたいと思いまして。

jr8ppg, 2021-06-14:
了解です。うちはFT-2000です。

jj1cvh, 2021-06-14:
ICOM機の挙動について、再び訂正です。
d3614f8 の修正を適用した状態では、ポーリング動作もトランシーブモード動作でも、LSB/USBが正しく切り替わりません。
３行コメントすると、ポーリング動作もトランシーブモード動作でもLSB/USBが正しく切り替わります。

ポーリング動作でも、
 if FPollingCount = 0 then begin
のFPollingCountは0になっていました。

jr8ppg, 2021-06-14:
FT817でも動きそうですけどね。

jj1cvh, 2021-06-14:
ICOM機は、 d3614f8 の修正を適用せずとも、URigControl.pas の TICOM.SetMode() を３行コメントすれば、ポーリング動作もトランシーブモード動作でもLSB/USBが正しく切り替わりました。
FT817の現象は不明です。

JO3JYE, 2021-06-14:
おはようございます。
FT-817は d3614f8　の修正がないと、SHIFT+Bでもbandのクリックでも、LSB/USBの切り替えができません。
SHIFT+MやmodeのクリックによるSSB/CWの切り替えだとLSB/USBが正しく変わります。

SHIFT+Bの時は、main.pas L1869の TContest.ChangeBand() も呼ばれていますし、モードのSSB判定もちゃんと反応しています。なのに、  MainForm.RigControl.Rig.SetMode(CurrentQSO);　が正しく動いていない雰囲気です。処理の二度づけみたいになっているはずなんですが、なんでだろう？？

ft817.setmodeですが、ft847.setmodeの派生です。単にモード切替のコマンドを発行し、コマンド発行後のackを受け流すためにwaitを入れて、ポーリング再開です。ということで、こちらは大丈夫かな?と思っています。

jr8ppg, 2021-06-14:
おはようございます。不思議現象ですね。
https://github.com/jr8ppg/zLog/blob/master/zlog/URigControl.pas#L3549 の待つ時間が実は短いとかですかね。
色々検討しましたがそれ位しか思いつきませんね。困ったな～

jr8ppg, 2021-06-14:
ICOM機TODO
https://github.com/jr8ppg/zLog/blob/master/zlog/URigControl.pas#L2006 の意図するところを思いだす。
https://github.com/jr8ppg/zLog/blob/master/zlog/URigControl.pas#L2248 のFPollingCountをクリアするあたりは必要かどうか。
この辺を整理するとOKですね。

jj1cvh, 2021-06-15:
FT-817ですが、URigControl.pasのTFT817.hogeかTFT847.fugaのどこかに問題があると私も思います。
やはり一度 d3614f8 の修正は取り消していただいた方が良いと思います。

SHIFT+Bは、Main.pasのTContest.ChangeBand()で
 MainForm.RigControl.Rig.SetBand(CurrentQSO);
と
 MainForm.RigControl.Rig.SetMode(CurrentQSO);
を実行

Band欄クリックは、Main.pasのTMainForm.BandMenuClick()で
 QSY()
を実行、QSY()から
 RigControl.Rig.SetBand(CurrentQSO)
と
 RigControl.Rig.SetMode(CurrentQSO)
を実行

これらのSetBandとSetModeの間にWaitを入れて改善するなら、jr8ppgさんご指摘のTFT817.SetFreqのsleepを調整するなどして原因を追い込めそうに思います。

jr8ppg, 2021-06-15:
取り消しコミットしました。

JO3JYE, 2021-06-19:
皆様の推察通り、FT-817固有の現象でした。
SetFreq→SetModeとコマンドを連投すると、一定量のwaiがないと後から投げたSetModeが働きませんでした。

TFT817.SetFreq　の最後にあるsleepの時間を調整しています。
当方所有の個体（FT-817　元祖）では　sleep(106)より大きくすると、SSBモード時のLSB/USB自動切替が正しく働くようになりました。+20%ほどの余裕を持たせて、sleep(130)　としています。
TFT817.SetMode　のsleepも、制御コマンド連投時に備えて　100→130　としておきます。(不要かもしれませんが)

同じく所有する、FT-857Dでテストしたところ,sleep(100)でも問題なく動きました。FT-857Dでテストしていると気が付かなかったことでしょう。(857の場合、sleep(60)前後にまで縮めるとLSB/USBの切替が働かない現象が出ます)

jr8ppg, 2021-06-19:
ICOM機TODO
> https://github.com/jr8ppg/zLog/blob/master/zlog/URigControl.pas#L2006 の意図するところを思いだす。

これはFPollingCountがオーバーフローしてマイナス値になった時の処理でFPollingCountがなんぼまでカウントされるか考えると、
48hour=2880min.=172800sec.で75ms毎にポーリングすると2,304,000カウントなので実際は通ること無く十分ですね。
（Integerの最大値は2,147,483,647）

> https://github.com/jr8ppg/zLog/blob/master/zlog/URigControl.pas#L2248 のFPollingCountをクリアするあたりは必要かどうか。

０かどうか判定する必要が無いのでクリアも不要。
TICOM.SetMode()とTICOM.SetFreq()の修正が必要。

jr8ppg, 2021-06-19:
TICOM.SetMode()とTICOM.SetFreq()のfinallyに、FPollingCount := 0; を入れればいい話なのか？

Get band and mode alternately 無効なら関係なさそう
Get band and mode alternately 有効なら、1/2の確立で周波数は読み取れるが、3206行の(FPollingCount < 2)があるので、モード取得が先になったら周波数は読み取らない。

JO3JYE, 2021-06-20:
icom機が手元にないのでコードからの想像ですが…

今の状態だと、FPollingCountを０に戻すコードが削除されているので、FPollingCountはオーバーフローした時に０へ戻す以外は、永遠に増え続けるだけと見ます。
となれば3206行の　　FPollingCount < 2 があると、後半の条件が成立するのは、最初の時だけなので、使用状態だと成立することは事実上起こらないような気がします…。

常に周波数→モードの順で取得させたい、という事だと思いますので、

>TICOM.SetMode()とTICOM.SetFreq()のfinallyに、FPollingCount := 0; を入れればいい話なのか？

も対応の一つかもしれません。

JO3JYE, 2021-06-20:
コードを延々眺めてながらネチネチ迷って投稿したら、PPGさんの修正が先になってしまいました。
ということで、先ほどのはお忘れください。

jr8ppg, 2021-06-20:
トランシーブ時はVFO回すとか操作しないと新しい情報が来ないのですよね。KENWOODは来るのだけど。
仕方ないのでポーリングするしかないわけです。
@jj1cvh さんこれで試してもらえますかね。

jr8ppg, 2021-06-20:
@JO3JYE さん
色々ありがとうございます。
オフバンドすると周波数表示を赤くしたのでやってみて下さい。

jj1cvh, 2021-06-20:
ご対応ありがとうございます。
MLでご報告した、ICOM機のトランシーブモード動作が2.7.0.3でおかしくなっている件について、 f7cfa69 を32bitでビルドし、正常に動作することを確認しました。

(2.7.0.3)
Use Transceieve Mode 有効
Get band and mode alternately 有効
のトランシーブモード動作でSHIFT+BやSHIFT+Mを押すと、リグのバンドやモードはちゃんと1回で切り替わるが、zLogのBand欄やMode欄、Rig Controlウィンドウの表示内容は、1回目では変更が反映されず、もう1回押すと反映される。

Use Transceieve Mode 無効
Get band and mode alternately 有効
のポーリング動作であれば、正常に1回で切り替わり、LSB/USBもバンドに応じて正しく切り替わる。

(f7cfa69)
Use Transceieve Mode 有効
Get band and mode alternately 有効
のトランシーブモード動作でも、
Use Transceieve Mode 無効
Get band and mode alternately 有効
のポーリング動作でも、正常に1回で切り替わり、LSB/USBもバンドに応じて正しく切り替わる。

1200Mのバンド変更のIssue #171から派生した、本件のICOM機のLSB/USBがバンドに応じて正しく切り替わらない現象は、これにてcloseとさせていただきます。

中期的には、
Use Transceieve Mode 有効
Get band and mode alternately 無効
のトランシーブモード動作でも、zLogが周波数やモードを変更した時は、周波数とモードをリグから1回ずつ取得するよう改善をお願いしたいと思います。
- 定期的なポーリング動作
- zLog起動時、zLogで周波数・モードを変更した時、zLogのOptionsウィンドウを閉じた時のポーリング動作
を整理いただく必要があると思いますので、Issue #169, #171 で引き続き議論させていただきたいと思います。

closed 2021-06-20 by jj1cvh

jr8ppg, 2021-06-20:
色々ご確認ありがとうございました。