jr8ppg/zLog #172: 64bit版でリグコン時にアプリケーションエラー

jj1cvh, 2021-06-12:
(動作環境)
- Windows 10 2004 64bit, LIFEBOOK S935/K
- USBシリアルIF(FT232RL)
- IC-910(CI-V Transceive: ON)
- zLog version: 2.7.0.0 64bit, 2.7.0.1 64bit
	- zlog.iniを消去してから以下を設定
	- Rig1: COM5 19200 IC-910/911
	- Use Transceieve Mode: チェック有
	- Get band and mode alternately: チェック無

(現象)
1. Options を開いて[OK]を押すと、下記画像の「Application Error」ウィンドウが連続して2つポップアップ。
[image]
2. リグのVFOを回すと、「Application Error」ウィンドウが1つポップアップした後、下記画像の「アプリケーションエラー」ウィンドウがポップアップし、[OK]を押すとzLogが強制終了される。
[image]

(補足)
- 32bit版は問題無く動作。
- OptionsのRig1: None 300 None の初期設定をCOM5 19200 IC-910/911に変更して[OK]を押した時点ではエラーが発生せず、リグコンも正常に動作。もういちどOptionsを開き[OK]を押すとエラー発生。

jr8ppg, 2021-06-12:
うーん、６４ビットだから影響あるってところが見当たりませんなー
https://github.com/jr8ppg/zLog/blob/master/zlog/URigControl.pas#L3761 で、
TCommPortDriver(Sender).Tagの値が1,2以外になったのかな

jj1cvh, 2021-06-12:
Delphi 10.3を入れているデスクトップPC(Win10 Pro 2004)では問題なく動作してしまい、変数の状態を調査できませんでした。

前述のノートPC(Win10 Pro 2004)では、以下の組み合わせのいずれも問題が発生しました。
何かPC環境に問題があるのかもしれません。
USBシリアル: FT232RL(自作)、CH341(LD-C101)
リグ: IC-910、IC-756ProII

jj1cvh, 2021-06-16:
64bit版で問題が再現する前述のノートPCにDelphi 10.3をインストールしてデバッグ実行したたところ、
   Rig.Xit := fOnOff;
https://github.com/jr8ppg/zLog/blob/ZLOG2702/zlog/URigControl.pas#L3845
の実行時にACCESS_VIOLATIONのデバッガ例外通知が発生することが分かりました。
Delphiのスクリーンショットを添付します。
zLog version: 2.7.0.2 64bit
Win10 Pro 64bit 2004, IC-756ProII, FT232RL
[image]

続くEnd;行(L3846)にブレークポイントを設定し、zlog.iniに Rig1: COM5 19200 IC-756ProII が設定された状態でzLogを起動すると、ブレークポイントを1回通過しますが、その時点ではエラーは発生しません。
zLogが立ち上がった後、Options を開いて[OK]または[Cancel]を押すと、2回目のブレークポイント通過直前にデバッガ例外通知が発生します。

Rig.XitをDelphiの監視式に設定すると、1回目(zLog起動直後)のブレークポイント通過時のRig.Xitの値はFalse(正常)ですが、2回目(Optionsウィンドウを閉じた時)のブレークポイント通過時は
Rig.Xit | E2003 Undeclared identifier: 'Rig'
となっており、識別子'Rig'が壊れてしまってるようです。

なお、同じPC環境で32bitビルドすると、2回目のブレークポイント通過時もRig.Xitの値は正常です。
なぜ32bitと64bitで異なるのか、またPC環境で差異があるのか、全く分からないのですが、もし私が試せるこがありましたらお知らせください。

jr8ppg, 2021-06-16:
たぶんわかりました。
潜在的な不具合で３２ビットはたまたま動いていると思います。
下図のようにTRigControl.Stop()に、FCurrentRig := nil; を入れると改善すると思います。
[image]

どうしてこうなるか
OptionsでOKボタンを押した後、RigControl.ImplementOptions;を呼び出して
RIG関係を再構築しますが、その最中にDoVfoChange()が呼ばれると、
RigControl.Rig(FCurrentRig)の値が参照する先がStop()によってFreeされているので、アクセス違反となります。
３２ビットはたまたまメモリー上に残っていた旧Rigオブジェクトを参照してそれっぽく動くけど
６４ビットはそうではない。と言うことかな～と思います。

きちんとした修正は、Rigオブジェクト構築中はTRig.UpdateStatus;が呼ばれても平気な様にしないと
いけないかなと思います。（上記修正で平気になりそうですが）

ただいま出張先なので机上のみです。違ってたらまた考えます。

jj1cvh, 2021-06-16:
FCurrentRig := nil; で直りました!
Delphi上ですが、32bit/64bit releaseビルドを動作させ、問題が解消することを確認しました。
zlog令和 64bit 2.7.0.2が(プレリリース版ではなく)リリース版として公開されている状況を考えると、各局のPC環境で影響が波及する前に、優先度高めでご対応いただけますと幸いです。

jr8ppg, 2021-06-16:
詳しい説明と確認ありがとうございました。
この手の問題は迷宮入りしがちですので大変助かりました。
６D前にはリリースですね。

jj1cvh, 2021-06-16:
838abf8 を64bitでビルドし、問題の起きていたPCでアプリケーションエラーが解消していることを確認しました。
ご対応ありがとうございました。

closed 2021-06-16 by jj1cvh