jr8ppg/zLog #174: リグコントロール設定でICOM機の設定見直し

jr8ppg, 2021-06-13:
Get band and mode alternately欄がわかりづらいので、設定画面見直し

> トランシーブ動作時はグレーアウト、ポーリング動作時は
> Get freq. and mode (ICOM only, slow) <-- デフォルト
>  Get freq. only (ICOM only,fast)
> のような文言のラジオボタンで選択する方が、意味が分かりやすいかなと思いました。
> alternatelyという文言は、チェックONは交互に取得、チェックOFFは同時に取得、と受け取ってしまいそうです。

jr8ppg, 2021-06-22:
スペースが無いのでコンボボックスで試作。
ClusterやZ-Linkは別タブ（Network等）に出して、場所を作らないと苦しいかな。

jr8ppg, 2021-06-22:
icomoptionブランチです。

jj1cvh, 2021-06-22:
ご対応ありがとうござます。
icomoptionブランチの最新コミット b695d6c で、IC-756ProIIで正常に動作することを確認しました。

Trasceive、Polling + Get freq. and mode (slow)、Polling＋Get freq. only (fast)
の3通りとも、zLog->RIG、Rig->zLogのいずれも問題なく、とても良くなったと思います。
敢えて言えば、「Transceive」は、ICOMセットモードの表記に合わせて「CI-V Transceive」とした方がICOMユーザには分かりやすいかなと思いました。

Transceiveモード時にzLogでモードを変更してもMode欄に反映されない #169 も、正しく反映されるようになりました。
Mode欄が元のモードの表示に一瞬戻る挙動は残ってますが、個人的には、#169 と #174 はcloseしていただいても結構です。

[Hardware]タブについては、見た目も分かりやすくなって何ら違和感はありませんが、[Hardware]タブ内のスペースを増やしたいのでしたら、ICOM onlyの枠を[Rig control]タブに移動する手もあるかと思います。

jr8ppg, 2021-06-22:
早速の確認ありがとうございます。
あまり違和感ない感じでしたら、表記の修正など微調整してFIXします。
スペースがない件は新たに入れるものが出てくるまで先送りします。

jr8ppg, 2021-06-23:
少し微調整しました。これでリグ側と合わせて貰えばわかりやすいですかね。
CI-V Transceive On
CI-V Transceive Off (Polling)

jj1cvh, 2021-06-23:
確認しました。ご対応ありがとうございました。

jr8ppg, 2021-06-24:
masterへマージ完了

closed 2021-06-24 by jr8ppg