jr8ppg/zLog #184: 獲得済みマルチのスポットがバンドスコープで赤(wanted multi)になる

jj1cvh, 2021-07-17:
zlog Version 2.7.0.5

telnet clusterからスポットが届いた際、spcファイルに登録されている局のバンドマップ(Windows > Band Scope)の表示色は
交信済み 黒
未交信局、未獲得マルチ  赤
未交信局、獲得済マルチ 緑
となってほしいのですが、3つ目の
未交信局、獲得済マルチ
が緑ではなく赤になってしまいます。

jj1cvh, 2021-07-17:
旧zlogに添付のzlog.spcの仕様はパワーコード有でしたが、zLog令和はパワーコード無にする必要があるのでしょうか？
https://github.com/jr8ppg/zLog/wiki/Super-Check-(N%EF%BC%8B1)#zlogspc%E3%81%AE%E4%BB%95%E6%A7%98

jr8ppg, 2021-07-17:
プログラムを確認したところ、パワーコードの有無は考慮していませんね。

ここですね。マルチの末尾が0-9以外なら除外して比較するようにすれば良いかと思います。
https://github.com/jr8ppg/zLog/blob/master/zlog/UzLogQSO.pas#L1990

jr8ppg, 2021-07-17:
現象は確認。修正案での改善も確認。
あとは考慮漏れがあるのか無いのか、、、

jr8ppg, 2021-07-17:
rateexブランチで修正しました。

jj1cvh, 2021-07-17:
プログラムの流れをつかめてないのでアレですが、末尾のA-Zを削除してマルチを抽出する方法で、あらゆるコンテストに対応できるのかがちょっと気になります。

Main.pas の
MyContest.MultiForm.ExtractMulti
でマルチを抽出する方法なら、各種コンテスト(組み込み、ユーザ定義とも)に対応できそうな気がしたのですが、いかがでしょうか？

jj1cvh, 2021-07-17:
例えば町村マルチのあるオール山口やオール群馬などは、末尾がアルファベットですね。
- ＪＡＲＬ山口県支部ホームページ [第２０回オール山口コンテスト規約](https://jarl33yamaguchi.jimdofree.com/%E3%81%8A%E7%9F%A5%E3%82%89%E3%81%9B%E3%81%AE%E3%82%B3%E3%83%BC%E3%83%8A%E3%83%BC/%E4%BB%A4%E5%92%8C%EF%BC%93%E5%B9%B4%E5%BA%A6%E3%81%AE%E8%A1%8C%E4%BA%8B/%E7%AC%AC%EF%BC%92%EF%BC%90%E5%9B%9E%E3%82%AA%E3%83%BC%E3%83%AB%E5%B1%B1%E5%8F%A3%E3%82%B3%E3%83%B3%E3%83%86%E3%82%B9%E3%83%88%E8%A6%8F%E7%B4%84/)
- JARL群馬県支部 [オール群馬コンテスト](https://www.jarl.com/gunma/contest2021.html)

jj1cvh, 2021-07-17:
https://github.com/jr8ppg/zLog/blob/8dcd3c4ccac3929f1a824801f8d5765749d38403/zlog/USpotClass.pas#L473
で、Sp.NumberをExtractMultiでマルチに変換してから代入すれば良いのかもしれません。

jr8ppg, 2021-07-18:
そうですね。
ExtractMultiで作ってみました。これでOKと思います。

jj1cvh, 2021-07-18:
ありがとうございます。
パワーコード有のSPCファイルに対し、Field Dayとオール群馬の手持ちログで検証し、期待通りの色分けになることを確認しました。
大変良くなったと思います。ありがとうございました!

これにて本件はクローズさせていただきますが、検証を通じて追加で2点気付いたことがあります。
もし修正可能な内容でしたら、別建てでissueを立てていただけますと幸いです。

1. 2.4Gとそれ以上のバンドでCtrl+Nでバンドマップに登録すると、バンドマップの周波数がおかしな数字になる。
2. ALL ASIAのバンドマップは獲得済マルチ(カントリー)のチェックが行われず、獲得済カントリーも赤になる。

closed 2021-07-18 by jj1cvh