jr8ppg/zLog #185: CWサイドトーンの音量調整機能

jr8ppg, 2021-07-17:
CWサイドトーンの音量調整ができると良い。

jj1cvh, 2021-07-18:
rateexブランチの 2d8d3b9ac をビルドし、正常動作を確認しました。
Phoneの変調にちょうど良いPCスピーカ音量10の状態で、CW/RTTYタブのUse Side ToneのVolumeを100から5に落とすと、期待しているサイドトーン音量になりました。
ご対応ありがとうございました。

jj1cvh, 2021-07-18:
別件ですが、CW/RTTYタブのCQ maxがUse Side Toneの下にあるのは変ですね。CQ rpt. interval (sec)あたりに移動するのが適切に思います。
また、zLogのOptions内でPhoneのCQ maxを定義する場所がないような気がします。

jj1cvh, 2021-07-18:
CW,Phoneがらみでさらに別件ですが、Options > Hardware タブのPTT設定はPhoneでも機能するので、
CW/ PTT port --> PTT port
CW PTT control --> PTT control
Reverse Signal lines --> Reverse DTR(Key) RTS(PTT) to RTS(Key) DTR(PTT)
みたいな感じの方が分かりやすいかな、と思いました。

jr8ppg, 2021-07-18:
音量確認ありがとうございます。

CQ Maxの位置について確認しました。確かにおかしいですね。
修正しました。

> Reverse Signal lines --> Reverse DTR(Key) RTS(PTT) to RTS(Key) DTR(PTT)

これは確かにそうなんですが、長すぎて入らないので苦肉の策って感じです。
（さらに苦肉の策がマウスを上に持って行った際のToolTip表示）
当初とは画面レイアウトも変わっているので、試しに変えてみましたがやはり長いです。

jj1cvh, 2021-07-18:
CQ maxの位置、CW PTT control --> CW/PTT control の修正を確認しました。
ありがとうございました。

jj1cvh, 2021-07-25:
引き続き Version 2.7.0.7 で正常動作を確認しましたので、本件はcloseします。
ご対応ありがとうございました。

JO3JYE, 2021-08-09:
サイドトーンの音量調整を実装してから、再びサウンドデバイスが存在しないＰＣで起動すると、一般保護エラーが出るようになりました。
起動して、メインフォームが立ち上がったタイミングです。再生デバイスを無効にして空にすれば再現できると思います。
済だったのですが、もう一息よろしくお願いします。

(これまた、FDで実家から運用する際、サウンドデバイスのないPCで起動したことで気が付きました。)

jr8ppg, 2021-08-10:
おっとそうでしたか。了解しました。

JO3JYE, 2021-08-11:
早速の対応ありがとうございました。
問題解消を確認しました。

jr8ppg, 2021-08-12:
確認ありがとうございました。

closed 2021-08-12 by jr8ppg