jr8ppg/zLog #190: Z-Serverのバンド表示をオペレータ表示に切り替える機能がほしい

jr8ppg, 2021-07-25:
Z-Serverのバンド表示をオペレータ表示に切り替える機能がほしい

jr8ppg, 2021-07-27:
間違えた #191 でコミットしてしまった

jj1cvh, 2021-07-27:
最新コミット 98f873f をビルドし、期待していた動作になることを確認しました。
従来の[OK]ボタンは不要ですので、[OK]ボタンを削除してBand/PCNAME/OPNAME/CALLの選択ボックスを追加するのは、ナイスアイデアだと思います。

1点、以前のバージョンから生じていますが、惜しい点がありまして、zLogのQSO入力行で10GHz&Upを選択すると、Z-Server Messages画面のIDが
10GMHz>
となります。

実用上は何ら問題ありませんが、一応ご報告まで。

jr8ppg, 2021-07-30:
10Gの表記は昔からここの定義ですね。
色々なところで使われていて長くするとあちこちで影響出そうです。
https://github.com/jr8ppg/zLog/blob/master/zlog/UzlogConst.pas#L72

jj1cvh, 2021-07-30:
いえ、長さの話ではなく、「10GMHz」じゃなくて「10GHz」ですよね、ということです。

jr8ppg, 2021-07-30:
おっと、それは気がつきませんでした。
格好悪いですね。

jj1cvh, 2021-07-30:
ちなみに、Z-Server 2.2fの画面上でも「10GMHz」と表示されます...

jr8ppg, 2021-07-30:
まあ、同じ仕組みですからね、、、
MHz取って良いですかね？

jj1cvh, 2021-07-30:
「10G」でも良いですよ。
Z-Server画面の「Current Frequencies」ウィンドウは、「10G」と表示しますね。

jj1cvh, 2021-07-30:
最新コミット 22fd181 をビルドし、期待していた動作になることを確認しました。
プロンプトの文字数を少なくして6文字＋>でも良いかなとは思いますが、これでもOKです。
ご対応ありがとうございました。

closed 2021-07-30 by jj1cvh