jr8ppg/zLog #191: Z-ServerのChat内容の改良

jr8ppg, 2021-07-25:
Z-ServerのChat内容をzLogクライアント側で保存する or コピーする機能が欲しい

jr8ppg, 2021-07-27:
こんな感じで良いのではないかな

jr8ppg, 2021-07-27:
https://github.com/jr8ppg/zLog/commit/f796e2b0c38d7d9158f584cd4e6ec194204d07ce は #190 の機能

jj1cvh, 2021-07-27:
最新コミット 98f873f をビルドし、正常動作を確認しました。

その前のコミット 47cb4f3 でzLogを再起動すると、同一日のzlog_chat_yyyyMMdd.txtの内容がZ-Server Messages画面に表示されるため、Z-Serverに既に接続できているのかと一瞬勘違いしてしまう件については、未接続時にZ-Server Messagesの入力欄をグレーアウト(入力不可)とする追加対応をしていただくことで、大変分かりやすくなりました。

ご対応ありがとうございました。

jj1cvh, 2021-07-28:
Z-Server 1.2fに接続して試したところ、Z-Server画面にチャットが表示されず、他のzlogクライアントにも中継されてないようです。

jr8ppg, 2021-07-28:
Z-Server 1.2fで試してみましたが、大丈夫そうですけど、、、

jj1cvh, 2021-07-28:
Z-Server Messagesウィンドウの動き #189 
に本来書くべきところ、間違って今朝こちらに書いてしまったので、引き続きこちらで。

今朝はおかしな動作だったのですが、今改めて試すと正常です。
今朝と同じZServer.exeとzlog.exe(98f873f 32bit)を使っているのに、変ですね。
また再現したらご報告します。

jj1cvh, 2021-07-30:
Z-Serverに中継されない現象は、その後再現しないため、いったんクローズします。
ご対応ありがとうございました。

closed 2021-07-30 by jj1cvh