jr8ppg/zLog #194: リグコンで指定したリグに対応するバンドがないコンテストでSHIFT+Bを押すとzLogが固まる

jj1cvh, 2021-07-28:
zlog Version 2.7.0.7, 32bit および 64bit, Win10 2004

(再現方法)
- ALL JAを指定してzLogを起動
- Options > Hardware で、Rig1 のRigにIC-910/911を指定
- SHIFT+Bを押すと、zLogが固まって応答しなくなる
-- CPU負荷が50%前後になり、zLogタイトルバーに「応答無し」と表示される

(補足)
- リグを持っていなくても(COMポートに何も接続しなくても)再現します。
- Delphi上のDebugビルドでも再現します。
- BAND欄をクリックしてバンドを変更した場合は、固まりません。

jr8ppg, 2021-07-29:
現象確認。無限ループ。
例えばALL JAでは1.9-50Mしか使用しないのに、当該RIGは144-1200Mまでなので
有効なバンドが見つからず無限ループする。

jr8ppg, 2021-07-29:
ACAGだと問題ない様だ。

jr8ppg, 2021-07-29:
コンテストで使用するバンドを優先することにする。RIGが対応していないバンドに対するコマンドは出ないので問題ないでしょう。

jr8ppg, 2021-07-30:
実機で試してみると、RIGが対応していないバンドに対するコマンドは出ないけど、ポーリングした瞬間RIGのバンドに変わってしまうのでいまいち。やはりRIGが対応しているバンド内でトグルした方がマシ。
コンテストが必要とするバンドとRIGのバンドが交わらないケースはどうしたものか。

jj1cvh, 2021-07-30:
まだ試せてないのですが、FD以降の課題としていったん戻していただき、別branchで検討する、というのでいかがでしょうか？

jj1cvh, 2021-07-30:
運用中でzLogが固まるエラーは困りものなので対処したいところではありますが、コンテストに対応するBANDが無いリグをリグコンするパタンは、今まで発覚しなかったことからしてレアケースかなと思います。
- 前回起動時のリグコン設定がzlog.iniに残ったまま、意図せずzLogを起動した場合
- トラバタを使う場合 (これはこれで検証が必要)
あと何かありますかね。

jr8ppg, 2021-07-30:
あとは
・単純に間違った
くらいかと思います。
無限ループは避けたいですね。思案中です。

jr8ppg, 2021-07-30:
思い切って大幅修正しました。これでOKと思います。
コンテストが必要とするバンドとRIGのバンドが交わらない場合は、SHIFT+Bを押してもバンドは変わりません。

jj1cvh, 2021-07-30:
最新コミット 22fd181 をビルドし、期待していた動作になることを確認しました。

試しに10Gトラバタ利用を想定して、Rig1のXVTをチェックして周波数オフセットを指定してIC-910(1200MHz)をリグコンすると、色々とおかしな挙動があって使い物になりませんでしたが、無限ループに陥る現象は解消していますので本件はクローズさせていただきます。
ご対応ありがとうございました。

closed 2021-07-30 by jj1cvh