jr8ppg/zLog #206: Shift+Mを押してもモードがCWのまま

jj1cvh, 2021-09-19:
Version 2.7.1.1

(再現方法)
 zlog.iniを削除
 全市全郡、ALL Band、PH/CWを指定してzLogを起動
 Band 1.9Mの状態でSHIFT+Mを押しても、モードがCWのまま
 他バンドに変更しても同様にCWのまま

jr8ppg, 2021-09-19:
現象確認
原因は[これ](https://github.com/jr8ppg/zLog/commit/d184da6431769655be2eab923f677bb0c3b7e326#diff-0ce98c0a77074b070d4d34fd6db4f98cd4df5776100931b6f2f4d495c0106e26L1861)ですが、何でこうなったのだろう。
周辺の修正も無いし記憶にも無いので謎。

jj1cvh, 2021-09-19:
2f9726d をビルドし、正常に動作することを確認しました。

修正箇所を見ても何が悪かったのか理解できてませんが、、ご対応ありがとうございました。

jj1cvh, 2021-09-19:
Issue #201 の d184da6 で、どういうわけか[L1861](https://github.com/jr8ppg/zLog/commit/d184da6431769655be2eab923f677bb0c3b7e326#diff-0ce98c0a77074b070d4d34fd6db4f98cd4df5776100931b6f2f4d495c0106e26L1861)の
aQSO.Mode := TMode(Integer(aQSO.Mode) + 1);
を
aQSO.Mode := TMode(Integer(aQSO.Mode));
に変更してしまっていた、ということですね。理解しました。

closed 2021-09-19 by jj1cvh