jr8ppg/zLog #212: Band Map画面をマウスのホイールを回したらすぐスクロールしてほしい

jr8ppg, 2021-09-23:
スクロールに関しては、zLogの縦スクロールバーが普通のアプリと異なっている感触で、つまみ(ノブ)の長さが、縦方向のウィンドウ高さから期待する長さよりもなぜか短くて、ホイールでノブをだいぶずらさないとスクロールが始まらないことに起因しています。

jj1cvh, 2021-09-23:
http://mrxray.on.coocan.jp/Delphi/plSamples/071_TDBGrid_ScrollBar.htm#05
> マウスホイールや [Page Up], [Page Down] キーによるスクロールは，選択レコードは移動しますが，選択レコードが上端または下端になるまでは表示内容はスクロールを開始しません．

元々のリクエストは「Band Map画面をマウスのホイールを回したらすぐスクロールしてほしい」というものでしたが、zLogメインウィンドウもまさにこんな感じのホイール操作になりますね。
ホイール操作はつまみ(ノブ)を上下させる操作ではなく、選択レコードの移動操作なので、つまみ(ノブ)の長さが云々というのは関係なかったですね。

それなりのQSO数をロギングしている場合、最終QSOのあとに空のQSO行が22行くらいあるため、最終QSO行を見つけるのにしばらく迷子になる現象も、併せて改善方法をご検討いただけますと幸いです。

jr8ppg, 2021-09-23:
BandScopeとメイン画面では事情が異なっています。
とりあえず現状の説明です。

・BandScope　ホイールをクルクルするとリスト内選択位置（点線表示）が動いて、端まで来るとスクロールします。
・メイン画面　最初（１局目～スクロールが発生するまで）だけ、データの無い行が空欄になって見苦しいのでダミー行を追加しています。