jr8ppg/zLog #255: Cabrillo出力の改善

jr8ppg, 2021-11-29:
M/M種目以外はTXNRを0/1とする。
SingleOPは0固定とする。

jr8ppg, 2021-11-30:
Cabrillo作成側は修正した。次は設定側。

jr8ppg, 2021-12-01:
設定側もできた。
TXNOは
M/Mは0-9，M/SとM/2は0-1のみ設定可能とした

jj1cvh, 2021-12-02:
- 令和2.7.1.4の挙動について
既にお気づきかもしれませんが、M/S選択時は本来、zLog起動画面のTx#ではなく、「,RUN」と「,MULTI」コマンドでRunning StationとMulti Stationを切り替えることで、Tx#の代わりにRとMがzLog内でそれぞれ記録され、Cabrillo出力時はそれぞれ0と1が出力されるのがあるべき姿だと思います。
zLog起動画面でRunning Station用PCにTx#0、Multi Station用PCにTx#1を設定する人向けに、0-1のみ設定可能とするのもアリだと思います。ただ、M/SでzLogを起動した直後のタイトルバーに「Running Station」「Multi Station」の表示がされないのは、zlog.iniの[Categories]のMultiStnが反映されないという初期化の不具合な気もします。
また、zLog2.2eや令和2.7.1.4では、zLog内で記録したRとMが、Cabrillo出力時に1と2に変換されてしまってましたので、0と1に変換するようにこの機会に直していただけるとよろしいのではないかと思います。
(参考)
https://zlog.org/zlogwin_v22.pdf
 p.32 M/S もしくは2TX の場合の特記事項 
https://www.cqww.com/rules_faq.htm#MS
 このFAQの中に、RUN SIGNALは0、MULTIPLIER SIGNALは1とすることが書かれています。

- v2800ブランチの最新コミット 427aa46 の挙動について
Multi-Op/Single-Tx 選択時の「,RUN」「,MULTI」コマンドの挙動がおかしくなっているように思います。
具体的には、タイトルバーの表示と、zLog内のOp欄に記録されるR,Mの表示がおかしいです。
(例) zlog.iniを削除した状態で、CQWW,Multi-Op/Single-Tx,Tx#0 を設定し、zLogを起動した場合
「,RUN」コマンド実行 --- タイトルバーが[Tx#1] Running Stationになり、QSOを入力するとOp欄にMが記録される
「,MULTI」コマンド実行 --- タイトルバーが[Tx#2] Multi Stationになり、QSOを入力するとOp欄に2が記録される

jr8ppg, 2021-12-03:
いつも確認ありがとうございます。
M/Sで1/2になるのは私の方でも確認していまして、v2800で一部直したのですがまだ直すところがありますね。

> 「,RUN」コマンド実行 --- タイトルバーが[Tx#1] Running Stationになり、QSOを入力するとOp欄にMが記録される
> 「,MULTI」コマンド実行 --- タイトルバーが[Tx#2] Multi Stationになり、QSOを入力するとOp欄に2が記録される

例えば上記の場合、1/2で見ているところを0/1にしてしまったので動作が逆になったみたいです。

関連して #235  も直してしまおうと考えています。
またしばらくお付き合いください。

jr8ppg, 2021-12-03:
M/M時
TXNRは0-9まで設定可能（足りない？）

M/S時
Multi StationかどうかはTXNR=1かどうかで判定
Running Station/Multi Stationの表示はM/Sのみとする

M/2時
,RUN1、,RUN2コマンド追加
,TXNRコマンドでも0 or 1しか設定できない

jj1cvh, 2022-02-23:
2.8.0.5でタイトルバーの表示を確認したのですが、zLog起動直後と、Options画面で[OK]を押した後で、表示内容が異なっています。

zlog起動直後は
zLog for Windows
となっていますが、Options画面をいったん開いてそのまま[OK]を押した時と同じ表示になるようにして頂いた方が良いと思います。

また、Options画面で[OK]を押した後の表示ですが、タイトルバーの文字列の先頭は常に「zLog for Windows」としていただき、そのあとに追加の文字を入れていただけませんでしょうか？

理由は、UWSC等を使ってzLogを他のソフトと連携させる際に、zLog入力画面を特定するためにタイトルバーの先頭の文字列で一致判定する場合があるためです。

M/M, M/2
 (2.8.0.5) [TX#0] zLog for Windows -
 (修正案)  zLog for Windows - [TX#0] -

M/S
 (2.8.0.5) [TX#0] zLog for Windows - Running Station -
 (修正案)  zLog for Windows - [TX#0] Running Station -

jr8ppg, 2022-02-24:
おはようございます。

> zlog起動直後は
> zLog for Windows
> となっていますが、Options画面をいったん開いてそのまま[OK]を押した時と同じ表示になるようにして頂いた方が良いと思います。

そうですね。これは直しましょう。

> また、Options画面で[OK]を押した後の表示ですが、タイトルバーの文字列の先頭は常に「zLog for Windows」としていただき、そのあとに追加の文字を入れていただけませんでしょうか？

UWSCって何でしょうかね。
目的の文字列がタイトルバーの先頭から入っていないとだめな感じですね。

zLogを二つ起動した際に、どっちのウインドウか直ぐに判別できるようにとのことで先頭に入れています。
ちょっと考えたいと思います。

jr8ppg, 2022-02-24:
> zlog起動直後は
> zLog for Windows
> となっていますが、Options画面をいったん開いてそのまま[OK]を押した時と同じ表示になるようにして頂いた方が良いと思います。

起動すると、コンテスト選択＞ファイル選択と進みますが、ファイル選択しない（キャンセル）場合、ご指摘の状況になります。

jj1cvh, 2022-02-24:
起動直後のタイトルバーが「zLog for Windows」になる件は、ご指摘の通りでして、zloファイルを選択せずにCancelする操作を行っておりました。

また、masterブランチの fba69b8 にて、上記の対応と、タイトルバーの文字列の先頭を「zLog for Windows」とする対応をしていただきましてありがとうございます。正常動作を確認しました。

実は、UWSCでは部分文字列で一致判定しており、
 [TX#0] zLog for Windows - Running Station -
のような場合も特に問題無いことが分かりましたので、他の方から要望があれば[TX#0]が先頭にあっても大丈夫です(今さらスミマセン)。

以下は余談になります。
UWSC
https://www.vector.co.jp/soft/winnt/util/se115105.html
はフリーの自動化ツールで、N1MM+で使用している方はたまにネットで見かけますが、zLogで使用している方が他にいらっしゃるかどうかは分かりません。

私の場合、マルチオペのリモート運用で、ローカルPC上のzLogでロギング操作、ローカルPC上のRS-BA1で遠隔リグの送受信操作をする場合に、zLogとRS-BA1の画面を行ったり来たりするのがあまりに大変なので、UWSCの自作スクリプトで連携させ、zLog画面上でF1やTABや;キーを押すと、RS-BA1からCQや相手CALL+NRやTU+自局CALLを送信し、zLog画面に戻ってくるよう工夫していた時期がありました。
 GETID("zLog for Windows", "TMainForm")
でzLogのウィンドウのIDを取得するのですが、解説を確認したところタイトルは一部分だけでOKでした。
http://uwsc.s1007.xrea.com/script/function/getid/

closed 2022-04-11 by jr8ppg