jr8ppg/zLog #265: 2Radio対応、入力欄を左右配置

jr8ppg, 2021-12-28:
まずは、入力欄をR1/R2で左右配置して操作ミスを少なくしたい。

jr8ppg, 2022-01-12:
それっぽくなってきた

JO3JYE, 2022-01-16:
改良作業お疲れ様です。
SO2RモードでのR1,R2,R3,TRはokです。ありがとうございました。

SO1Rの従来バージョンの表示で、R1,R2まで正常ですが、R3を送るとRIG3にはなりますが、それ以降の入力を受け付けなくなります。TRコマンドでもrig1,2,3と切り替わり、RIG3のあとが同じ状態です。(一般保護エラーが出る)
main.pasのTMainForm.EditKeyPress　内4160行目のifのところでしょうか。

jr8ppg, 2022-01-16:
確認ありがとうございます。
ご指摘の通りです。SO1R時はRIG3を除外しないといけませんね。忘れてました。

jr8ppg, 2022-01-16:
残作業
RIG3のON/OFF機能

JO3JYE, 2022-01-16:
対応ありがとうございました。
SO2R時のクリック動作、SO1Rへ戻したときのRig3での一般保護エラーは出なくなりました、のでokです。

SO1R時のmaxrig指定が２で固定され、変更できなくなっている感じですが・・
maxrigですが、リグコントロールできるRIG数＋１でよさそう。
４以上の出番はあるのでしょうか?というのがふと疑問になってきました。

jr8ppg, 2022-01-17:
そうなんですよ。４以上の出番は無いです。
色々考えた結果、リグコントロールできる２台以降はRIG3の１台分で良いのです。なのでSO2RではMAXRIG=3固定。
SO1Rの場合はMAXRIG=2固定。リグコンできない２台目なら設定しなければいいし、３台目がいるならSO2R画面を使う。
なんかこれで良いのではないかと思っています。
従来画面はSO1Rと言うより、MOP(M/S,M/M,M/2)用なのかもしれません。
対応できないのは３台４台をリグコンしたいと言ったケースですね。これはSO2Rとは違うニーズであると思います。

JO3JYE, 2022-01-17:
最初はmaxrigが変えられない点が引っ掛かりましたけど、冷静に考えると、２か３の二択でした。
という事で、おっしゃる通り、パターン分類していくと、ご提案通りですね。
maxrigが4以上はまた別の相談でしょうか・・

jr8ppg, 2022-01-26:
ショートカットキー影響範囲
#48 CTRL+K
#61 ALT+M
#69 ALT+W
#101 PSE QSL
#102 NO QSL
#106 QUICK MEMO3
#107 QUICK MEMO4
#108 QUICK MEMO5

JO3JYE, 2022-01-26:
対応ありがとうございます。
CTRL+K、ALT+M、ALT+Wは大丈夫です、確認できました。
101～107のショートカットデフォルトはどこに割り当てていましたか?失念してしまいまして(^^;;

RIG3を選択した状態で、optionを開いてSO2Rを解除すると、一般保護エラーが出ます。
RIG1に戻るという、実装漏れかと思います。

よろしくお願いします。

jr8ppg, 2022-01-26:
了解！です。

JO3JYE, 2022-01-27:
RIG3の件、okです。
ありがとうございました

jr8ppg, 2022-01-27:
とりあえずRIG3でCWもできるようになったと思います。もう少し検証が必要ですが。

JO3JYE, 2022-02-03:
こちらでも、対策前に再現を確認し、今は大丈夫なことを確認しました。

JO3JYE, 2022-02-08:
CQ Repeatのbreak、ＯＫです。ありがとうございました。