jr8ppg/zLog #267: 2Radio対応、RIG1とRIG2で交互にCQする機能

jr8ppg, 2021-12-30:
RIG1とRIG2で交互にCQする機能が欲しい

jr8ppg, 2022-01-22:
とりあえずWinKeyerで動作確認

JO3JYE, 2022-02-28:
手直しありがとうございます。

オーこれで出来た!　と思っていましたが一点だけ。
rig3のチェックが入ったままだと,Rig1-2の交互CQができない。
rig3のチェックを抜くとrig1-2になるので、こうするしかないのでしょうか‥

rig3にある、1,2のラジオボタン、今はどちらかがonですが、両方offも認めたほうが良いのかもです。
ONがあれば、ONしたものとrig3。両方offならrig1-2の交互CQで、すべての組み合わせが出来そうです。

よろしくお願いします

jr8ppg, 2022-02-28:
確認ありがとうございます。
そうですね、チェックボックスにしてみますか。
あと、それをトグルするショートカットも用意する予定です。

JO3JYE, 2022-03-01:
動作を見ていて気が付いたのですが、現状でもRIG1、RIG2のラジオボタンをうまく操作すると、RIG3→RIG1→RIG2→RIG3・・
という3つでの交互じゃない、順繰りのCQができます。
チェックボックスに変えた際、RIG1とRIG2の同時押しを許可しておくと面白いかもしれません。
ヘッドホン等をzlogと連動して切り替えるハードを用意するのが前提になりますが・・
個人的には面白いと思います。
ご検討いただけると幸いです。

jr8ppg, 2022-03-01:
おっと、コメント見逃していました。
両方ONやってみますか。

JO3JYE, 2022-03-02:
ＯＫです。
対応ありがとうございました！

jr8ppg, 2022-04-20:
今後は #285 へ

closed 2022-04-20 by jr8ppg