jr8ppg/zLog #272: 2Radio対応、SO2R Neo対応

jr8ppg, 2022-01-05:
JH5GHM局のSO2R Neoへの対応
・リグ切り替えコマンド
・音声切り替え（ブレンド）コマンド

jr8ppg, 2022-01-08:
実機確認メモ
・最初から9600bps
・CH340 http://www.wch-ic.com/downloads/CH341SER_EXE.html

jr8ppg, 2022-01-08:
echo backでの次の文字送信OK

jr8ppg, 2022-01-15:
残作業はPTT制御に合わせてRXの切り替えだけど、PTT制御をWinKeyerに任せているのでどうしたものか。
#138 参照

serial echo backを使って、１文字ずつ電文送信完了を確認してPTT OFFとすれば可能か。

kondou, 2022-01-15:
keyerにWinkeyerUSBを使用してもNeo内蔵のものを使用していても、いずれの場合もecho backあるいはESCをtapしたという情報を使用するしかないと思います。CWではこの方式でよくても他のモードは別の情報を使用することになるかと思いますが。

参考になるかどうかわかりませんが、SLだとモードに関わらずメッセージ(≠個々の文字)が終了した際にメッセージ送出が終了したというメソッドを呼んでいてその中でAFを切り替えるように実装しています。

jr8ppg, 2022-01-16:
コメントありがとうございます。
やはりecho backで対応するしか無いですね。
やってみます。

jr8ppg, 2022-01-17:
CW Keyboadも

kondou, 2022-01-17:
CW Keyboardはキーボードでkeyingのことを指しているかと思います。ロギングソフトによりけりかと思いますが、送電中の文字が目視できない(表示されない)場合、サイドトーンが聞こえなければtypoをしたかどうかが不明瞭になります。このあたりを考慮した実装にするのがよいのではないかと思います。

ちなみにSLでは表示が何もされないのでtypoに気づけないことがあります。なのでこの場合のAFは送信中のリグを選択するようにしています。

jr8ppg, 2022-01-18:
なるほど、了解しました。
キーボードを打ちながら反対のRIGの音を聞く方はあまりいないと思いますので、とりあえずは送信中のRIGすることにします。

jr8ppg, 2022-01-18:
CW Keyboadできた。次はワンタイムキャンセル機能

jr8ppg, 2022-03-08:
Auto RX Select=ONの際に、入力欄の切替をどうするか
やはりキャレット位置はRX側であるべきか
Inv.CQと整合を取りたい。