jr8ppg/zLog #276: Z-Server接続時、他のPCで修正したQSOがある場合にMergeを行うとアプリケーションエラーになる

jr8ppg, 2022-02-23:
Z-Server接続時、他のPCで修正したQSOがある場合にMergeを行うとアプリケーションエラーになる

再現手順
(1) PC1とPC2でzLogを起動し、Z-Serverに接続
　  ※ 同一PC内の別フォルダのzLogでも大丈夫です
(2) PC1で新規QSOを1行入力
(3) PC2をZ-Serverから切断 (Network > Disconnect Z-Server)
(4) PC1で既存QSO行をダブルクリックし、QSOレコードを修正(例：599->589)して[OK]で確定
(5) PC2をZ-Serverに再接続 (Network > Connect to Z-Server)し、Mergeを選択
(6) PC2でアプリケーションエラー画面がポップアップ

jr8ppg, 2022-02-23:
現象確認

jr8ppg, 2022-02-23:
修正のあったQSOのを指すQSOIDのメモリー二重解放

jj1cvh, 2022-02-23:
早速ご対応いただきありがとうございます。
masterブランチの b119143 にて、「無効なポインタ操作」エラーが発生しなくなったことを確認しました。

ちょっと思っていた動作と異なる点がありまして、PC2切断中にPC1で修正したQSOが、再接続したPC2側ではdupeとして表示されます。
PC2は、PC1と同様な表示(修正後の1QSOのみ)になることを想定していたのですが、修正前のQSOもPC2側には残しておくべきでしょうか？
いろんなケースを考えないといけないかもしれないので、merge時の挙動はどうなるのが正解なのか、ちょっと自信がないです。

以下は、PC1とPC2のスクリーンショットです。PC2切断中に、PC1でCWをSSBに修正した例です。
(画像上) PC1
(画像下) PC2
[image]

jr8ppg, 2022-02-23:
おはようございます。
ご指摘の現象は私の方でも気になっていたのですが、昨日は時間切れでした。
少なくとも後の交信（SSBの方）がDUPEってのはいまいちですね。引き続き調べます。

jr8ppg, 2022-02-24:
現象確認
QSOのMerge処理はおそらく開発途中らしくZ-Server側が新しい場合の処理は動かないものでした。
Z-Serverも手直しを行って、Z-Server側が新しい場合はQSOを更新するように修正。

jr8ppg, 2022-02-24:
Z-Serverの https://github.com/jr8ppg/zServer/commit/be0a5008515fd5bf4341067d8653da01407c60a4 も必要です。

jj1cvh, 2022-03-03:
3/2に公開されたzLog Version 2.8 beta 2(2.8.0.6)とZ-Server Version 2.8.0.1で試してみました。
PC2切断中にPC1で修正したQSOが、再接続したPC2側でMerge操作をすると正常に反映されることを確認しました。
しかしながら、修正を行っていない普通のログが、Merge操作でzLogに反映されない状態になっているようです。

再現手順
(1) PC0とPC1でzLogを起動
　 ※ 同一PC内の別フォルダのzLogでも大丈夫です
(2) PC0をZ-Serverに接続(Network > Connect to Z-Server)し、PC0で新規QSOを1行入力
(3) PC1をZ-Serverに接続(Network > Connect to Z-Server)し、Mergeを選択
(4) PC1にログが反映されない

なお、(3)でDownloadを選択した場合は、PC0のログがPC1に正常に反映されます。
また、Z-Server 2.7.1.2であれば、Mergeを選択してもPC0のログがPC1に正常に反映されます。

jr8ppg, 2022-03-04:
現象確認

jr8ppg, 2022-03-04:
これで良い感じかと思いますがどうかな

jj1cvh, 2022-03-04:
zLog 8b64a1b と Z-Server Version 2.8.0.1 にて正常動作を確認しました。ありがとうございました。

jr8ppg, 2022-03-04:
確認ありがとうございます。