jr8ppg/zLog #279: Z-Server接続時、QSO削除が遅い

jr8ppg, 2022-03-09:
Z-Server側の変更に対応した修正

jr8ppg, 2022-03-10:
https://github.com/jr8ppg/zServer/commit/6b5a0b5bebfff994a1d50e345dcf21bca46063e1 が必要

jj1cvh, 2022-03-11:
Z-Server Version 2.8.0.2とzLog f4df4ed6 で試したところ、PC1のzLogで複数QSOを削除しても、PC2のzLog上ではQSOが残ったままになりました。1つのQSOを削除した場合は正常です。

jr8ppg, 2022-03-11:
確認ありがとうございます。
現象確認しました。

jr8ppg, 2022-03-11:
Z-Serverに新しくコマンドを追加した際は、zLog側にも実装が必要

jr8ppg, 2022-03-11:
これでOK

jj1cvh, 2022-03-11:
Z-Server Version 2.8.0.2とzLog 2c4a754c で試しました。
PC1のzLogで複数QSOを削除すると、PC2のzLog上もQSOが削除されるようになりましたが、PC2のzLogの負荷が上がり、2000QSOの場合はPC2のQSO削除まで3分くらいかかる感じです。
PC1とZ-Serverの削除はすぐに終わります。

jr8ppg, 2022-03-11:
なるほど。
zLog側は時間かかりそうなところ見当たらなかったのですが
何かあるようですね。ちょっと考えます。

jr8ppg, 2022-03-12:
zLog側はDUPEチェックが時間かかっていました。

jj1cvh, 2022-03-12:
Z-Server Version 2.8.0.2とzLog 1d55990 にて正常動作を確認しました。
PC1側の約2000QSOの削除が、PC2側に約5秒で反映されるようになりました。
ご対応、ありがとうございました。

jr8ppg, 2022-03-14:
確認ありがとうございます。

closed 2022-04-11 by jr8ppg