jr8ppg/zLog #310: 相対パスでPath / Super Check フォルダを指定したい

jj8dan, 2022-06-15:
Pathの各フォルダを相対パスで記載したいです。
USBメモリスティックで各マシンにzLogフォルダを配布した際に
毎回Pathを書き換えるのが大変だからです。

例えばD:&yen;HAM&yen;zLog&yen;zLog.exe とした場合にCFG/DAT を
CFG
.&yen;CFG&yen;
D:&yen;HAM&yen;zLog&yen;CFG
が同じフォルダを参照するように動作したい。
(環境変数も使えると自由度が増します %userprofile% など)

【現状】各フォルダを、CFGs, LOGs, Backups, Sound, Plugin, Superchecksとして

CFG/DAT : CFGsと書くと見に行っている
Logs : LOGsと書いても見に行かない（最後に開いたログフォルダが開く）
Backup : Backupと書くとアプリケーションエラーが発生する(書けないので)
┗.&yen;Backup&yen;と書けばBackupフォルダを作成してそこにバックアップファイルを作成する
Sound : 未確認
Plugins : 最初からフルパスが書いてある！
Super Check : Superchecksと書いてそれが存在していれば見に行く

【蛇足】
Path指定の5項目は逆スラッシュが&yen;で、Super Checkは\\なのもちょっと気になります

jj8dan, 2022-06-16:
【現状】というのはzlogwin_28010_x86_20220411 環境という意味です（情報が足りなかった）。

jr8ppg, 2022-11-13:
下記の件は、このIssueの修正の中で一緒に直す。
https://github.com/nextzlog/todo/issues/106

jr8ppg, 2022-11-13:
従来設定からの移行をどうやってやるか

jr8ppg, 2022-11-13:
まだ途中

jr8ppg, 2022-11-15:
これで良さそう。OptionsでOKクリック時に各種フォルダも作成する。
プラグインのDLLがpluginsフォルダにダウンロードされないのも修正した。

jr8ppg, 2022-11-15:
環境変数か

jr8ppg, 2022-11-15:
環境変数に対応

jr8ppg, 2022-12-16:
ルートフォルダを起動フォルダとしたいとのこと。%ZLOG_ROOT%とかにしますか。

jr8ppg, 2022-12-19:
@jj8dan 
仕様案
各種ファイルへのアクセスパスは、絶対パスならその値、相対パスならzLogルートフォルダ＋サブフォルダの組みとする。
絶対パスとして扱うのは、"C:\","D:\","E:\"などで始まる場合又は"\\"で始まる場合で、それ以外は相対パスとして扱う。
zLogルートフォルダが空の場合、又は「%ZLOG_ROOT%」の場合は、zLog.exeのフォルダを指すものとする。
また、zLogルートフォルダは絶対パスのみで相対パスは指定しない。

---

例１）絶対指定
```
zLofルート：空欄
CFG/DAT　 ：d:\zlog\cfgdat
log　　 　：d:\zlog\logs
backup  　：d:\zlog\backups
voice　 　：d:\zlog\voice
plugin  　：d:\zlog\plugin
spc　　 　：d:\zlog\spc
```

例２）相対指定１
zLogルートが空欄又は”%ZLOG_ROOT%”の場合は、zlog.exeの場所を指します。
```
zlog.exeの所在地：d:\zlog
zLofルート：空欄 or %ZLOG_ROOT%
CFG/DAT　 ：cfgdat
log　　 　：logs
backup  　：backups
voice　 　：voice
plugin  　：plugin
spc　　 　：spc
```

例３）相対指定２
```
zlog.exeの所在地：d:\zlog
zLofルート：d:\zlog
CFG/DAT　 ：cfgdat
log　　 　：logs
backup  　：backups
voice　 　：voice
plugin  　：plugin
spc　　 　：spc

```
上記、例１）例２）例３）は全て下記のフォルダをアクセスします。
```
CFG/DAT　 ：d:\zlog\cfgdat
log　　 　：d:\zlog\logs
backup  　：d:\zlog\backups
voice　 　：d:\zlog\voice
plugin  　：d:\zlog\plugin
spc　　 　：d:\zlog\spc
```
他のPCに丸ごとコピーしてそのまま動くのは例２）です。

---

例４）相対指定３
zlog.exeとは別のフォルダ使用する場合。
```
zlog.exeの所在地：d:\zlog
zLofルート：%USERPROFILE%\zlog     <---環境変数参照
CFG/DAT　 ：cfgdat
log　　 　：logs
backup  　：backups
voice　 　：voice
plugin  　：plugin
spc　　 　：spc
```
↓下記の様に解釈
```
環境変数USERPRFILEが c:\users\jr8ppg の場合下記の通りです。
CFG/DAT　 ：c:\users\jr8ppg\zlog\cfgdat
log　　 　：c:\users\jr8ppg\zlog\logs
backup  　：c:\users\jr8ppg\zlog\backups
voice　 　：c:\users\jr8ppg\zlog\voice
plugin  　：c:\users\jr8ppg\zlog\plugin
spc　　 　：c:\users\jr8ppg\zlog\spc

```
---

例５）絶対相対の混在
```
zlog.exeの所在地：d:\zlog_reiwa
zLofルート：%ZLOG_ROOT%
CFG/DAT　 ：cfgdat
log　　 　：logs
backup  　：e:\backup_files
voice　 　：d:\member_voice_files
plugin  　：plugin
spc　　 　：spc
```
↓下記の様に解釈
```
CFG/DAT　 ：d:\zlog_reiwa\cfgdat
log　　 　：d:\zlog_reiwa\logs
backup  　：e:\backup_files
voice　 　：d:\member_voice_files
plugin  　：d:\zlog_reiwa\plugin
spc　　 　：d:\zlog_reiwa\spc
```

jj8dan, 2022-12-19:
この案は最高です。これでやりたいことが全部できます。

jr8ppg, 2022-12-19:
じゃあ、これですすめますね。

jj8dan, 2022-12-19:
明示的に書く話("logs"ではなく".\logs")は別件ですね。
"logs”と”.\logs”は、”%zlog_root%\logs"と解釈して良いと思います。

%zlog_root%があるドライブのルート(C:¥やD:¥)を示す方法としての”\logs”や"\\"は分けて考えるか？
需要なさそうだから"\logs" = %zlog_root%\logs", "\\" = "%zlog_root%"でいいかなぁ。
どうしても指定したい人は絶対パスで書いてくださいで困らないでしょう。環境変数で工夫してもらってもいい。

トリッキーなのも放置しましょう"..\\..\logs"とか

jr8ppg, 2022-12-20:
例６）.\\と..\\
zLogルートがきちんと設定されていれば。「.\\」や「..\\」も機能します。
```
zlog.exeの所在地：d:\zlog_reiwa\bin
zLogルート：%ZLOG_ROOT%
CFG/DAT　 ：.\cfgdat
log　　 　：..\logs
backup  　：..\..\backup
voice　 　：voice
plugin  　：plugin
spc　　 　：spc
```
この様に解釈されます。
```
CFG/DAT　 ：d:\zlog_reiwa\bin\cfgdat
log　　 　：d:\zlog_reiwa\logs
backup  　：d:\backup
voice　 　：d:\zlog_reiwa\bin\voice
plugin  　：d:\zlog_reiwa\bin\plugin
spc　　 　：d:\zlog_reiwa\bin\spc
```

jr8ppg, 2022-12-20:
各ファイルを開くダイアログのフォルダ初期値は、現状通りとします。
例）
・logフォルダに設定が無い場合
　フォルダ初期フォルダ＝zLogルート
・logフォルダに設定がある場合
　フォルダ初期フォルダ＝zLogルート+logフォルダ

現状、明示的に”最後に開いたフォルダ”にはできないけど、困ることは無いと思う。

jr8ppg, 2023-01-22:
済

closed 2023-01-22 by jr8ppg