jr8ppg/zLog #344: FT-1011対応

jr8ppg, 2022-09-01:
FT-1011対応

jj1cvh, 2022-09-01:
早速着手いただきましてありがとうございます。
まずは現状をお伝えします。
YAESU機のCATに詳しくないため、間違っていたらスミマセン。

やりたいこと
- FT-1011で、バンドマップダブルクリックでリグコンしたい

現象
- zLog 2.8.2.2, FT-1011(ROM Version 1.06), USBシリアルケーブルのCAT端子の2-1間(SERIAL OUT-GND間)プルダウン抵抗1kΩ
- Hardware | Rig1: COMx 4800 FT-1000 None 選択すると、CATランプが連続点灯、RIGがCAT通信し続けてRIG操作が不能になる
- zLogのRig Controlウィンドウの表示内容も異常になる
  - VFO AとVFO Bは2秒ごとに、7000.000kHz-->0.000kHz-->0.000kHz-->0.000kHz-->7000.000kHzを繰り返し表示する
  - 0.000kHzの時、zLogのmodeがotherになる
- zLogを起動したまま、リグ電源OFFし再度電源ONすると、CATランプが消灯してRIGが操作可能になり、zLog->RIGのリグコン(Band変更、Mode変更、バンドスコープダブルクリック時の周波数&Mode設定)も可能になる。

予想される原因
- FT-1011は、コンファームリクエスト(コマンド10)のVFO A&B読み出しコマンド(00 00 00 03 10)でVFO A&Bデータ(32byte)を要求しても、数秒間かけて1636byteの全データを返答するため
  - FT-1000MPやFT-1000後期型、FT-1021(ROM6.0搭載)は 10 P1 ※ ※ P4 に対応
  - FT-1011やFT-1000前期型、FT-1021(ROM6.0非搭載)は、10 ※ ※ ※ ※にしか対応しておらず、全データを返答

リクエスト
1. FT-1011(no polling)の新設: zLog->Rigの片方向通信とし、コマンド10を一切発行しない(pollingもしない)
2. 片方向通信のみでも、バンドスコープダブルクリック時にzLogのBandとModeを切り替え

---
リクエストの1.については、以下の修正で対応できることを確認しました。
これでやりたいことの8割は実現できています。
もとにしたソースコードは、ちょっと前の、8/24のcommit eb13313 です。
```diff
diff -u zLog/zlog/URigControl.pas zLog-FT1011/zlog/URigControl.pas
--- zLog/zlog/URigControl.pas	2022-08-18 20:27:17.629533600 +0900
+++ zLog-FT1011/zlog/URigControl.pas	2022-08-26 19:08:04.000000000 +0900
@@ -296,6 +296,10 @@
     procedure SetVFO(i : integer); override;
   end;
 
+  TFT1011 = class(TFT1000)
+    procedure PollingProcess; override;
+  end;
+
   TMARKV = class(TFT1000MP)
     procedure ExecuteCommand(S: AnsiString); override;
     procedure RitClear; override;
@@ -1412,6 +1416,12 @@
             rig._maxband := b28;
          end;
 
+         if rname = 'FT-1011' then begin
+            rig := TFT1011.Create(rignum);
+            rig._minband := b19;
+            rig._maxband := b28;
+         end;
+
          if rname = 'FT-920' then begin
             rig := TFT920.Create(rignum);
             rig._minband := b19;
@@ -2121,6 +2131,10 @@
    WriteData(_nil3 + AnsiChar($03) + AnsiChar($10));
 end;
 
+procedure TFT1011.PollingProcess;
+begin
+end;
+
 procedure TFT847.PollingProcess;
 begin
    FPollingTimer.Enabled := False;
diff -u zLog/zlog/UzlogConst.pas zLog-FT1011/zlog/UzlogConst.pas
--- zLog/zlog/UzlogConst.pas	2022-06-12 12:46:34.165568700 +0900
+++ zLog-FT1011/zlog/UzlogConst.pas	2022-08-26 19:16:26.000000000 +0900
@@ -37,7 +37,7 @@
   TXLIST_MS = '0,1';
 
 const
-  RIGNAMES : array[0..17] of string =
+  RIGNAMES : array[0..18] of string =
 ('None',
  'TS-690/450',
  'TS-850',
@@ -52,6 +52,7 @@
  'FT-991',
  'FT-100',
  'FT-1000',
+ 'FT-1011',
  'FT-1000MP',
  'MarkV/FT-1000MP',
  'FT-1000MP Mark-V Field',
```

jj1cvh, 2022-09-01:
リクエストの2.は、上記の修正を行ったzlog.exeで検証した結果、以下の事象が発生したことによります。
やりたいことの残り2割なので、リクエストの1.よりは優先度は低いです。
FT-1011のために対応することによって、双方向通信で普通にリグコン可能な他のリグに悪影響が生じないか、気になっています。

事象
- バンドマップダブルクリック時に、リグの周波数とモードは正常に切り替わる一方、zLogのBandが切り替わらない
- 「Use Estimated Mode by Freq.」にチェックしていても、zLogのModeが意図しないモードに切り替わる

再現手順
(あらかじめzlog.iniを削除)
1. zLogをField　Dayで起動
2. zLogのBandとModeを1.9 CWから28 FMに変更
3. Settings > Options で以下を選択し[OK]
   - Hardware | Rig1: COMx 4800 FT-1011 None
   - Hardware2 | PacketCluster: TELNET ccc.jg1vgx.net 7373
   - BandScope | Bands: 7 MHz
4. Windows > Packet Clusterで[Connect]
5. zLogのBandとModeを28 FMから21 SSBに変更(リグの周波数とモードも正常に切り替わる)
6. 7MHzバンドスコープのスポット(CW帯)をダブルクリック
7. リグは、周波数とモードが正常に切り替わる(例: 7.01050MHz CW)
8. zLogは、Bandは21から変わらず、ModeがFM(Options > [OK]時のMode)に切り替わる ★7 CWに切り替わってほしい

「Options > [OK]」だけでなく、「Options > [Cancel]」でも再現します。
また、RIG1などのOptions設定がzlog.iniに記録された状態で、zLogを起動し、Options画面を開かなかった場合は、zLog起動時のModeに切り替わる現象を確認しています。

---
以下は、参考情報です。

7M4KSCさんのブログ
https://blog.goo.ne.jp/baristas_latte/e/a040b362a45196f59467c0646c065f05
で、FT-1011のRIGからの受信は「ROM依存」となっていますが、FT-1011のROM依存性については情報を見つけられませんでした。

一方、FT-1011の海外モデルであるFT-990は情報がいろいろあり、多くのソフトはROM1.3でないとまともに動かず、ROM1.2だとリグの返答中は3秒間ロックアウトするそうです。

https://groups.io/g/ft990/topic/cat_control_i_give_up/79953277
Note: ROM version 1.2 always returns 1492 bytes for the UPDATE command. This is the problem with 1.2 because the 990 is locked out for 3 seconds during the time the 1492 bytes are returned.

DXLab Commanderは、FT-990(ROM1.3用)とFT-990_12(ROM1.2用)が用意されていて、FT-990_12はUpdateの[Freq & Mode]ボタンをクリックした時だけリグ情報を取得するように工夫されているようです。
http://www.n0agv.com/yaesu/Yaesu_FT-990_Difference_Rom1.2_zuRom_1.pdf
With the DXLAB suite there are two choices for a FT-990. One choice is for ROM 1.2. In that case it sends the UPDATE command only one time when the UPDATE button in the COMMANDER window is clicked on. This results in a 1492 byte reply from the 990 and it gets the VFO and mode info from the 1492 bytes. For ROM 1.3 the special UPDATE option that returns the two VFO frequencies is used. It keeps sending
several UPDATE commands every second but it only takes 64 milliseconds to get the info and COMMANDER can update the window to display the 990 vfo frequencies in realtime.

hamlibも、従来のROM1.3向けのft990に加えて、最近はROM1.2向けのft990_v12を用意しているようです。
https://github.com/Hamlib/Hamlib/tree/master/rigs/yaesu

jj1cvh, 2022-09-01:
YAESU機に詳しい7M4KSCさんとJO3JYEさんをアサインさせていただきました。

今回、FT-1011対応を行おうとしているのですが、zLogに表示するリグ機種名について、もしよろしければご意見をお聞かせください。

FT-1011選択時は、zLog->RIGの片方向通信のみとし、リグの周波数情報とモード情報を取得しないようにしたいと考えています。
FT-1011は、ROMがはんだ実装されており容易に交換できないので、JAの多くのFT-1011は古いROMであり、現状のzLogでFT-1000を選択するとRIGがCAT通信し続けてRIG操作が不能になる事象が発生するのではないかと想像しています。
※私自身はYAESU機に全然詳しくないので、あくまで想像です。

FT-1011に限らず、FT-1000前期型やFT-1021(ROM6.0非搭載)でも同様にRIG操作が不能になるようなら、zLogのRig選択欄に表示する機種名は「FT-1011」ではなく、zLogユーザにもうちょっと分かりやすい表示にしたほうが良いかもしれないと思っています。

少なくとも、他のリグと異なり、片方向通信であることを明示するため、例えば「FT-1011(no polling)」とか、「FT-1011(zLog-->Rig only)」など、zLogに表示する機種名を一工夫したほうが良いかもしれないとも思っています。

7M4KSC, 2022-09-02:
FT-1011旧・新とすると、ROMのバージョンは２種類しか表せないので、もうROM ver.を表記してしまうのはどうでしょう？
FT-990 ROM1.2
FT-990 ROM1.3
FT-1011 ROM1.06
FT-1011 ROM1.07以降
みたいな感じ。
（ICOM,Kenwoodは素直で古い機種でも動作しますが、Yaesu機は入手しておいた方が良いなぁ・・・）

jr8ppg, 2022-09-02:
なるほど。
周波数設定のコマンド送付は成功したが、その後zLog側のバンドが切り替わらないと。
→コマンド送付後の、周波数読み取りでzLog側のバンド切替をおこなっているため。

と言う事は、コマンド送付後、内部で折り返して周波数情報を読み取ったことにしてあげればよろしいのかな。

jr8ppg, 2022-09-02:
これでどうでしょうかね？

jj1cvh, 2022-09-02:
@jr8ppg さん、ご対応ありがとうございます。送出されるコマンドをモニタする限りではよさそうな感じですが、FT-1011(ROM1.06)実機に接続した確認結果は別途報告させていただきます。

@7M4KSC さん、FT-1011は、 ROM1.06 だけでなくROM1.07以降も存在するのですか! 初めて知りました。ROM1.07以降は、zLogでFT-1000を選択すれば双方向のリグコンが普通に出来る、と理解してよろしいでしょうか？

また、FT-1000とFT-1021は、ROMバージョンに関わらず、FT-1000を選択すれば普通に双方向のリグコンができ、FT-1011(ROM1.06)のようにリグが固まって操作できなくなるような問題は発生しない(はず)、と理解してよろしいでしょうか？

もしそうであれば、zLogのRig欄に表示されるリグのリストは、以下のような感じでいかがでしょう＞皆様、ご意見をお寄せください

- 現状 (commit f47d438)
FT-100
FT-1000
FT-1011
FT-1000MP
MarkV/FT-1000MP <--このくらいが表示文字数の限界

- 変更案
FT-100
FT-1000/1011(1.07-)
*FT-1011(1.06)
FT-1000MP
MarkV/FT-1000MP

*をつけたのは、PC->Rigの片方向通信通信で、他の機種とは異なる挙動を示す(RigのVFO情報を取得しない)ことをマーキングしたいなと思ってのことです。
文字数の問題もあり、かといって対応機種を1行1機種で列挙するとリストが長くなるため、対応機種の詳細は、これまで通り以下で説明するのが良いのかな、という気がしてます。
https://github.com/jr8ppg/zLog/wiki/%E3%83%AA%E3%82%B0%E3%82%B3%E3%83%B3%E3%83%88%E3%83%AD%E3%83%BC%E3%83%AB

jr8ppg, 2022-09-03:
なんだかややこしいですね。
FT-990は国内には無いのでパスするとして、ROM 1.07はFT-1000と同等で、1.06が今回の対応。
FT-1011(Set only)とかにしてもわかりづらいかな。うーむ。

7M4KSC, 2022-09-03:
とすると、（ROMバージョン書きは廃棄として）こんなのでどうでしょ？
FT-100
FT-1000/1011
*FT-1011(旧ROM)
FT-1000MP
MarkV/FT-1000MP

（こうなると、８Gより前の古いFT-847が気になってくる・・・。そもそもCATは動くのだろうか？）

jj1cvh, 2022-09-06:
ft1011ブランチのcommit f47d438 について、FT-1011(ROM Version 1.06)の実機確認結果をご報告します。

1. zLog->Rigの制御は、おおむね期待通りに動作。
2. 周波数折り返しの副作用として、バンドスコープダブルクリック後は、バンドスコープへの手動登録(Ctrl+N、Shift+Ctrl+N)が実質的に使えない状態になる。
3. バンドスコープダブルクリック時に、バンドスコープのスポット周波数に対して、リグのVFO周波数が数100Hzずれることがある。

2.については、たまに使う程度の機能ではありますが、再現性が確認できてます。
3.については、現象発生のタイミングやズレ幅(700Hzや300Hzやズレ無しなどいろいろ)の規則性がまだ良くわからない(まったくのランダムというわけでもない)状況のため、ひとまず2.の事象について以下に記載します。
※ 3.は、zLogのRig ControlウィンドウのVFO Aもズレているか否かも未確認の状態ですので、別途ご報告します。

zlogwin_v22.pdfより
- Ctrl+N （登録後に入力フィールドをクリアする）または Shift+Ctrl+N （登録後に入力フィールドをクリアしない）でコールサインフィールドに入力されている局とコンテストナンバーを登録できます。周波数はリグから読み込みますが、リグコントロールをしていない場合は周波数を入力するよう促されます。

現象
- バンドスコープダブルクリックにより、zLogのVFO Aに周波数折り返しのスポット周波数を取得した後、リグのVFOダイヤルを回して呼び回りしながら、バンドスコープにコールと周波数をメモ的にCtrl+Nで登録しようとすると、周波数入力画面がポップアップせず、取得済みのスポット周波数でバンドスコープに登録されてしまう。
- zLog起動後、またはOptions画面を開いて閉じた後、zLogのVFO Aが0.000kHzの状態であれば、Ctrl+Nで周波数入力画面が期待通りポップアップされる。

周波数折り返しとは別のアプローチとして、Options > Hardware | Rig1 のRIG欄がNoneの場合も含めて、zLogのVFO Aが0.000kHzの状態でバンドスコープダブルクリックした時は、zLogのBandとModeを切り替える方法もあり得るかな、と思いました。
※ Mode切り替えは、「Use Estimated Mode by Freq.」チェック時のみ

ただし現状では、RIG欄がNoneの状態でバンドスコープダブルクリックすると、zLogのBandは変更されるもの、zLogのModeは変更されませんでした。

---
zLogのRig欄に表示されるリグのリストについて、FT-1011は旧ROMということをzLogユーザに伝えることに重きを置くか、PC->zLogの片方向という機能的な違いに重きを置くかによって、どういう表記が良いのか変わるかなと思いました。
前者なら、英語表記を前提として「FT-1011(old ROM)」とする手もありますが、個人的には後者で以下の感じが良いかなと思いました。
FT-1011(Set only)よりは意味が分かりやすいかなと思ったのですが、いかがでしょう。

- 変更案２
FT-100
FT-1000
FT-1011(PC->Rig)
FT-1000MP
MarkV/FT-1000MP

@7M4KSCさん
以下の2点が、あいにく良くわかりません。
- FT-1011(ROM1.07以降)は、zLogでFT-1000を選択すれば双方向のリグコンが普通に出来るのか
- FT-1000とFT-1021は、ROMバージョンに関わらず、FT-1000を選択すれば普通に双方向のリグコンができ、FT-1011(ROM1.06)のようにリグが固まって操作できなくなるような問題は発生しない(はず)なのか

既に分かっている、あるいは想定されている情報があるようでしたら、
https://github.com/jr8ppg/zLog/wiki/%E3%83%AA%E3%82%B0%E3%82%B3%E3%83%B3%E3%83%88%E3%83%AD%E3%83%BC%E3%83%AB
に追記、修正いただけるとありがたく存じます。

JO3JYE, 2022-09-06:
実機も情報も無いため傍観しておりました。(ずいぶん前にFT-1011も所有していましたが、20年以上前にお別れしてしまいました。)
バンドスコープが事実上使えなくなる件、こんな対応はいかがでしょうか?

・周波数折り返しの部分は今まで通り。
・TFT1011.PollingProcess　を作成。
　ここでRIGへの問い合わせコマンドを投げるかわりに、　周波数　0　のデータを取得したことにする。
　1回ぽっきりいいので、1回呼ばれた後は　PollingTimer　を止める
・PollingTeimerはsetfreqのあと1回だけ動かす。

これで、周波数　０　のデータが来たことになれば、バンドスコープへ追加するときに周波数入力を求められるようように思います。
本来は、PollingProcessでは、周波数問い合わせのコマンドを投げるだけなので、ちょっと無理があるかも？

PPGさん、このアプローチでいかがでしょうか？
無理だったらすみません。

JO3JYE, 2022-09-06:
TFT1011.Pollingprocessで　周波数で０のデータを用意
 TFT1011.ExecuteCommand　で　周波数　０　を取り出す。
でしょうか？
これで成立するか?ご検討いただければ。。。

7M4KSC, 2022-09-07:
>FT-1011(ROM1.07以降)は、zLogでFT-1000を選択すれば双方向のリグコンが普通に出来るのか

CATのコマンド表では、FT-1000を選択していれば、FT-1011の新ROMだと動作すると思われます。
（ヤエスなので実機だと何が起こるかわかりませんが・・・）

>FT-1000とFT-1021は、ROMバージョンに関わらず、FT-1000を選択すれば普通に双方向のリグコンができ、FT-1011(ROM1.06)のようにリグが固まって操作できなくなるような問題は発生しない(はず)なのか

はい、FT-1021が作られて、その後FT-1000の発売ですが、そのまま動くと思われます。FT-1021とFT-1000で取扱い説明書の中のCATコマンド一覧に違いは見られません。（ヤエスなので・・・というのは上記と同じですが）

jr8ppg, 2022-09-07:
@JO3JYE 
最初の１回じゃだめかな。バンドスコープからジャンプした後に、一度PollingProcessを開始して現在freqを0にしてやるって感じかな～と思います。

jr8ppg, 2022-09-07:
> PollingTeimerはsetfreqのあと1回だけ動かす。

って書いてありましたね

jr8ppg, 2022-09-07:
JYEさん正解！
これでどうでしょうかね。

jj1cvh, 2022-09-08:
jr8ppgさん、JO3JYEさん、ありがとうございます。
ft1011ブランチの9/7付のcommit 27b4eda にて、Ctrl+Nで周波数を手動登録できることを確認しました。

一方、FT-1011での一連の検証で、FT-1011とは直接関係なく発生していると思われる現象を2つ見つけました。

1. CW帯のスポットをバンドスコープでダブルクリックすると、ダブルクリック前後でリグのモードが変わる場合、バンドスコープのスポット周波数に対して、リグのVFO周波数が数100Hzずれる
2. リグ周波数を取得できない場合(Rig1がNoneやFT-1011の場合など)に、Ctrl+Nで周波数を手動登録する際、zLogのBAND(例えば7MHz)と異なるバンドの周波数(例えば14050 kHz)を登録すると、zLogのBAND(例えば7MHz)のバンドスコープ画面に、登録した周波数(例えば14.050 MHz)の情報が掲載されてしまう。

特に1.については、FT-1011以外のYAESU機でも現リリース版のzLog 2.8.2.2で発生している予感がします。
リリース版のzLog 2.8.2.2で構いませんので、特にYAESU製のいろんな機種で、皆さんに検証していただけますと幸いです。
他機種でも発生しており、改良と検証に手数がかかるようでしたら、別のIssuesを立てて頂ければ、と思います。

以下、1.についての詳細です。

FT-1011の検証結果では、CWピッチの700Hz(FT-1011初期値; ディップスイッチで変更可)と同じズレが生じました。
例えば、リグが7.007.00 LSB、zLogが7MHz SSBの状態で、7.011.5のCW帯のスポットをダブルクリックすると、リグの周波数とモードは、
7.007.00 LSB --> 7.011.50 CW
にはならず、
7.007.00 LSB -->  7.012.20 CW (+700Hz)
になります。

zLogが送出するCATコマンドの設定周波数は正常で、周波数折り返しによりzLogのRig Control画面にちらっと表示されるVFO A周波数は、7.011.50でした。

zLogが送出するコマンドは、周波数設定、モード設定の順で2コマンドのため、
7.007.00 LSB -(周波数設定)->  7.012.00 LSB -(モード設定)-> 7.012.20 CW (+700Hz)
の順で切り替わることにより700Hzズレてしまうことが原因ではないか、と想像しています。

周波数設定、モード設定の順でzLogがコマンドを送信しており、かつリグのモード切り替え時にVFO表示周波数が変わる機種では、同様の現象が発生するのではないかと想像しています。

実のところ、FT-1011は私が持っているリグではなく、知人宅に置いてあるものを検証してもらっているのですが、FT-897DMでも、ダブルクリック前後でモードが変わる場合に周波数がズレる現象が発生するとのことです。
一方、IC-756ProIIでは発生しないそうです。

解決策としては、
周波数設定、モード設定、周波数設定
の3つのコマンドを送出する必要がありそうな気がしています。

jr8ppg, 2022-09-08:
jj1cvhさん
それは、CWのUSB/LSB問題ですね。SSBからCWに変わるときにside bandが一致していないとその分だけずれる話と思います。7MHzのSSB(LSB)->CW(USB)ですね。
ご提案通りモード変更後はもう一度周波数セットをしてみようと思います。

ICOMはCWのside bandが他社とは逆ですので再現手順を変えないとだめです。
21MHzでSSB(USB)->CW(LSB)とした場合はどうでしょうか。

KENWOODのside bandはYAESUと同じです。

jj1cvh, 2022-09-11:
> ICOMはCWのside bandが他社とは逆ですので再現手順を変えないとだめです。
> 21MHzでSSB(USB)->CW(LSB)とした場合はどうでしょうか。

周波数ズレ対策(side bandずれ対策 5a5a911)を行う前の状況からご報告します。

ICOM機は、SSB(LSB)-->CW(LSB)でも、SSB(USB)-->CW(LSB)でも、バンドスコープダブルクリック時の周波数ズレは発生しませんでした。
※IC-756ProII, IC-910, IC-9100, IC-7300, IC-9700 で確認

一方、YAESU機(FT-1011, FT-897(FT-817を選択))の場合は、SSB(LSB)-->CW(USB)で+700Hz、SSB(USB)-->CW(USB)でも+700Hzの周波数ズレが発生しました。

リグ単体で、リグのモードボタンでモードを切り替えると、YAESU機は下記のようにリグの表示周波数が700Hz変化しますが、ICOM機は変化しません。
zLogのバンドスコープダブルクリックで周波数ズレが発生する際のズレ量と、正負含めて一致しています。

＜FT-1011＞
LSB-->CW: +700Hz
USB-->CW: +700Hz
CW-->LSB: -700Hz
CW-->USB: -700Hz

＜FT-897＞
LSB-->CW: +700Hz
USB-->CW: +700Hz
CW-->LSB: -700Hz
CW-->USB: -700Hz
LSB-->CWR:+700Hz
USB-->CWR: +700Hz
CWR-->LSB: -700Hz
CWR-->USB: -700Hz

また、周波数ずれ(side bandずれ)対策後のcommit 2a1a80f については、FT-1011で検証中です。
修正内容は、バンドスコープダブルクリック時にzLogが送出するコマンドで、以下と理解しました。

(side bandずれ対策前)
  「周波数設定＋モード設定」
(side bandずれ対策後)
  zLogのモードが変更される場合は「周波数設定＋モード設定＋周波数設定」、変更されない場合は「周波数設定」

FT-1011は片方向通信なので、リグ側でボタン操作してしまうと、リグとzLogのモードが一致しない状態となるケースが想定されます。
リグのモードがSSB、zLogのモードがCWの状態で、CW帯のスポットをダブルクリックすると、zLogはCWのままなので「周波数設定」のみとなり、モードが切り替わらない現象が発生します。

FT-1011のリグ側でボタン操作してバンドやモードを切り替えた場合は、zLogのバンドやモードを手動で合わせてからQSOをロギングするので、通常は問題無いと思いますが、実運用でどのくらい問題にないそうか、もうちょっと考えてみます。

jj1cvh, 2022-09-11:
> 一方、FT-1011での一連の検証で、FT-1011とは直接関係なく発生していると思われる現象を2つ見つけました。
>
>     1. CW帯のスポットをバンドスコープでダブルクリックすると、ダブルクリック前後でリグのモードが変わる場合、バンドスコープのスポット周波数に対して、リグのVFO周波数が数100Hzずれる
>
>     2. [...]

1.について、周波数ずれ(side bandずれ)対策を行った commit 5a5a911 にて、周波数が+700Hzずれる現象が解消したことをFT-1011で確認しました。
LSB->CW、USB->CWいずれも、スポット周波数とリグ周波数が一致するようになりました。
ありがとうございました。

バンドスコープダブルクリック時の送信コマンドが、
> (side bandずれ対策後)
> zLogのモードが変更される場合は「周波数設定＋モード設定＋周波数設定」、変更されない場合は「周波数設定」

となったことについては、片方向通信のFT-1011に限って言えば、モードが変更されない場合は「周波数設定」ではなく従来通り「周波数設定＋モード設定」の方が良いです。
※リグ側でボタン操作してしまうと、リグとzLogのモードが一致しない状態となるケースが想定されるため

ただ、モードの不一致は、運用中に気を付ければ回避できる話なので、このままでもまぁ良いかなと思っています。

2.のCTRL+Nについては、いかがでしょうか？

jr8ppg, 2022-09-11:
2.は見落としていました。ちょっと考えます。

jr8ppg, 2022-09-12:
side bandずれ対策はbandscopeのoptionで設定できるようにする。

closed 2022-09-16 by jr8ppg