jr8ppg/zLog #369: ウインドウ(W)の各項目をタスクバーに収納

jr8ppg, 2022-10-13:
例えば「スコア(S)」を押してScoreを表示させて、タスクバーに収納して
「スコア(S)」を押すとScoreウインドウが出てきません。
タスクバーから選択しないと出てこないです。これは出てきて欲しいです。
(5)のWinキー+Dしてしまうと大変です。
ウインドウ(W)からでは出てこなくなります
(タスクバーで一個一個選択すると出てくる）

jj8dan, 2023-01-28:
スコア(S)でScoreウインドウを呼び出そうとするのですが、下記の手順を取ると復帰しないです。タスクバーから呼びだせば復帰します。環境：zlogwin_2842_x64_20230121　Windows10 x64 Pro 21H2

zLog for Windows ウインドウからスコアを呼び出す
[image]

Scoreウインドウが出てくる
[image]

Winキー+D でウインドウを閉じてタスクバーからzLog for Windows ウインドウを呼び出す
[image]

Scoreウインドウが最小化？されて全画面の左下に出てくる
[image]

この状態からzLog for Windows ウインドウから「ウインドウ(W)-スコア(S)」でScoreウインドウを呼び出すがScoreウインドウは出てこない
[image]

タスクバーから呼びだします
[image]

タスクバーから呼びだすと出てきます
[image]

ウインドウ(W)-スコア(S)でScoreウインドウが出てきてほしいです。

jr8ppg, 2023-01-28:
現状の動作は合っていると言えば合っているので、続きは #403 にて。

closed 2023-01-28 by jr8ppg