jr8ppg/zLog #497: マルチオペ：スクロールが新QSOで戻ってしまう

jj8dan, 2023-08-25:
Z-Link時に、新しいQSOが届くと画面スクロールが最新に戻ってしまいます。
Dupe言われて過去QSOを探しにいっている時に新QSOが届くと修正ができないのです。
（現状のままでやるなら、一旦、Z-Link切ればいいのですが）
ただ、画面スクロールを過去のままにいつまでもするのも困りものです。
例えば、スクロールそのままでTwitter的に「新しいQSOがあります」とうながすとかですかね。

jr8ppg, 2023-08-25:
> Dupe言われて過去QSOを探しにいっている時に新QSOが届くと修正ができないのです。
DUPEと言われたら「ありがとうございました」でスルーするのが良いと思いますが、それはそうとして問題点はわかりました。
対応考えます。

jj8dan, 2023-08-25:
自分ではDupeのつもりでなくてDupe言われると過去QSOのコールサインをミスコピーしているので探しにいかなければ1QSO逃してしまいます。

jr8ppg, 2023-08-25:
過去のQSOは直さないで、新しいQSOをそのままログインしてしまえばいいのですよ。

jj8dan, 2023-08-25:
何時何分Dupeです、ありがとうございました、QRZ と言われて交信成立しないのでそこに新しいQSOはないのです。もう一回交信できれば過去QSOを訂正する必要はないです。最近はDupeを許容するスタイルの方が多いですが中にはかたくなな方もいます。と、いった背景があります。

jr8ppg, 2023-08-25:
呼びに回って相手側がDUPEなんですね。私は暇なときならともかく、面倒なのでそのままにしてますわ。
KCJみたいなログ照合のある場合は別ですけど、uniqueになるだけなので。

jr8ppg, 2023-08-27:
こんな感じかな
[image]

jr8ppg, 2023-09-11:
交信リストスクロールのみの場合に対応

jr8ppg, 2023-09-18:
実際にXPOコンテストで使ってみると、他のzLogでの交信データを受信すると、何でも無いのにお知らせがでてウザいので、表示条件の見直しを行った。

jj8dan, 2023-09-20:
> 実際にXPOコンテストで使ってみると、他のzLogでの交信データを受信すると、何でも無いのにお知らせがでてウザいので、表示条件の見直しを行った。

私のところでもなんでもないのに出ました。コールサイン欄でスペース押すと消えました。

jr8ppg, 2023-09-22:
済み

closed 2023-09-22 by jr8ppg