jr8ppg/zLog #510: Z-Server接続時の処理に「接続のみ」を追加する

jr8ppg, 2023-09-30:
Z-Server接続時の処理に、「Merge」「Download」があるが、どちらもやりたくない場合のために「Connect only」を追加する。
Mergeは処理に時間がかかり、コンテストのある局面ではやらない方が良い場合がある。
DownloadはzLog側のみにあるQSOが無くなってしまう。

jr8ppg, 2023-09-30:
こんな感じで
[image]

jr8ppg, 2023-09-30:
これはOKでしょう

jr8ppg, 2023-10-01:
デフォルトは「マージ」に変更
[image]

jj8dan, 2023-10-01:
OKを押すと３択した１つが動作するのですが、Cancelを押した場合はどう動作します？
今まで(例えば2.8.6.1)は、

- すでにZ-Linkは確立している
- マージかダウンロードを選んでOKを押すとそれが実行される
- Cancelを押すと今回の「接続のみ」動作が結果的にが実施される（接続自体はキャンセルされない）。

今後Cancelはどう扱いますか？Disconnectします？それかCancelボタン自体なくすか。
(ウィンドウ左上にはすでに「Connected...」と出ているのでわざわざDisconnectしなくてもいいか)

jr8ppg, 2023-10-01:
Cancelは「接続のみ」と同じですね。3番目です。「×」も同じです。

「OK」なら接続、「Cancel」なら接続しないの方が一般的は解釈なのでそうしたいですね。

closed 2023-10-01 by jr8ppg