jr8ppg/zLog #524: CabrilloでACAGの区ナンバーも出るようにする

jr8ppg, 2023-10-11:
Cabrillo形式でexportするとACAGの区ナンバーの後の電力符号が切れてしまう。

jr8ppg, 2023-10-11:
zlistwではＮＲ部が6桁超えてもそのまま出している。
```
zlistw
QSO:  7000 CW 2023-04-30 2101 JR8PPG        599 106H   JR8YLF/8      59  36H    0
QSO:  7000 PH 2023-10-10 2112 JR8PPG        59  010103H JR8YLF/8      59  010104M 0

zLog
QSO:  7000 CW 2023-04-30 0005 JR8PPG        599 106H   JR8YLF/8      599 106M   0
QSO:  7000 PH 2023-10-07 2242 JR8PPG         59 010103 JR8YLF/8       59 010101 0
```

jr8ppg, 2023-10-12:
各フィールドは規定の桁数未満の場合は、規定の桁数までスペースでパディング。
規定の桁数以上の場合はデリミタにスペースを1個つける。

jr8ppg, 2023-10-12:
済み

jj1cvh, 2023-10-12:
Cabrilloで固定長を超えるの文字数の場合にデリミタとしてスペース1個をつけるのは、正しい処理のようですね。

https://wwrof.org/cabrillo/cabrillo-specification-notes/
> While the specification implies fixed columns for all data, this is not a requirement.  Most contest sponsors accept any space or tab character as a delimiter between data. The major contest sponsors such as CQ and ARRL do accept space or tab as a delimiter. I.e., data does not need to be in specific columns.

jr8ppg, 2023-10-12:
確認ありがとうございます。ちゃんと書いてあるんですね。

jr8ppg, 2023-10-13:
この際だからサマリー部も入力できるように変更。

jr8ppg, 2023-10-20:
とりあえずOK

closed 2023-10-20 by jr8ppg