jr8ppg/zLog #589: KENWOODのリグコントロールでの不具合

jr8ppg, 2024-04-29:
初期状態が2）で何かの操作で1）になったとのこと。
1）RTSの設定をNoneにしてもハンドシェイクにしてもフロー制御Noneになる
2）RTSの設定をハンドシェイクにしてもNoneにしてもフロー制御ハードウェアになる

再現確認まだ

jj8dan, 2024-05-12:
2.9.1.1
|I/F|Serial|RTS|DTR|Rig to PC|PC to Rig|Hang-up|
|-|-|-|-|-|-|-|
|USB-Serial|9pin full|ハンドシェイク|None|OK|OK|None|
|USB-Serial|9pin full|None|None|OK|OK|None|
|USBIF4CW Gen.3|TXD,RXD,GND| ハンドシェイク|None|Fail|Fail|ハードウェア設定を選ぶと発生、USBケーブルを抜くと回復|
|USBIF4CW Gen.3|TXD,RXD,GND| None|None|OK|OK|None|

|Magical Calling|メインダイヤルを回す度にTS-570からBeep|
|-|-|
|☑使用する|鳴る|
|☐使用する|鳴らない|

やっとまとまりました。TS-570からBeepが鳴るのはMagical Callingを使うと発生します。これをシリアルポートのRTS制御と混同していたので訳がわからなくなっていました。

jr8ppg, 2024-05-12:
KENWOODはPCからコマンドを送るといちいちピッってなるので、メニューでOFFにすると良いですよ。
TS-570はメニュー01です。

jr8ppg, 2024-05-12:
対応なし。done

closed 2024-05-12 by jr8ppg