jr8ppg/zLog #632: 入力中のコールサインがtelnet(RBN)に記録されたものかを点検する機能

jucky154, 2024-07-11:
ZLOの部員からzLogにも入力中のコールサインが過去のTELNETに記録されたものかを点検する機能がほしいという意見をもらいました
https://n1mmwp.hamdocs.com/manual-windows/check-window/
のようなことをイメージしているようです

TELNETに記録されたコールサイン（データの質を保つために3回以上記録されたコールサイン）で, N+1的なことをする感じだと思っています

現状、自分はよくやくZyLO側からzLogの一部関数を呼び出せるようになったレベルなのでどう実装すればいいかわからない状況で, 意見を書いてしまい申し訳ないです…
自分の方では総務省APIを叩く機能を改良して, 入力完了（Spaceキー入力などでカーソルがナンバーに移った際）したコールサインが5~6文字で日本のコールサイン形式に一致する際には総務省APIを叩き, 存在しないコールサインの場合はDLL側からzLogのMainForm.WriteStatusLine関数を叩いて, メインウィンドウの下に表示できるようにする機能追加で, 部員の要望の一部を叶えようかなと思っているところです

jr8ppg, 2024-07-14:
なるほど。それはアフターコンテストの処理ってことですよね。

jucky154, 2024-07-14:
アフターコンテスト(コンテスト終了後)にコールサインチェックして、修正すると規約に引っかかる部分がありそうなので、
イメージ的にはスーパーチェック・パーシャルチェックなどと併用して、スーパーチェックの補助的に使用することをイメージでした

CWで信号が弱く聞き取りにくいコールサインをTELNETのデータをうまく活用して、解読しようというものです。

jr8ppg, 2024-07-14:
なるほど、建前と言うか規約に抵触しない建て付けが必要ですね。
参考情報ですがWin-Testは終了後のログ書き出し時に、そう言うレポートを出力してくれます。

jr8ppg, 2025-01-26:
案

データの登録
1. スポット受信
2. スーパーチェックのリストへの存在チェック
3. あればRBN出現回数を＋1する
4. なければリストに追加してRBN出現回数を1にする

照合
5. コールサイン入力
6. スーパーチェックとN+1実行
7. マッチしたコールのRBN出現回数がn回以上の場合に表示を変える（RBN照合済みなど）（ｎは2～5を想定）

jr8ppg, 2025-02-12:
できた