jr8ppg/zLog #722: zLog を使用していて気付いたこと

irukabando, 2024-12-01:
zLog をコンテストや通常交信で使っており，気づいたことを書き留めていたので，もし，取入れらるところなどありましたら取り入れていただけますと幸いです。

- 送信 RST を 599 固定でなく，579 などに変更できるようにできないか
- cw message に小文字を入力すると送信されない
- cw message で，[BK], [BT] 等が使えなかった。
- QSO Party のログとして使用しようとしたときに、rcvd 欄に日本語が入力できないのをなんとかできないか
- jcc jcg 判定マクロがあると，個人的に通常交信の送信メッセージを作成するときに便利
- cw キーイング中に設定画面を誤って開いて cw の pc キーイングを止めると cw monitor の進捗が途中で停止した状態になる
- プレ QSO 時にナンバー入力を求めないようにすることはできないか（設定したコンテスト期間以前の交信の際もナンバーを入力しないとエラーが出た）
- cwmessage 送信にショートカットキーを割り当てられないか
- cw keyboard 送信が上手くいかないときがある（入力中に送信されている場合、途中で送信が止まってしまう）

もう改善されているものがありましたらすみません。

よろしくお願いします。

jr8ppg, 2024-12-03:
レポートありがとうございます。
取り急ぎ確認しました。

> 送信 RST を 599 固定でなく，579 などに変更できるようにできないか

手で入力するか、SHIFT+RやSHIFT+Sで変更可能です。

> cw message に小文字を入力すると送信されない

小文字は別の意味を持たせているので送信できません。

> cw message で，[BK], [BT] 等が送信されなかった。私の入力が間違っているのか，ご確認いただけますとありがたいです。

確認しました。2.8.1.1 で動かなくなってました。  #723 で対応します。

> QSO Party のログとして使用しようとしたときに、電話での交信の際はオペレーターネームを日本語で入力したいが，rcvd 欄に日本語が入力できなかった。また，qso party 用の紙ログ出力を実装できないか。

そうなんです。私はQSOパーティーだけ紙ログ使ってます。それかローマ字で入力しています。
印刷は誰か作ってって感じ。

> $Q の値について，jcc/jcg 判定マクロがあると，個人的に通常交信の送信メッセージを作成するときに便利

うーん、どういった物でしょう？

> cw キーイング中に設定画面を誤って開いて cw の pc キーイングを止めると cw monitor の進捗が途中で停止した状態になる

V2.9系ではハードウェア設定を開くとCW Monitorは消えると思います。

> プレ QSO 時にナンバー入力を求めないようにすることはできないか（設定したコンテスト期間以前の交信の際もナンバーを入力しないとエラーが出た）

確認しました。確かにそうですね。SHIFT+Enterで回避願います。

> cwmessage 送信にショートカットキーを割り当てられないか

できないことも無いけど、割り当てるキーが無いのです。２ストロークキーですかね。
例）CTRL+M の後に1～9を押すと、CW MessagePadの1～9に対応した物を送る

> cw keyboard 送信が上手くいかないときがある（入力中に送信する機能、途中で送信が止まってしまう，1文字送信されて停止することもしばしばある）

これは聞かないですね。しばしばあるなら何か考えないといけませんが、特定のＰＣとＩ／Ｆなのか、ＰＣやＩ／Ｆの種類を問わずなのかなどの情報が欲しいですね。

irukabando, 2024-12-03:
ご連絡ありがとうございます。

コールサイン欄右の RST は送信 RST の欄ということでしょうか...　ただ，CW の送信メッセージを $R としているのですが，RST を変えても常時 599 が送信されてしまいますが，これも致し方無いでしょうかね...　手打ち，cw keybord 等がひとまずの解決策でしょうか...

cw message について，小文字は送信できない仕様とのこと承知しました。
ショートカットキーについては，確かに，割り当てるキーの問題で厳しいかもしれないですし，割り当てたとしても，押しにくいかもしれないですね...

最新版で，cw monitor の進捗が設定画面を開くと止まること確認しました。ありがとうございます。

プレ QSO のエラーについては，Shift + Enter で回避できるとのこと承知しました。

cw keyboard の送信が止まることについては，最近あったこととして，「73」と入力したときに，7 まで送信され，3 は送信されない，その後，「TU」などの追加の文字を入力すると 3TU が送信される現象が起きていました。毎回ではありませんが，1日運用していて何度か発生している状況です。装備は IC-7100M に digiaid を介してキーイングを行っています。すみませんが，これ以外 CW 対応リグは所有しておりません。
PC 依存性については，検証できておりません。あまり聞かないとのことも承知しました。運用にものすごい影響が出ているわけでもないですし，I/F については，digiaid しか所有しておりませんが，PC 依存性についてはしばらく検証してみたいと思います。

QSO パーティー関係のところにつきましては，ローマ字で提出しても受理されているという話も聞きますし，ローマ字で使用したいと思います。紙ログ出力は，Golang プラグインなどで自作に挑戦するかもです。

どうぞ，よろしくお願いします。

jr8ppg, 2024-12-03:
ごめんなさい。そうでした。画面に出ている599は受信RSTですね。送信RSTは変更できません。

closed 2026-02-22 by irukabando