jr8ppg/zLog #835: リグコントロールでバンド変更コマンドへの対応

jr8ppg, 2025-07-30:
YAESU
・FT-2000以降でBSコマンドが使用可能。

KENWOOD
・TS-2000以降でBD/BUコマンドが使用可能。

ICOM
・対応コマンド無し

jj1cvh, 2025-08-31:
YAESUやKENWOODと同じ動きをするかは分かりませんが、IC-756ProII, IC-7300, IC-9700の取説を見た限りでは、ICOMは1A 01コマンド(バンドスタッキングレジスターの内容設定)があります。

但し、IC-910/Dは1A 01コマンドが別の機能に割り当てられていました。

また、理由はわかりませんが、HamlibのIC-7300ではコメントアウトされていました。
https://github.com/Hamlib/Hamlib/blob/master/rigs/icom/ic7300.c
// RIG_PARM_BANDSELECT disabled until Icom can describe the return from 0x1a 0x01
//#define IC7300_PARMS (RIG_PARM_ANN|RIG_PARM_BACKLIGHT|RIG_PARM_TIME|RIG_PARM_BEEP|RIG_PARM_BANDSELECT)

jr8ppg, 2025-08-31:
情報ありがとうございます。マニュアル見てみます。

jr8ppg, 2025-11-26:
Use band up/down commands ・・・ YAESU FT-2000とそれ以降、 KENWOOD TS-2000とそれ以降。(BU/BD)
Use band select command ・・・ YAESU FT-2000とそれ以降(BS)

BU/BDはQSY先バンドがわからないため、AntSelectが正常動作しないと思われる。（QSY先はAI受信でわかるため）
BSは特に不都合はないと思われる。

jr8ppg, 2026-01-19:
実機確認できてないけどクローズ

closed 2026-01-19 by jr8ppg